prenticedavid/MCUFRIEND_kbv #92: Change the data port on bluepill

BadWolf-i, 2019-04-26:
Hi, I want to transfer the DATA pins to another port (port b), I changed the pins in my mcufriend_shield.h file to my own and changed the data port accordingly, but in the end I got a white screen .. I use stm32duino but I can switch to the HAL library if needed..
[image]
what am i doing wrong?

prenticedavid, 2019-04-26:
Choose your wiring scheme.
Edit the defines in LCD_ID_readreg sketch

Verify that your wiring is correct.   i.e. the registers all read correctly.

Then post your defines.

Life is much simpler if you follow the default BLUEPILL wiring scheme.
There is a MY_BLUEPILL scheme in special.h
And there are several other schemes in the extras/unused folder. 

Oh.   You will get more readers if you choose a less unpleasant colour scheme when posting code.

David.

BadWolf-i, 2019-04-26:
Thank you for answering so quickly! So I changed the LCD_ID_readreg sketch as you said, and got these results, in my opinion everything is as it should be. (I am using ILI9486)
[image]
[image]

P.S. This is a standard theme of VS Code.

BadWolf-i, 2019-04-26:
Also i noticed here that it is not necessary to use the CS pin, you can put GND on this leg of the display. or am i wrong?

prenticedavid, 2019-04-26:
Your code should be
```

// configure macros for the data pins
#define write_8(d)    {DATA_PORT->REGS(BSRR) = 0x00FF << 16; DATA_PORT->REGS(BSRR) = (d) & 0xFF; }
#define read_8()      (DATA_PORT->REGS(IDR) & 0xFF)
//                                         PB7 ..PB0
#define setWriteDir() {GP_OUT(DATA_PORT, CRL, 0xFFFFFFFF); }
#define setReadDir()  {GP_INP(DATA_PORT, CRL, 0xFFFFFFFF); }

```

You had omitted the BSRR statement that clears the pins.

Yes,  it is a good idea to say DATA_PORT but very few punters will ever have a contiguous port.
You might just as well changed GPIOA to GPIOB.  And edited the comment line.

David.

BadWolf-i, 2019-04-26:
This is very very strange. I changed what you specified, but I still got a white screen. But the strangest thing is that I loaded graphictest_kbv and he wrote ID = 0x8286, I checked many times, loading LCD_ID_readreg which says that everything is fine and right after that this test, so that the problem with the connection can be eliminated. I'm completely confused ...

[image]

prenticedavid, 2019-04-27:
PB2 is BOOT1 pin.     I suggest that you avoid it.
PB4 is JTRST pin.      Roger's init restores it to GPIO.   The library does it for STM core.

I suggest that you use an existing wiring scheme.   The data bus does not need to be  contiguous port pins (but it does make the  SPECIAL simpler to write)

I get worried when punters suggest hard wiring CS low.    It implies you view convention with contempt.    

David.

BadWolf-i, 2019-04-27:
Thank you for the answer and help to me.  It's a shame that it's not possible to properly transfer data port.  But why do I want to transfer a port that is designed to release the analog inputs that I need.  And also release some PWM outputs.  I specifically tried to merge the port, without changing the order of the outputs, so that it would be easier to carry out the transfer as much as possible.

prenticedavid, 2019-04-27:
I did not design the BluePill.   If you want contiguous pins on GPIOB use PB4-PB11 
But it is not unreasonable to use PB0-1, PB3-PB7 and any random pin to replace PB2.

Do you actually use the BOOT0 and BOOT1 jumpers?

You could replace the BOOT1 jumper with a 10k resistor.   It will retain its Boot function without interfering  with the TFT.

David.

BadWolf-i, 2019-04-27:
No, I do not use jumper. I program the bluepill with a StLink v2, so I do not use these jumper, and they are in a zero position. Essentially, I do not need them. 

If I use foam PB4-11, I can use foam PB0-3 for my needs? Thanks.

prenticedavid, 2019-04-27:
If the jumpers are removed from BOOT1,  PB2 should work fine in your original scheme i.e. using PB0-PB7.

I use STlinkv2 for programming BluePill.   But I generally use Nucleo boards.

David.

BadWolf-i, 2019-04-27:
But it does not work .. I checked the outputs with the oscilloscope, and it showed that on all lines except for 3-4 there are some signals. I tried to directly turn on and off these pins, in the program at the expense of a blink  and 4 pin generated the signal and 3 no signal. Is it possible to transfer 3 pins to another, for example 8? To test my theory .. Thank you in advance ..

prenticedavid, 2019-04-27:
Please write an accurate message.   You can always edit any typos.

I would expect all the GPIOB pins to work.  If PB3 or PB4 do not work,  it means that Roger's core has not disabled JTDO or JTRST.

Seriously.   You want to choose your pins.   Test them.   Then ask for help.
USE_PIONSCOR_BLUEPILL which is in extras/unused/special2 uses data pins DB2 on PA15.  All others on PB0-1, PB3-15 for a 16-bit data bus.

David.

BadWolf-i, 2019-04-28:
Well, I managed to successfully run the display. This requires:
1. Disable JTAG for correct use of PB3-4. for this you need a file that is located in
` .platformio\packages\framework-arduinoststm32\STM32F1\variants\generic_stm32f103c\board.cpp`
 and replace
`#ifndef`  to  `#ifdef`
 in the line `#ifndef CONFIG_MAPLE_MINI_NO_DISABLE_DEBUG`
then in this file
 `.platformio\packages\framework-arduinoststm32\STM32F1\system\libmaple\stm32f1\include\series\gpio.h` 
replace  `AFIO_DEBUG_NONE = AFIO_MAPR_SWJ_CFG_NO_JTAG_NO_SW`
to  `AFIO_DEBUG_NONE = AFIO_MAPR_SWJ_CFG_NO_JTAG_SW`
Done.

2. It is necessary to ensure the normal operation of PB2, it is on the jumper BOOT1, so it needs to be removed, and also replace the resistor R4 on the jumper.

Note: If PlatformIO does not want to download the firmware. Generally download it through the STM32 ST-LINK Utility. For this, in PlatformIO, we compile the project, then the `.pioenvs\genericSTM32F103C8\firmware.bin`  file (located in the project folder) open in the STM32 ST-LINK Utility and flashed. I hope this will be useful to somebody.

prenticedavid, 2019-04-28:
I would not trust PlatformIO
Surely it is safer to use the regular Arduino IDE.

When it is proven on Arduino,  you can test with PlatformIO

Thanks for the tip about Roger's Core.   I know that it disables JTRST.   I would expect it to disable JTDO as well.
I use the STM Core.   I would presume that most people would prefer STM.     Of course the "older" stm32duino members use Roger's Core.

It is not reasonable for me to ask / suggest a user to alter any System files.
There are specific Maple functions to enable/disable things like JTRST. 
And I can put regular CMSIS statements in my GPIO_INIT() e.g. AFIO->MAPR

JTAG is a bit of a mystery on the F103.   SWD works pretty well.

I have hardware for BLUEPILL, MY_BLUEPILL,  USE_PIONSCOR_BLUEPILL, NUCLEO64, NUCLEO144.
I have no intention of testing every random wiring.
Thanks for your report about PB3 (and PB4).    I will investigate Roger's Core.

David.

BadWolf-i, 2019-04-28:
You also can write in the top of seup() function this: `disableDebugPorts();` But i don`t test it.

closed 2019-04-29 by BadWolf-i