prenticedavid/MCUFRIEND_kbv #93: STM32H743ZI Nucleo board successfully tested.

MagicianT, 2019-05-10:
HI,

All it takes to run a library on stm32h743zi (nucleo-144) is to modify 
~/utility/mcufriend_shield.h  in two locations:
`#define IS_NUCLEO144 ( defined(ARDUINO_NUCLEO_F207ZG) \
                   || defined(ARDUINO_NUCLEO_F429ZI) || defined(ARDUINO_NUCLEO_F767ZI) \
                   || defined(ARDUINO_NUCLEO_L496ZG) || defined(ARDUINO_NUCLEO_L496ZG_P) \
                   || defined(ARDUINO_NUCLEO_H743ZI) \
		    )
`
and

`#elif defined(STM32F767xx)            <-  this already exist, help you find where insert a new code
#warning DELAY macros untested yet

#define WRITE_DELAY { WR_ACTIVE8;WR_ACTIVE2; } //216MHz
#define IDLE_DELAY  { WR_IDLE2;WR_IDLE; }
#define READ_DELAY  { RD_ACTIVE16;RD_ACTIVE16;RD_ACTIVE4;}

#define GPIO_INIT()   { RCC->AHB1ENR |= RCC_AHB1ENR_GPIOAEN | RCC_AHB1ENR_GPIOCEN | RCC_AHB1ENR_GPIODEN | RCC_AHB1ENR_GPIOEEN | RCC_AHB1ENR_GPIOFEN; }
#define PIN_OUTPUT(port, pin) PIN_MODE2((port)->MODER, pin, 0x1)

#elif defined(STM32H743xx)   <-  new part.
#warning >STM32H743xx< DELAY macros untested yet

#define WRITE_DELAY { WR_ACTIVE8;WR_ACTIVE2; } //200MHz
#define IDLE_DELAY  { WR_IDLE2;WR_IDLE; }
#define READ_DELAY  { RD_ACTIVE16;RD_ACTIVE16;RD_ACTIVE4;}

#define GPIO_INIT()   { RCC->AHB4ENR |= RCC_AHB4ENR_GPIOAEN | RCC_AHB4ENR_GPIOCEN | RCC_AHB4ENR_GPIODEN | RCC_AHB4ENR_GPIOEEN | RCC_AHB4ENR_GPIOFEN; }
#define PIN_OUTPUT(port, pin) PIN_MODE2((port)->MODER, pin, 0x1)
`
Good news, library runs flawlessly. The not so good news, is that for unknown for me reason H743  board is about 5 (!!!) times slower than F767 running build-in graphics_kbv benchmark.
There is option to accelerate by replacing:  
`#define WRITE_DELAY { WR_ACTIVE; } //200MHz
#define IDLE_DELAY  { WR_IDLE; }
#define READ_DELAY  { RD_ACTIVE;}
`
still 2.5 times slower.

prenticedavid, 2019-05-11:
In the Adafruit Tests Report : what does F_CPU say?

I have no idea how STM32H743 compares to F767.   I presume that the GPIO writes are similar to any Cortex-M4

Either your F_CPU is much slower or the compiler is doing really SLOW writes to the hardware pins.
Put a breakpoint in the debugger or just inspect the LSS file.

David.

MagicianT, 2019-05-11:
Here is the report:

 Serial took 0ms to start
ID = 0x9486
MCUFRIEND 2.99 UNO
FillScreen               875563
Text                     25855
Lines                    515536
Horiz/Vert Lines         71397
Rectangles (outline)     40012
Rectangles (filled)      2134143
Circles (filled)         276067
Circles (outline)        221123
Triangles (outline)      143537
Triangles (filled)       690744
Rounded rects (outline)  89524
Rounded rects (filled)   2318388
Total:7.40sec
ID: 0x9486
F_CPU:400.00MHz

 All frequency settings are correct. I 've done some testing, extracting write_8() macros out of the library, and verifying cortex-M7 access time to GPIO ports.

```
#define REGS(x) x

#if defined(STM32F767xx)
#define GPIO_INIT()   { RCC->AHB1ENR |= RCC_AHB1ENR_GPIOAEN | RCC_AHB1ENR_GPIOCEN | RCC_AHB1ENR_GPIODEN | RCC_AHB1ENR_GPIOEEN | RCC_AHB1ENR_GPIOFEN; }
#elif defined(STM32H743xx)
#define GPIO_INIT()   { RCC->AHB4ENR |= RCC_AHB4ENR_GPIOAEN | RCC_AHB4ENR_GPIOCEN | RCC_AHB4ENR_GPIODEN | RCC_AHB4ENR_GPIOEEN | RCC_AHB4ENR_GPIOFEN; }
#endif

// configure macros for the data pins
#define DMASK ((1<<15))                         //#1
#define EMASK ((1<<13)|(1<<11)|(1<<9))          //#3, #5, #6
#define FMASK ((1<<12)|(1<<15)|(1<<14)|(1<<13)) //#0, #2, #4, #7

#define write_8(d) { \
        GPIOD->REGS(BSRR) = DMASK << 16; \
        GPIOE->REGS(BSRR) = EMASK << 16; \
        GPIOF->REGS(BSRR) = FMASK << 16; \
        GPIOD->REGS(BSRR) = (  ((d) & (1<<1)) << 14); \
        GPIOE->REGS(BSRR) = (  ((d) & (1<<3)) << 10) \
                            | (((d) & (1<<5)) << 6) \
                            | (((d) & (1<<6)) << 3); \
        GPIOF->REGS(BSRR) = (  ((d) & (1<<0)) << 12) \
                            | (((d) & (1<<2)) << 13) \
                            | (((d) & (1<<4)) << 10) \
                            | (((d) & (1<<7)) << 6); \
    }

//                                             PD15                PE13,PE11,PE9          PF15,PF14,PF13,PF12
#define setWriteDir() { setReadDir(); \
                        GPIOD->MODER |=  0x40000000; GPIOE->MODER |=  0x04440000; GPIOF->MODER |=  0x55000000; }
#define setReadDir()  { GPIOD->MODER &= ~0xC0000000; GPIOE->MODER &= ~0x0CCC0000; GPIOF->MODER &= ~0xFF000000; }

  volatile  uint32_t temp1, temp2;

void setup()
{

  GPIO_INIT();
  setWriteDir();
/*
  __HAL_RCC_CSI_ENABLE() ;
  __HAL_RCC_SYSCFG_CLK_ENABLE() ;
    HAL_EnableCompensationCell();  
*/
}

void loop()
{
  // H743 = 2.777 MHz
  // F767 = 4.908 MHz
  
    temp1 = 0xAAAAAAAA;
    temp2 = 0x55555555;
    
    for( uint32_t i = 0; i < 0x1000000UL; i++) {  
      write_8(temp1);
      write_8(temp2);
      }
}
```
I think, that stm32H743 internal architecture is mistakenly design, stm assigns GPIO to AHB4.  stm32F767 has access over AHB1.  Just to remind that 400 MHz core is not necessarily faster than 216 MHz, in some aspects it may be slower by factor 5 !.

prenticedavid, 2019-05-11:
I bought a F767 Nucleo but never used in anger.
I will see what ASM instructions are generated compared to F4.
Either way,   the ILI9486 still has a minimum 60ns write cycle.    A faster core can do faster maths but you have to limit the GPIO to comply with 60ns.
From memory,   Adafruit Tests on 320x480 will never be quicker than about 1.6 seconds e.g. with an Xmega.
Nucleo-144 have random port pins.    2.5 seconds is a reasonable expectation for F767 and H743..

I will have a look later.   Rendering JPEGs should be faster with F7 or H7 than M4 or Xmega.

David.

prenticedavid, 2019-05-11:
I ran the regular graphictest_kbv with an ILI9486 on the Nucleo-F767

-Os:  total=1.47sec fillScreen=153ms
-O1:  total=1.50sec fillScreen=160ms
-O3:  total=1.42sec fillScreen=156ms
-Og:  total=2.19sec fillScreen=259ms

So my guesswork was not far wrong.
STM32Cube shows that the APB peripheral clock is always limited to relatively low speeds.

How did you manage to get longer  times for your F767 ?

Please Fork the library.   Then you / I can compare code.

I have not looked at generated instructions.   (I am going out)
But displaying JPEGs with F767 is impressive.

David.

MagicianT, 2019-05-11:
Its' not like " paître les moutons " in your "kishlak".
I have issue with stm32 -HHHH-743 board. With stm32-FFFF-767 my numbers are the same like yours.
Will ask on the stm-community web hosting.

prenticedavid, 2019-05-11:
I had to Google  " paître les moutons " in your "kishlak".
I am still not sure what you mean.

If the GPIO has a much slower clock,  you can change the WRITE_DELAY macros.
Because of pipelining and cacheing you can not just use NOPs to create the cycle timing.
That is why I use multiple volatile statements e.g. WR_ACTIVE

It would be handy if you Fork.   Then between us we can get this issue resolved.

Incidentally,   the Adafruit Tests may be visual but the effective performance depends on:
GPIO for drawPixel(),  fillRect() and the raw computing power for calculating graphics.

The tests that illustrate this best are angled Lines, Filled Screen,  Filled Triangles.

David.

closed 2021-08-24 by prenticedavid