prenticedavid/MCUFRIEND_kbv #95: LCD ID 0x4747 using STM32F103C8T6

nikjayswal, 2019-05-27:
LCD_ID_readreg:
ngle 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 47 47	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 3F 3F 3F 3F 3F	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 01 01	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 80 80	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00 Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03	GAMMA-P
reg(0x00E1) 49 49 49 49 49 49 49 49 49 49 49 49 49 49 49 49	GAMMA-N
reg(0x00EF) 49 49 49 49 49 49	ILI9327
reg(0x00F2) 10 10 10 10 10 10 10 10 10 10 10 10	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

Hi , i want to use SPI Pins but no idea how to connect them for displaying anything like normal tft display i want to use DMA for speedup Display.

prenticedavid, 2019-05-27:
You have a Himax HX8347-D display with 8080-8 parallel interface.

MCUFRIEND_kbv only supports 8080-8 and 8080-16 parallel interfaces.

It is very unlikely that your display has access to the interface pins.    If you can configure the interface for SPI,  you will have to find a library that supports SPI HX8347-D.

David.

nikjayswal, 2019-06-11:
Thank You David for Replay i need more help for another issue getting white screen.

[DMA_Buffer_1.9.4_Menu.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3277610/DMA_Buffer_1.9.4_Menu.zip)

the display getting White Screen when enabling TouchScreen library in void loop i had commented in Attached Arduino File but if i disable touchscreen in void loop is working normal.

Thank You in Advance.

Nik.

prenticedavid, 2019-06-11:
You have:

```
  TSPoint_kbv p = ts.getPoint();
 
 if (p.z > MINPRESSURE && p.z < MAXPRESSURE){
     p.x = map(p.x, 499, 721, 0, 240);
     p.y = map(p.y, 112, 887, 0, 340);
     button = p.z/10;
 }

```
1.  you have omitted to restore XM, YP to OUTPUT after calling ts.getPoint()
2.  you are mapping to 240, 340 when a HX8347 controller is naturally 240x320
3.  your calibration looks pants.  e.g. TS_LEFT=499,  TS_RT=721
4.  you appear to have chosen LANDSCAPE_REV which implies pixels would be 320x240
5.  read the ID before calling tft.begin(ID)
6.  otherwise you will install a different screen and nothing will work.

Measure the X, Y Touch pin resistance with a DMM.   
Then use the correct XM,XP YM,YP pins in the constructor.
Run the calibration sketch.   (which gives the PORTRAIT calibration)

David.

nikjayswal, 2019-06-11:
It work if i disable the dma in void setup the display and touchscreen both works properly but if i enable dma " display showing only white and if i disable touchscreen settings it works "

prenticedavid, 2019-06-11:
Please design and write your sketch properly.

I suggest that you do not use DMA for the ADC.    The Touchscreen library is using the ADC in polled mode.

It is wise to test your project in stages.   e.g. test Serial, test TFT, test Touch, ...
Take notes.   You will see if and where problems arise.

I did not write the Adafruit_Touchscreen library.    Touchscreen_kbv is only a simplified version of the Adafruit library.
You could probably adapt for DMA.   I doubt if there is any advantage in using DMA.

 David.

nikjayswal, 2019-06-12:
Thank you David it working now

Nik.

nikjayswal, 2019-07-01:
As you told that SPI not supporting in this Himax HX8347-D display library so i search in Google and found this mbed library can it work in Arduino ?

https://os.mbed.com/users/dreschpe/notebook/qvga-tft-with-hx8347d-controller/

prenticedavid, 2019-07-01:
Every TFT controller that I have seen can use SPI.   (except SSD1963)
However very few TFT modules give access to the hardware IM# pins.

Likewise,  the modules that do use SPI seldom give access to the parallel pins.

The HX347-D module in your link does have two IM# pins.
Some ILI9341 boards from BuyDisplay and Adafruit have got IM# pins.
The HX8357 board from Adafruit has got some IM# pins.

If you actually buy a board with IM# pins,   I can show you how to use SPI.

Most parallel modules let you change between 8080-8 and 8080-16 interface mode by changing an 0R surface-mounted resistor on the ribbon.

David.

nikjayswal, 2019-07-01:
My TFT doesn't have IM pins but only SD Pins can it use for SPI I don't want use SD Card

prenticedavid, 2019-07-01:
No.   You can **only** use the 8080-8 interface that is hard-wired to your Shield.

Life would be simpler if you just explained what you actually want to do.
https://forum.arduino.cc/index.php?board=7.0 would also be a more appropriate forum to ask this sort of question.

The 8080-8 interface is faster than the SPI interface.
You use more pins for any parallel interface.

There are still "spare" GPIO pins left on a BluePill e.g. for external Sensors, SPI, I2C ...

David.

nikjayswal, 2019-07-01:
The 8080-8 interface takes more pins than SPI can i use I2C in this LCD ? And if use I2C how can i connect them and use it

nikjayswal, 2019-07-01:
Did you have any other to reduce pin ?

prenticedavid, 2019-07-01:
Yes,   BUY a proper SPI display e.g. https://www.ebay.co.uk/itm/3-3V-240x320-2-4-SPI-TFT-LCD-Touch-Panel-Serial-Port-Module-with-PCB-ILI9341/401087406846

Since you have a 3.3V BluePill there is no need for any level-shifters.

David.

nikjayswal, 2019-07-02:
can shift register like 74HC165 can be use for converting SPI to parellel to drive TFT ?

nikjayswal, 2019-07-02:
This tft takes more pins and buying one more tft takes too much time and costly too

prenticedavid, 2019-07-02:
74HC165 is Parallel to Serial.
You have a Parallel Display.
It works fine with a BluePill.
The BluePill drives the HX8347-D display as fast as the HX8347-D controller can handle.
You have several spare GPIO pins.
Please put your brain into gear.
David.

nikjayswal, 2019-07-03:
thank you David.

nikjayswal, 2019-07-03:
i don't closing this post may be help to other...

prenticedavid, 2019-07-03:
I am happy to answer questions.
But first of all you need to think about what you want to do.
And what you want to ask.

David.

nikjayswal, 2019-07-06:
I had bought 74HC164 but I don't know how to connect and which library should i use

prenticedavid, 2019-07-06:
Ok,  so 74HC164 is Serial to Parallel.

What do you want me to about it?

Read the mcufriend_how_to.txt
Configure a SPECIAL to match your wiring for MCUFRIEND_kbv  
Run all the MCUFRIEND_kbv examples.
Ask questions about a specific example.

What country do you live in?
Are you still at school?    Or at University?

I will help with _well presented questions_ about MCUFRIEND_kbv.
General questions about electronics would be better answered on Tutorial Forums or via printed textbooks.

David.

nikjayswal, 2019-07-07:
74HC164 has DSA,DSB = may be MOSI(Data pins), CP = Clock Pin, MR = Reset pin. But doesn't have SS, MISO Pins i m confused how to connect them in my tft display i had connected D0 to D7 in my tft but other 4 pins how to connect them. 
I am not a school or university student i am a hobiest to do this work in part time i m not a professional in Arduino.

prenticedavid, 2019-07-07:
Whether you are a student or hobbyist you  stlll need to run MCUFRIEND_kbv examples first.

Bad English is fine.   Just compose your message carefully in your own language.    
Use Google Translate..

There will be Electronics Tutorials and printed textbooks in your local language.
What country do you live in?

David.

nikjayswal, 2019-07-07:
I had run examples from Mcufriend and it works good via direct parellel interface but it takes more pins also not working touchscreen after eneble drawling in either works good e.g drawing square or full screen but when i drawline touchscreen not working https://github.com/prenticedavid/MCUFRIEND_kbv/issues/102

so i decided to transfer using SPI interface it takes less pins and use via buttons not using touchscreen

prenticedavid, 2019-07-07:
There are Touch examples.
Run each Touch example.
Report any problem. Quote which example.

nikjayswal, 2019-07-07:
Touch examples works great i had only problem occurs when i drawline in display as shown on given link

nikjayswal, 2019-07-07:
As shown in image x = 1 or 2 only

prenticedavid, 2019-07-07:
Your #102 link is not written correctly.
Read the mcufriend_how_to.txt file to see how to use a SPECIAL.

David.

nikjayswal, 2019-07-07:
[without tft.drawFastVLine.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3365966/without.tft.drawFastVLine.zip)
[With tft.drawFastLine.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3365967/With.tft.drawFastLine.zip)

prenticedavid, 2019-07-07:
I asked you to ZIP up your project.    i.e. post the .INO file and your mcufriend_special.h file.
It is convenient to put multiple files in a single ZIP.
Not post a video.

I can't read your source code from a shaky video.
And I don't believe that you have run the Touch examples from the MCUFRIEND_kbv library.

If you have run an example,  please post the two "Calibration" lines that you replaced in the example.

David.

nikjayswal, 2019-07-08:
All files with Library

[TFT_Touch_Error.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3367077/TFT_Touch_Error.zip)

Touchscreen Calibartion Example edited Not Works
[TouchScreen_Calibr_native.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3367082/TouchScreen_Calibr_native.zip)

prenticedavid, 2019-07-08:
Thanks for the ZIPs.   This means that I can see your problem(s).

mcufriend_special.h:
```
// only define one "USE_XXX" macro at any time
...
#define USE_MY_BLUEPILL
#define USE_NIKJAYSWAL_BLUEPILL
...
//LCD pins  |D7  |D6  |D5  |D4  |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST|
//STM32 pin |PB15|PB14|PB11|PB10|PB9|PB8|PB7|PB6| |PA3|PA2|PA1|PA0|PA8|
...
```
You have placed your  USE_NIKJAYSWAL_BLUEPILL block before the USE_MY_BLUEPILL block
So it will compile your _NIKJAYSWAL_ special ok.    But you have disobeyed the one USE_xxx advice.

Touchscreen_Calibr_native.ino

```
int XP = PB6, YP = PA0, XM = PA1, YM = PB7;  //most common configuration

```

You have not measured the X and Y resistance with a DMM.   The Calibration sketch MUST have correct pin definitions.   Assuming that your Error.ino file has the correct pins. 

```
int XP = PB7, YP = PA1, XM = PA0, YM = PB6;  //your pins

```

Only you have the shield on your desk.  Only you can measure the resistance.   Only you can post the resistance in your message.   
e.g. LCD_RS and LCD_D0 = 500R.   LCD_CS and LCD_D1 = 300R.
I am sceptical.    Please confirm the actual resistances that you measured with a DMM.

David.

nikjayswal, 2019-07-08:
LCD_RS and LCD_D0 = 12.2 Mohms
LCD_CS and LCD_D1 = 13.25 Mohms

nikjayswal, 2019-07-08:
LCD_RS and LCD_D0 = 10.5 MOhms
LCD_CS and LCD_D1 = 9.3 MOhms.

prenticedavid, 2019-07-08:
13500000R is a lot more than 500R.
9500000R  is a lot more than 300R.

I do not think that you will ever understand my advice.
I suggest that you ask an Indian friend or neighbour for advice.

Life is much simpler with a regular  Uno and regular Mcufriend shield.
If you were taught in school about male pins and female sockets you plug the Shield into the Arduino.

Run the Calibration sketch.
It detects the pins.    It reports the Calibration values.

You copy-paste  the Serial Terminal to  your message.

David.

nikjayswal, 2019-07-08:
Thank you David for help and replying but i am tired now and don't want to use touchscreen I will try to use only buttons interface

nikjayswal, 2019-07-08:
hey David can  you tell me how to change DATA pins in mcufriend_special.H ?

prenticedavid, 2019-07-09:
No.   There seems little point in my replies.
Ask for help from Indian friends.
They can show you  how to use a multimeter.
They can show you how to plug your shield into  an Arduino.

Come back wen you have made progress.

David.

nikjayswal, 2019-07-09:
LCD_WR and LCD D7 = 365R and other pins not showing any resistance i doubt that my tft touchscreen is damaged

nikjayswal, 2019-07-09:
Can you please change the pins into LCD_CS PB6
LCD_CD PB7
LCD_WR PB8
LCD_RD PB9
LCD_RESET PA8

LCD_D0 PB12
LCD_D1 PB13
LCD_D2 PB14
LCD_D3 PB15
LCD_D4 PA0 
LCD_D5 PA1
LCD_D6 PA2
LCD_D7 PA3

prenticedavid, 2019-07-09:
> I had removed all pins from tft and check with DMM . The DMM Showing LCD_WR and LCD D7 = 365R and other pins not showing any resistance i doubt that my tft touchscreen is damaged

Yes.   It is damaged.    I would expect LCD_RS and LCD_D6 to have about 500R.    If you do not get less than 1000 ohms the screen is broken.    Hint.   A cracked glass means "broken"

> Can you please change the pins into

No.

nikjayswal, 2019-07-09:
[image]

Screen not broken but not showing any resistance

nikjayswal, 2019-07-09:
Could you please change the pins i need analogpins

LCD_CS PB6
LCD_CD PB7
LCD_WR PB8
LCD_RD PB9
LCD_RESET PA8

LCD_D0 PB12
LCD_D1 PB13
LCD_D2 PB14
LCD_D3 PB15
LCD_D4 PB10
LCD_D5 PB11
LCD_D6 PA0
LCD_D7 PA1

prenticedavid, 2019-07-09:
No.

nikjayswal, 2019-07-09:
How could i change my own pin configuration for data pins ?

nikjayswal, 2019-07-09:
I m confused why you telling me " No " if i hert you then i say sorry.

prenticedavid, 2019-07-09:
Read your messages.     They make no sense to a non-Indian.
Ask for help from fellow Indians.     They might understand you.

I am only an English speaker.    If my answers are not clear, say so.    Perhaps your fellow Indians can advise me how to write better answers.

David.

nikjayswal, 2019-07-09:
David my indians friends/advicer is not genius like you. i will try to write better question/issues. Now i will write the messages in Google translate and then after i will send to you. Ok

prenticedavid, 2019-07-09:
India is a big country with lots of very clever microcontroller programmers.
I presume that there are Indian versions of StackExchange,  GitHub,  Arduino,  ...
And there are many fellow countrymen who can assist you.

It has taken many days and messages to discover that you had never measured Touch pins or run examples on a Uno.

I am only English speaker.  So I have sympathy for readers that are not native-English speakers.
But I find it very difficult when I receive messages that are clearly wrong.     If you have not run examples just say so.

Please compose accurate messages in your own language.   Then use Google Translate.

David.

closed 2019-07-09 by nikjayswal