prenticedavid/MCUFRIEND_kbv #96: Screen goes blank on tft.Print

leversole, 2019-06-15:
I am new to TFT LCD stuff...but trying to learn...my sketch draws three buttons, adds text and waits for a button to be touched, then writes a message to the serial window. All of this works as expected...instead of the serial window, I want to update the screen with a message. No matter what I have tried, when I press the button (with the tft.print, the other buttons still work), it makes the screen go blank (white/grey). I feel I am missing something simple...here is the loop part of the sketch...thanks for any insight

void loop() {
  
  TSPoint p = ts.getPoint();

  if (p.z > 100) {  //did we touch the pad?

     p.x = map(p.x, TS_MINX, TS_MAXX, 0,240);  //map the touch point x to the screen x
     p.y = map(p.y, TS_MINY, TS_MAXY, 0,320);  //map the touch point y to the screen y

    //Serial.print("X = "); Serial.print(p.x);
    //Serial.print(" Y = "); Serial.println(p.y);
    //Serial.println(tft.height());
    //Serial.println(tft.width());

    if(p.x>20 && p.x<220 && p.y>30 && p.y <70) {  //did we touch within button 1
       
        //Serial.println("you pressed button 1");
        tft.setCursor(20, 20);
        tft.setTextSize(4);
        tft.setTextColor(BLUE);
        tft.print("What is up?");
        
    }
    
    if(p.x>20 && p.x<220 && p.y>80 && p.y <120) {  //did we touch within button 2
        Serial.println("you pressed the button 2");
    }    

    if(p.x>20 && p.x<220 && p.y>130 && p.y <170) {  //did we touch within button 3
        Serial.println("you pressed the button 3");
    }    
  
    delay(300);
         
  }
}

prenticedavid, 2019-06-15:
Look at button_simple.ino:

```
bool Touch_getXY(void)
{
    TSPoint p = ts.getPoint();
    pinMode(YP, OUTPUT);      //restore shared pins
    pinMode(XM, OUTPUT);
    digitalWrite(YP, HIGH);   //because TFT control pins
    digitalWrite(XM, HIGH);
    bool pressed = (p.z > MINPRESSURE && p.z < MAXPRESSURE);
    if (pressed) {
        pixel_x = map(p.x, TS_LEFT, TS_RT, 0, tft.width()); //.kbv makes sense to me
        pixel_y = map(p.y, TS_TOP, TS_BOT, 0, tft.height());
    }
    return pressed;
}

```
or any other Touch example:

```
    tp = ts.getPoint();   //tp.x, tp.y are ADC values

    // if sharing pins, you'll need to fix the directions of the touchscreen pins
    pinMode(XM, OUTPUT);
    pinMode(YP, OUTPUT);
    // we have some minimum pressure we consider 'valid'

```

There is no need to write HIGH unless you have a shield that shares a Touch pin with LCD_CS pin.

I can only provide examples.   You have to study them.    But I would appreciate any advice or suggestions for improvement.

David.

leversole, 2019-06-15:
Thank you, not sure if I was clear, I am not having issues with the button press, it is working, I am having issues with writing to the screen. If I comment out the bottom 4 lines below, pressing the button prints to the serial screen just fine, when I try to write to the screen with any combination of the last four lines, the screen goes black (but the serial print still works.

 //Serial.println("you pressed button 1");
    tft.setCursor(20, 20);
    tft.setTextSize(4);
    tft.setTextColor(BLUE);
    tft.print("What is up?");

Thanks

leversole, 2019-06-15:
Not sure I understand, but instead of doing it with getPoint I incorporate d Touch_XY and it is working...scratching my head, but that is okay! ;-)

Thank You

prenticedavid, 2019-06-15:
The function in button_simple.ino is called `bool Touch_getXY(void)`

The Touchscreen library leaves the XM, YP pins in INPUT mode.
**Any** call to the Touchscreen library will ensure the TFT crashes.   Hence you should call pinMode() immediately after a call.

My examples factorise the logic to use a single helper function e.g. Touch_getXY() which looks after pinMode()

My usual advice for Arduino libraries:   RUN EVERY EXAMPLE SKETCH

Develop your project by starting with a "similar" example.   Adapt in stages to implement your requirements.
The Arduino Forum is probably the best place to discuss Arduino topics.    GitHub is more suitable for technical issues.

David.

leversole, 2019-06-15:
I understand now! Thank You! I naturally go into shortcut mode, and end up overlooking important things!

prenticedavid, 2019-06-15:
No problem.

If you have any suggestions to improve examples or comments,   I would be very grateful.

David.

closed 2021-08-24 by prenticedavid