prenticedavid/MCUFRIEND_kbv #100: STM32F103C8T6 with TFT, 8 bits, not standard connection

AntonioFromBrazil, 2019-06-26:
I am trying to put a generic STM32F103C8T6 --> STM32F103C8T6 -> https://www.aliexpress.com/item/32891197661.html to work in 8 bits parallel mode with a TFT ILI9488 using the library MCUFRIEND_kbv.

The connection was made as the following: TFT signals D0, D1, D2, D3, D4, D5, D6, D7 connected to STM32F103C8 PB8, PB9, PB10, PB11, PB12, PB13, PB14, PB15. TFT signals RD, WR, RS, CS connected to STM32F103C8 PA0, PA1, PA2, PA3. TFT reset signal connected to microprocessor reset signal.

I do not know what I forgot but my changes is not working :(

Does anybody would help me pointing where and what the changes should be made to put my project working ?

Thanks a lot.

Antonio

prenticedavid, 2019-06-26:
I suggest that you use the "official" wiring e.g.

```
//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1**

```
Then the library will work out of the box.

Yes,  you can use a SPECIAL wiring if you want.   But it involves you editing mcufriend_shield.h and mcufriend_special.h

Please post your "SPECIAL" if you want me to help.

David.

AntonioFromBrazil, 2019-06-26:
Thanks David. The problem with the "official" wiring is that almost every A/D pins are used and my application need them as A/D ... So the solution is edit the modules mcufriend_shield.h and mcufriend_special.h that is exactly what I am trying to do...

I did not underestand what you mean with "Please post your "SPECIAL" if you want me to help."...

prenticedavid, 2019-06-26:
If you look in extras/unused/mcufriend_special2.h
```

// ######################### STM32 PB8-PB15 RAVI_KANCHAN ###################
#define USE_RAVI_KANCHAN_BLUEPILL
...
#elif  defined(USE_RAVI_KANCHAN_BLUEPILL) && (defined(__STM32F1__) || defined(STM32F103xB)) // MAPLECORE or STM32CORE 
#warning USE_RAVI_KANCHAN_BLUEPILL

#if defined(ARDUINO_NUCLEO_F103C8)   //regular CMSIS libraries
#define REGS(x) x
#define GPIO_INIT()   { RCC->APB2ENR |= RCC_APB2ENR_IOPAEN | RCC_APB2ENR_IOPBEN | RCC_APB2ENR_IOPCEN | RCC_APB2ENR_IOPDEN | RCC_APB2ENR_AFIOEN; \
        AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_1;}
#else                                                                  //weird Maple libraries
#define REGS(x) regs->x
#endif

#define WRITE_DELAY { }
#define READ_DELAY  { RD_ACTIVE; }
#define GROUP_MODE(port, reg, mask, val)  {port->REGS(reg) = (port->REGS(reg) & ~(mask)) | ((mask)&(val)); }
#define GP_OUT(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x33333333)
#define GP_INP(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x44444444)
#define PIN_OUTPUT(port, pin) {\
        if (pin < 8) {GP_OUT(port, CRL, 0xF<<((pin)<<2));} \
        else {GP_OUT(port, CRH, 0xF<<((pin&7)<<2));} \
    }
#define PIN_INPUT(port, pin) { \
        if (pin < 8) { GP_INP(port, CRL, 0xF<<((pin)<<2)); } \
        else { GP_INP(port, CRH, 0xF<<((pin&7)<<2)); } \
    }

#define PIN_HIGH(port, pin)   (port)-> REGS(BSRR) = (1<<(pin))
#define PIN_LOW(port, pin)    (port)-> REGS(BSRR) = (1<<((pin)+16))

#define RD_PORT GPIOA
#define RD_PIN  0
#define WR_PORT GPIOA
#define WR_PIN  1
#define CD_PORT GPIOA
#define CD_PIN  2
#define CS_PORT GPIOA
#define CS_PIN  3
#define RESET_PORT GPIOB
#define RESET_PIN  0

// configure macros for the data pins
#define write_8(d)    { GPIOB->REGS(BSRR) = 0xFF00 << 16; GPIOB->REGS(BSRR) = ((d) << 8) & 0xFF00; }
#define read_8()      ((GPIOB->REGS(IDR) & 0xFF00) >> 8)
//                                          PB15..PB8
#define setWriteDir() {GP_OUT(GPIOB, CRH, 0xFFFFFFFF); }
#define setReadDir()  {GP_INP(GPIOB, CRH, 0xFFFFFFFF); }

#define write8(x)     { write_8(x); WRITE_DELAY; WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; READ_DELAY; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }

```

Untested.  You might need to compare with an existing SPECIAL.    But you should be able to see the general technique.
I strongly advise you to use a specific GPIO pin for LCD_RST.

David.

AntonioFromBrazil, 2019-06-26:
It works !!!! Great !!! Thanks 1k
Antonio

closed 2021-08-24 by prenticedavid