prenticedavid/MCUFRIEND_kbv #101: Touch is not working for STM32 Blue Pil

Flo082002, 2019-06-30:
Hi,
I use a stm32 Blue Pil board for my ili9486 display.
This works fine. But the touch is not working because the touch pins from the display are no analog pins on the STM32.
I use the "official" wiring
```
//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| 
```
My Uno works with the display:  XP = 8, XM = A2, YP = A3, YM = 9;

prenticedavid, 2019-06-30:
Thanks for the well presented question.

Although the control pins are digital the databus pins are Analog.
So run the calibration with:
`XP = PB7, XM = PA0, YP = PA0, YM = PB8;`

David.

Flo082002, 2019-06-30:
Thanks for your help.
This works.

closed 2019-06-30 by Flo082002