prenticedavid/MCUFRIEND_kbv #102: TouchScreen Wrong values on 2.4 TFT

nikjayswal, 2019-07-03:
getting Wrong values from TouchScreen 

Code:

#define USE_NIKJAYSWAL_BLUEPILL
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>    // Core graphics library
#include <MCUFRIEND_kbv.h>

#include "TouchScreen_kbv.h"

#include <Fonts/FreeSans9pt7b.h>
#include <Fonts/FreeSans12pt7b.h>
#include <Fonts/FreeSerif12pt7b.h>

#include <FreeDefaultFonts.h>

#define LCD_CNTRL GPIOA
#define LCD_DATA  GPIOB

#define LCD_CS PB0 // Chip Select goes to Analog 3
#define LCD_CD PB1 // Command/Data goes to Analog 2
#define LCD_WR PA2 // LCD Write goes to Analog 1
#define LCD_RD PA3 // LCD Read goes to Analog 0
#define LCD_RESET PA8 // Can alternately just connect to Arduino's reset pin

#define LCD_D0 PB6
#define LCD_D1 PB7
#define LCD_D2 PB8
#define LCD_D3 PB9
#define LCD_D4 PB10
#define LCD_D5 PB11
#define LCD_D6 PB14
#define LCD_D7 PB15

#define BLACK 0x0000
#define NAVY 0x000F
#define DARKGREEN 0x03E0
#define DARKCYAN 0x03EF
#define MAROON 0x7800
#define PURPLE 0x780F
#define OLIVE 0x7BE0
#define GREY  0x8410
#define LIGHTGREY 0xC618
#define DARKGREY 0x7BEF
#define BLUE 0x001F
#define GREEN 0x07E0
#define CYAN 0x07FF
#define RED 0xF800
#define MAGENTA 0xF81F
#define YELLOW 0xFFE0
#define WHITE 0xFFFF
#define ORANGE 0xFD20
#define GREENYELLOW 0xAFE5
#define PINK 0xF81F

MCUFRIEND_kbv tft;

#define TS_MAXX 868
#define TS_MAXY 319

#define YP LCD_CD  // must be an analog pin
#define XM LCD_CS  // must be an analog pin
#define YM LCD_D0   // can be a digital pin
#define XP LCD_D1   // can be a digital pin
     
#define Xresolution 320 //128
#define Yresolution 240 //64

     #define MINPRESSURE 10
     #define MAXPRESSURE 1900
     TouchScreen_kbv ts(XP, YP, XM, YM, 300);   //re-initialised after diagnose
     TSPoint_kbv tp;                            //global point

int button;

//************************************************************************ Setting For Button Menu  *************************************************************

    int Menu[2];
    int up_button = PA7, down_button = PB0, left_button = PA4, right_button = PA5, E_button = PB13, B_button = PB12;
    int up_down;
    
    unsigned long previousMillis = 0;        // will store last time LED was updated
    // constants won't change:
    const long interval = 300000;           // interval at which to blink (milliseconds)

    int up_Button, up_Button_Last, down_Button, down_Button_Last, left_Button, left_Button_Last, right_Button, right_Button_Last;
    int Enter_Button, Enter_Button_Last, Back_Button, Back_Button_Last;
    int taskbr1 = 2,taskbr2 = 0;
//************************************************************************ END Setting For Button Menu  *************************************************************

//*********************************************************************   DMA Settings  *********************************************************************************

#include <HardwareTimer.h>
#include <STM32ADC.h>

#define pinLED  PC13
#define pinOUT  PA6

#define _SEND_BY_USB
#ifdef SEND_BY_USB
USBSerial usb;
#endif

// Channels to be acquired. 
// A0 (adc1 channel 1)
uint8 pins = PA6;

volatile bool trigger = 0;
long int L1, L2;

long int val;
long int Ary[320];

int LED = PC13;

void setup()  {

  pinMode(pins, INPUT_ANALOG);
 
  Serial.begin(9600); //Serial communication initiliser
  tft.reset();  // reset the TFT Display
  tft.begin (0x4747); // start the TFT Display "tft.begin (TFT Driver ID);"
  tft.setRotation (1);  //rotate the screen upto "0 to 3"
  // SwitAry[1] = 0;

    tft.fillScreen(BLACK);
    
}

  //////////////////////////////////*    Display *///////////////////////////////////////////////////////////
void capture(int y,int i) {
    i = i-1;
    val = analogRead(pins);
    val = val/8;  //val = map(val,0,4095, 0, 320);
    Ary[i] = val;
   // Serial.println(Ary[i]);
}

void plotter(int COLOUR,int pos)  {
  pos = pos - 1;
  L1 = L2;
  L2 = Ary[pos] - 140; //volblock[voldata] + 2;
  
    tft.drawFastVLine(pos, 0, 215, BLACK);
    tft.drawLine(pos - 2, L1, pos - 1, L2, COLOUR);  
}

 void readResistiveTouch(void)
{
    tp = ts.getPoint();
    pinMode(XM, OUTPUT);
    pinMode(YP, OUTPUT);
    
    Serial.println("tp.x=" + String(tp.x) + ", tp.y=" + String(tp.y) + ", tp.z =" + String(tp.z));
    delay(200);
}

void loop() {  

             readResistiveTouch();             
             unsigned long currentMillis = micros();

          if (currentMillis - previousMillis >= interval) {
            // save the last time you blinked the LED
             for(int tri = 0; tri<5; tri++){
              previousMillis = currentMillis;
              if(tri == 4){
                trigger = 1;
                }else {
                  trigger = 0;
              }
             }
             for(int i=0,x=0,j=0; i<1000,x<320 - 1,j<320-1; i++,x++,j++)
             {
                if(trigger == 1){  
                  capture(i, x);
                  plotter(YELLOW, j);
                  }else if(trigger == 0)
                  {
                    plotter(YELLOW, j); 
              }
             }
            }
            
}

[image]

[image]

nikjayswal, 2019-07-04:
when i comment "// tft.drawLine(pos - 2, L1, pos - 1, L2, COLOUR); " the touchscreen works but if uncomment the touchscreen values goes wrong.

prenticedavid, 2019-07-04:
I am happy to answer proper questions.

You are just sending gibberish.    

You could ZIP up your Arduino sketch and attach the ZIP.

But quite honestly,   you should start with the library examples.    And get these working first.

David.

closed 2019-07-08 by nikjayswal