prenticedavid/MCUFRIEND_kbv #103: 2.4" TFT on UNO R3, Width=0

baqwas, 2019-07-13:
Hello,

I took the plunge (after much hesitation regarding _self-inflicted wounds_) and true to form ran into an issue immediately with the **2.4" TFT HAT** on UNO R3.

I uploaded the **drawBitmap_kbv** code to the UNO and the Serial Monitor displayed the following information (I added the width & height print to the sample because the screen flashes momentarily but does nothing else):
```
Width=0
Height=320
ID = 0x9797
```
I used the standard IDE procedure to install the MCUFRIEND and AdaFruit GFX ZIP libraries and there are, of course, no compilation errors. Troubleshooting the Width=0 anomaly is beyond my pay grade/experience. Can anyone tell me what other "debug" statements can be inserted in the sample code to understand why the **TFT** is failing?

Kind regards.

P.S. I have tried the TFT HAT  on two different UNO boards with identical results - no display! The UNO boards are working fine with other attachments/sensors.

prenticedavid, 2019-07-13:
You have an SSD1297 controller.   It is **not** supported by  Release v2.9.8. (i.e. Library Manager)
It is supported by the Beta  (i.e. master branch).   But you have to "enable" the support.

From extras/mcufriend_how_to.txt:
`
SSD1297  240x320  ID=0x9797 #define SUPPORT_1289 (unstable readGRAM())

`
The diagnose_TFT_controller example should tell you the ID and whether it is supported.

ASk if you do not know how to use a Beta.

David.

baqwas, 2019-07-15:
Appreciate the quick response and the guidance.

- Cloned your repository
- Inserted the **#define SUPPORT_1289** statement prior to #include statements
- Made the utility/mcufriend_shield.h & utility/mcufriend_serial.h files available
- Verified: no compilation errors
- Upload: completed without errors
- Opened serial monitor at matching baud rate to see:
```
Serial took 0ms to start
ID = 0x9797
```
The display flashed momentarily but there was no further action on display or in serial monitor window. I suspect that I may have missed some other setting (but definitely some progress thanks to your quick response).

baqwas, 2019-07-15:
Sorry for the **newbie** mistake! Put the #define statement in the wrong (.ino) file. After pasting it in the MCUFRIEND_kbv.cpp everything seems to be working with diagnose_TFT_support. Expect the other routines to work fine too. Will test now but wanted to rush off this note to close the issue in case unnecessary cycles are being consumed by some good Samaritan.

baqwas, 2019-07-15:
Output from diagnose_TFT_support.ino:
```
Diagnose whether this controller is supported
There are FAQs in extras/mcufriend_how_to.txt

tft.readID() finds: ID = 0x9797

MCUFRIEND_kbv version: 2.9.9

PORTRAIT is 240 x 320

Run the examples/graphictest_kbv sketch
All colours, text, directions, rotations, scrolls
should work.  If there is a problem,  make notes on paper
Post accurate description of problem to Forum
Or post a link to a video (or photos)

I rely on good information from remote users
```
graphictest_kbv.ino worked _without even a minor hiccup_:
```
Serial took 0ms to start
ID = 0x9797
```
Many, many thanks, David, for the rapid response. Kind regards.

prenticedavid, 2019-07-15:
Someone sent me a SSD1297 Shield.   So I have encountered it in real life.

Everything worked but readGRAM() was unreliable.
Does it work on your screen?
i.e.. SOFTWARE SCROLL should move across the Penguin screens 
or it says readPixel() error.

David.

baqwas, 2019-07-15:
Hello David,

I saw your comment about **readGRAM()** being unreliable. I did not understand it, so in typical non-developer fashion, I glossed over it. I do see _relatively_ smooth scrolling along the major dimension in _both directions_ (for the Penguin screens). There is no error message in the output frame of the Arduino IDE and the Serial monitor window did not print anything after the two starter lines that were mentioned in my previous note. The code has been running uninterrupted for 18+ hours with my tactical focus on self-documenting the function calls for my own tests. As far as my untrained eye can tell, everything is running without any issue with the test case.

Please do not hesitate to let me know if I can be of service in any way by running other tests with this display (Of course, there is so much to learn that I forget my original mission was plot sensor outputs!)

Kind regards.

prenticedavid, 2019-07-15:
Each Penguin screen does:
SCROLL UP
SCROLL DOWN
BAND SCROLL
SOFTWARE SCROLL.
The GRAY Penguin screen does a final
INVERT SCREEN

You have to watch carefully.    e.g. to check directions and colours are correct.
Some controllers can't hardware scroll at all.   Some can only scroll the  full screen. 
Write-only controllers can never software scroll.
All Read-Write controllers (except 0x1620) can software scroll.

Your SSD1297 either shows a ReadPixel ERROR or the horizontal SOFTWARE SCROLL message corrupts. 

David.

baqwas, 2019-07-15:
Based on your comments, a cursory check with my display reveals:

- No _band_ scroll but color gradient observed
- No software scroll(?), I believe, because only the full frame is scrolled
- There is something on-screen after the scrolling (unsure whether it is related to the message - **readPixel() reads as BGR**)
- rest of the observations  OK as you described in your note (i.e. GRAY & INVERT)

Please allow me a day to get back to you with more authoritative responses:

- video of one entire run (I'll send you a link)
- step-by-step annotation of my understanding of the tests (for checklist purposes with respect to this model of the display)

Kind regards.

baqwas, 2019-07-16:
Hello David,
First pass at the _amateurish_ video on selected tests is at https://youtu.be/CpOBgOswxn4
Kind regards.

prenticedavid, 2019-07-16:
Thanks for the video.   Everything looks as I expected.

The readPIxel() message shows that reading Graphics Memory is is unreliable.

At least most other methods work ok.

David.

closed 2021-08-24 by prenticedavid