prenticedavid/MCUFRIEND_kbv #104: STM32 VLDISCOV

lgomesf, 2019-07-18:
First sorry for the English.
If it is of interest for you to make this change, simply add || defined (ARDUINO_STM32VLD) to the end of line 620 of mcufriend_shield.h that works perfectly with the VLDISCOV STM32 card.

prenticedavid, 2019-07-18:
I actually own a STM32VLDISCOVERY board.  I don't think that I have ever used it with a Parallel TFT.

It does not have Arduino header sockets.   So you can't plug a Shield into it.   So it is not the right place in mcufriend_shield.h

Of course you can add a SPECIAL and put it in mcufriend_special.h

David.

lgomesf, 2019-07-18:
The blue pill also can not connect the shield directly.
But okay, just wanted to help because I had the need and did not see this information anywhere, it works perfectly with the same pins of blue pill.

prenticedavid, 2019-07-18:
The STM32VLDISCOVERY is a nice kit even though there are M4 and M7 chips available now.

closed 2021-08-24 by prenticedavid