prenticedavid/MCUFRIEND_kbv #105: ESP32 devkit v1

igor-tvardyi, 2019-08-16:
I have ESP32 devkit v1 (30 pins)
https://www.aliexpress.com/item/33028843524.html

and display Opensmart 3" R61509V
https://www.aliexpress.com/item/33040167526.html

I did not find pin 17 and 16. So I tried to change them to 5, 18 or 18. 19. But no result.

Could you please help me to make my display work?

2 prenticedavid: Please tell why esp32 is not appropriate for paralel bus?

Thank you!

prenticedavid, 2019-08-16:
RX2 is GPIO16
TX2 is GPIO17

I strongly advise you to use a TTGO/Wemos D1 R32 board.
It has Arduino headers to receive a regular Mcufriend Shield.
You must make a minor hardware mod.

David.

igor-tvardyi, 2019-08-23:
Thank David for your response.

Display is quite unstable with my board. It is possible to get some picture only for few miliseconds. Then display freezes or becomes white.

I have pin "LED", which should be assigned to GND. This works with atmega328.
Is it correct for ESP32?

My pinout is:

RD - 2
WR - 4
RS - 15
CS - 33
RST - 32
LED - GND
D0 - 12
D1 - 13
D2 - 26
D3 - 25
D4 - TX2
D5 - RX2
D6 - 27
D7 - 14

prenticedavid, 2019-08-23:
I strongly advise you to use a TTGO/Wemos D1 R32 board.
It has Arduino headers to receive a regular Mcufriend Shield.
You must make a minor hardware mod.

Mcufriend TFT Shields plug into the Arduino headers.

I would not expect ESP32 to work with long flying wires.
It is a fast MCU.   The signals will have glitches.

ATmega328P is relatively slow.   It still seems to work with bad wires. 

David.

igor-tvardyi, 2019-08-23:
David, could you please take a look at the display.
https://ae01.alicdn.com/kf/HTB1BRWJcEGF3KVjSZFvq6z_nXXaX/3-0-TFT.jpg
It has a long plume.

I use 5cm wires which are soldered to 16 pin female header from the one side and two 8 pin headers from the other side.
Is it possible that display will not work with TTGO/Wemos D1 R32 board also (beacause of long plume on the display)?

prenticedavid, 2019-08-23:
Post a photo of your actual wiring.   It will show the total length of wires.

Header sockets to receive TFT and ESP32 with soldered wiring should be reliable.

Your TFT is 3.3V only.   It will be damaged by a 5V Uno. 

David.

igor-tvardyi, 2019-08-24:
[image]

My actual wiring. All wires have the same width and soldered to female headers.
Board is ESP32 devkit v1 (30 pins)

prenticedavid, 2019-08-24:
Your photo looks as good as you can get.   But each individual signal from TFT glass panel is about 200mm from the ESP32 GPIO pin.

I have one TTGO board  i.e. with Arduino headers.
And one ESP32 module hand-wired to a Protoshield with female headers.

The TTGO board has always been reliable.
The hand-wired arrangement "works" but has occasional glitches.

Obviously I always use displays that mate with Arduino headers.

I do have a 3.3V Open-Smart display that is smaller than yours (176x220 ILI9225).
I have a Protoshield Adapter that receives the 16x1 display header and routes it to Arduino males for mating with a regular Arduino.

I had never run the ILI9225 on an ESP32 before.
I have just plugged it into Adapter + TTGO.   It works fine.

I need to finish a "hardware mod" on my hand-wired ESP32 board.    I will try it later.
Which also gives the TTGO some time to develop a glitch.

My signal paths are about 100mm from glass to ESP32 chip.
Have you tried running the ESP32 at 80MHz ?

Please describe your glitches.   e.g. do they appear on Triangles, rounded rectangles, ... ?

Ah-ha.   While I was typing the TTGO is misbehaving !!

David.

igor-tvardyi, 2019-08-25:
ESP32:
Display detection: 20% - 0x0, 80% - 0xB509.
In 90% the display is white and can change intensity of white color.
In 9.9% the display can show horizontal (for landscape) random lines, which is not programmed in graphictest_kbv.
In 0.1% the display can show picture from scenario and freeze holding this picture. The picture I saw: "crop", "triangles", "colors" was with artifacts on the bottom.
Usually a picture I can get when I touch/release some pins 3.3V, GND or others.

Arduino Pro Mini 3.3V 8MHz:
Detection: 0x9101 which is wrong. When I force ID to 0xB509 the display shows noting as well as with 0x9101.
For this board (pro mini) I use GFX 1.5.0 (not latest) because binary is too large. 
Still I get warning message which can be translated as  "Not enough memory. Program may be unstable".

But I made it work with a library which is (probably) old modifications of MCUFRIEND library.
[[OPEN-SMART] TFT LCD 3.0inch R61509V.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3538486/OPEN-SMART.TFT.LCD.3.0inch.R61509V.zip)

prenticedavid, 2019-08-26:
If that is your result I would abandon the ESP32 with your display.

No,  I have not finished my "home made ESP32" board yet.
The TTGO works 99.99% but is not 100% reliable.

Quite honestly,   SPI displays are a better choice for ESP32 or ESP8266.   Use Bodmer's TFT_eSPI library.   e.g. 128x160, 240x320, 320x480 Chinese Red SPI displays

David.

closed 2021-08-24 by prenticedavid