prenticedavid/MCUFRIEND_kbv #106: SSD1963 7" display

RaviK147, 2019-08-17:
Hi, 
I am new to tinkering with TFT displays. I tried your your library with ILI9486 with Mega shiled and it worked perfectly. I am now trying to use a 7" display with SSD1963 driver with Mega. The code identifies the driver but tft width and heights show 0X320 in serial and display shows some grey lines. Please help me understand where I messed up. I am attaching my arduino code for your reference. Your help appreciated.

[SSD1963.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3511897/SSD1963.zip)

prenticedavid, 2019-08-17:
I presume that you are using a 40-pin ILI9486 display with a 40-pin Adapter Shield.
A 40-pin SSD1963 display with a 40-pin Adapter Shield should work too.

From extras/mcufriend_how_to.txt

```
17. If you do not have a standard Uno Shield, you can add a SPECIAL to the mcufriend_special.h
    Edit mcufriend_shield.h:  #define USE_SPECIAL
    Edit mcufriend_special.h: e.g. #define USE_MEGA_16BIT_SHIELD
    If your "special" is write-only,  the library can not read the ID.  It always returns 0xD3D3

```
Note that these Adapter shields are **write-only**.   `tft.readID()` will always return `0xD3D3`
So you must always force the ID e.g.

```
   tft.begin(0x9486);   // for ILI9486
   tft.begin(0x1963);   // for SSD1963

```
David.

RaviK147, 2019-08-17:
Thanks David for your prompt reply. Actually, 9486 is a 20 pin version in 8 bit. I followed the same shield by soldering over for SSD1963 . I also tried connecting SSD1963 pins directly to mega by mapping shield connection. With shield, ID returned is 0, but with direct wiring, it shows 1963 as ID, but TFT width and height is always 0x320. Tried forcing ID, but no success so far. I work for automation industry and have some experience in electronics and software, but not with displays. We are trying capture some additional data from our grinder to this display to monitor surface profile. Wondering what is missing, except thinking  that while tinkering with display, somebody may have fried the driver.

prenticedavid, 2019-08-17:
Please post a link to the actual ILI9486 screen you have bought.  e.g. Ebay sale
Please post a link to the actual SSD1963 screen you have bought.  e.g. Ebay sale

SSD1963 displays normally have a 40-pin header.   And use 16-bit parallel interface.
Plug into a 40-pin Adapter Shield.
The Adapter Shield contains level shifters for each signal  e.g. 16 data + 5 control pins.
As well as level shifters for SD card,  Touch controller, Flash chip, ...

David.

RaviK147, 2019-08-17:
Please find attachments.
[image]
[SSD1963RD070ATI-8BIT (4).pdf](https://github.com/prenticedavid/MCUFRIEND_kbv/files/3512079/SSD1963RD070ATI-8BIT.4.pdf)

RaviK147, 2019-08-17:
Hi, forgot to add that page 7 of the 7" display has pinout table for quick reference. Another observation was that 3.3V pin shows over 4.25 volts when connected, and voltage on RST pin was 4.78 volts. I found that odd considering there are level shifters on the shield.

prenticedavid, 2019-08-17:
Your ILI9486 display has level shifters U2, U3

I have never seen an SSD1963 with 8080-8 interface.
Your display **MUST** have level shifters.   Ok,  only 13 required for 8 data lines and 5 control lines. 

If you or your colleagues have attached SSD1963 to 5V GPIO from a MEGA2560 it is _probably_ damaged.

If you connect with 3.3V GPIO everything should work straight  out of the box.    No need for any SPECIALs at all.

LCD_ID_readreg should give a proper report.  e.g.
reg(0xA1) should say `01 57 61 01`

David.

prenticedavid, 2019-08-28:
Please post a schematic of your wiring.   Or a photo.

The graphictest_kbv example should show the correct ID on Serial Terminal.
And show ID on "Adafruit Tests" screen.
On Penguin screens it should scroll vertically.  And do SOFTWARE SCROLL horizontally.

You KNOW how things work from your ILI9486 display.
The SSD1963 should do everything the same.

Make notes on paper.   Describe what is "different"

David.

RaviK147, 2019-08-28:
Hi David, 
Did some more testing, but no progress so far. LCD_ID_readreg gives proper report. But pixel format using LCD_ID_readnew gives 00, 00
here is the the output-
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 08 08	Get Powsr Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0030) 00 00 00 00 00	PTLAR
reg(0x0033) 00 00 00 00 00 00 00	VSCRLDEF
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 01 57 61 01 FF	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B3) 00 00 00 00 00	Frame Memory
reg(0x00B4) 00 00	Frame Mode
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 01 EF	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C1) 00 00 01 00	Display Timing
reg(0x00C5) 00 00	Frame Rate
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00 00	Power Control
reg(0x00D1) 42 42 42 42	VCOM Control
reg(0x00D2) 00 00 00	Power Normal
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

Will appreciate any help with some clue of what might be wrong.
Ravi

prenticedavid, 2019-08-28:
If you have wired like that,  the library should work out of the box.

Ah-ha.  I see the problem.   In MCUFRIEND_kbv.cpp
```

#if defined(SUPPORT_1963) && USING_16BIT_BUS 
    case 0x1963:

```

Please remove the `&& USING_16BIT_BUS`

My apologies.   If you had run the diagnose_TFT_controller sketch you could have pasted the Serial Terminal report.

I would still like to see a photo of your wiring.   The SSD1963 is a 3.3V device.

David.

RaviK147, 2019-08-28:
Thanks for your support. Wiring is exactly the same as the one on UNO shield. The screen I have is SSD1963, 7" 800X480. LCD_ID_readreg gives me correct ID but pixel width and height shows as 0X320. I think that's where the problem is. Below is how we have wired. 
 LCD_RST A4 
 LCD_CS A3
 LCD_RS A2
 LCD_WR A1
 LCD_RD A0

 LCD_D0 8
 LCD_D1 9
 LCD_D2 2 
 LCD_D3 3 
 LCD_D4 4
 LCD_D5 5 
 LCD_D6 6
 LCD_D7 7

RaviK147, 2019-08-28:
Will send you the wiring pic and will also test by removing && USING_16BIT_BUS from cpp file.  Sounds like a plausible solution so far. Thanks a ton for your guidance.

RaviK147, 2019-08-28:
regarding SSD 1963 being a 3.3V device . We are using level shifter to convert logic levels from 5V to 3.3V and we have tested that also.

RaviK147, 2019-09-03:
Hi David, 
Your remedy worked. I would like to thank you for all the help. Much appreciated.

closed 2019-09-03 by RaviK147

IanBUK, 2021-07-14:
Ravik147, how’re you getting on with that ssd1963 display? I’ve seen some on sale and am tempted for a project I’m building. I’m using the Nano 33 BLE.

thanks,

prenticedavid, 2021-07-14:
I have never seen an SSD1963 with 8080-8 interface.

If you can configure the SSD1963 from its default 8080-16 interface,  it should work ok.
Change the CPP file as advised in my Aug 28 2019 reply.

David.

IanBUK, 2021-07-14:
I found a 9” model with 8080-8 on eBay: https://www.ebay.co.uk/itm/302077780265 and their website has this, which lists 8080-8 in the datasheet: https://www.buydisplay.com/7-inch-arduino-touch-screen-shield-ssd1963-library-for-mega-due

a 7” capacitive touch screen does sound nice.

prenticedavid, 2021-07-14:
I would avoid a 9 inch screen.   I prefer 5 inch to 7 inch.   Backlights are greedy on larger screens.

I have not looked at your link.   BuyDisplay generally provide jumpers / solder-bridges to configure their hardware.

David.

IanBUK, 2021-07-14:
Ok. The 7” would be more than sufficient for what I want.

Thanks for your advice.

Ian,