prenticedavid/MCUFRIEND_kbv #107: 2.8 inch button_simple question

stepready, 2019-08-26:
Hi David,
I am new to touchscreens. 
I am using a blue mcufriend 2.8 inch (ili9338 silkscreened on PCB) purchased from Banggood with your button_simple.ino sketch on a UNO R3. 
After using TouchScreen_Calibr_native.ino to get the ID (0x6809) and pins the sketch works regarding display and touch points.
My question is why do the ON and OFF buttons cycle through the justPressed and justReleased states when the button is kept pressed?
I don't know if this is the right site to ask my question but any help would be appreciated.
Thanks
Steve

prenticedavid, 2019-08-26:
Banggood are lying toads too !!

If the ID says 0x6809 you have a Raydium RM68090  **not** an Ilitek ILI9338.

The  justPressed() and justReleased() states should not cycle when the button is kept pressed?
You can always use the isPressed() state. 

Note that the Touch functions are reading the analog position of two resistive panels.   It will never be perfect.    It is wise to take multiple readings and then use the "best" one.    Which is what libraries do for you.    (some Touch libraries are better than others)

Users have great difficulty using  TouchScreen_Calibr_native.ino sketch.
I would appreciate any advice on how to improve it.

Mind you,   many users have difficulty plugging a shield into a Uno or Mega.

David.

stepready, 2019-08-26:
Thanks for your quick reply David.

I will try changing some of the simple_button sketch with the isPressed() state to see if it allows the pressed button to not toggle. Give me a day or 2 to reply.

Off topic - ID=0x6809 - Memories! I think the Motorola 6809 was one of the best 8/16 bit MCUs I ever programmed using assembler (showing my age now).

Thanks
Steve

prenticedavid, 2019-08-26:
Part numbers have "similarities".   Nothing more,  nothing less.

If I remember correctly Motorola M6809 was an excellent 8-bit cpu.
Raydium is a different company.   **R**M6809**0** is a different part number.

David.

stepready, 2019-08-31:
Solved!

If NUMSAMPLES is set to 2 in TouchScreen.ccp this leads to TSPoint p = ts.getPoint(); returning 0 because of small changes in the samples[0] and samples[1] comparison. Hence the pressed button showing pressed and not pressed when actually being pressed. 
I changed NUMSAMPLES to 1 in TouchScreen.cpp and now the buttons accurately shows the state of being pressed or not pressed.

Thanks
Steve

stepready, 2019-08-31:
Sorry David,
I mean that pressure p.z is set to 0 on return.
Steve

prenticedavid, 2019-08-31:
Personally,   I do not think the Adafruit_Touchscreen works very well.   Especially with NUMSAMPLES = 2.
NUMSAMPLES = 3 and above will give you a more reliable result.

NUMSAMPLES = 1 will give you lots of random pixels.

I suggest that you run a Paint program e.g. TouchShield_kbv
Experiment with different values for NUMSAMPLES.

This is why I use a "local" Touch library in the Calibration.
The Calibration values will work fine with every public Touch library e.g Adafruit_Touchscreen.
However the MINPRESSURE threshold will behave differently.   I suggest you use 200.

David.

closed 2021-08-24 by prenticedavid