prenticedavid/MCUFRIEND_kbv #108: Change pinout

skeltr, 2019-08-29:
Hello, 
I would like to use a 2.4 tft lcd and  ws2812 led strip on arduino nano.
The problem is that i need a pwm pin to control the ws2812b and i don't have enough pwm pin.
I would like to connect all the data pins of the screen to digital pins 0 to 7 but i don't know how to do this.
Sorry for my bad english.
Thank you in advance.

prenticedavid, 2019-08-29:
Surely PB2 (OC1B) and PB3 (OC2A) are both PWM pins.

It is quite possible to use different pins for the data bus.
But the original design of the library was to support Mcufriend Shields on regular Arduino headers,.

If you tell me which pins you want to use,  I can show you how to write a SPECIAL. 

David.

skeltr, 2019-08-29:
I would use the SD card so i have to use pins d10 to d13 for the sd card.
I would like connect the data pins of the screen (LCD_D0 to LCD_D7) to the digital pins 0 to 7 of the arduino nano.

prenticedavid, 2019-08-29:
Say what you mean e.g.
D0-D7 for data bus.
D8, D9 for PWM
D10-D13 for SPI 
A0-A4 for control
A5 spare GPIO
A6, A7 spare ANALOG

Yes,  that is easy to code.   And it works faster than regular wiring.
However "sketch upload" is unreliable.   Reading ID, GRAM, ... is unreliable.

But you can upload with LCD_D0, LCD_D1 disconnected.
And force the correct ID instead of using `tft.readID()`

David.

skeltr, 2019-08-29:
It is exactly what I need but i don't know how to do this, can you help me please ?

prenticedavid, 2019-08-29:
Add this block to mcufriend_special.h
#define USE_SPECIAL in mcufriend_shield.h
#define USE_SKELTR in mcufriend_special.h

```
#elif defined(USE_SKELTR) && defined(__AVR_ATmega328P__)      //regular UNO shield on UNO
#define RD_PORT PORTC
#define RD_PIN  0
#define WR_PORT PORTC
#define WR_PIN  1
#define CD_PORT PORTC
#define CD_PIN  2
#define CS_PORT PORTC
#define CS_PIN  3
#define RESET_PORT PORTC
#define RESET_PIN  4

#define DMASK         0xFF
#define write_8(x)    { PORTD = (x); }
#define read_8()      ( (PIND) )
#define setWriteDir() { DDRD |=  DMASK; }
#define setReadDir()  { DDRD &= ~DMASK; }
#define write8(x)     { write_8(x); WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }

#define PIN_LOW(p, b)        (p) &= ~(1<<(b))
#define PIN_HIGH(p, b)       (p) |= (1<<(b))
#define PIN_OUTPUT(p, b)     *(&p-1) |= (1<<(b))

```

Untested.  But it is pretty trivial.

skeltr, 2019-08-29:
I tried it with the graphictest_kbv but it doesn't work.
The screen stay completely white and don't display anything.

prenticedavid, 2019-08-29:
Did you edit the files:
#define USE_SPECIAL in mcufriend_shield.h
#define USE_SKELTR in mcufriend_special.h

And did you force the ID in the sketch e.g.
`
   tft.begin(0x9341);
`

David.,

skeltr, 2019-08-29:
I uncomment the #define USE_SPECIAL in the mcufriend_shield.h and include the #define USE_SKELTR in mcufriend_special.H and also force the ID in the sketch.

skeltr, 2019-08-29:
But it still not work

closed 2019-08-30 by skeltr

erfanjahanshahlo, 2020-07-19:
Hello. I just want to connect the data pin 1 that is connected to the digital pin 9 to the digital pin 0 or rx because I only need one pwm pin and I can not connect the whole d0-d7 to d0 to d7 Ardino because pcb I have it pre-designed and I just want to connect data bus 1 to pin 0 of the Arduino and I have to scratch part of the pcb. If a function was defined so that all data bus pins could be specified before begin, it would be very, very .... best. Thank you in advance.