prenticedavid/MCUFRIEND_kbv #109: NUCLEO-L152RE garbled display

shreyask21, 2019-09-08:
I have connected the 2.4" and 2.8" shield on Nucleo L152RE. The ID's read show wrong values. For 2.8" it shows ID=0x6868 but it should be 0x6809, on 2.4" it shows ID 0x0 but it should be 0x9340. I have tried this modules on UNO, they work fine. Is this issue with LSB pin? What should I do?

Output from LCD_ID_readreg:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-biRead Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 E3 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 02 82 27 04	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 02 02 02 02 02	ILI9481, HX8357-B
reg(0x00C0) 00 15 11 11 11 11 11 11 11	Panel Control
reg(0x00C8) 00 04 04 04 04 04 04 04 04 04 04 04 04	GAMMA
reg(0x00CC) 00 84	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 03 03	NVM Read
reg(0x00D3) 00 00 93 40	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 E3	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 86 10 09 16 08 3C 5A 4A 05 0D 09 1B 1F 0F	GAMMA-P
reg(0x00E1) 00 00 24 2A 00 0E 04 3D 64 4E 04 0B 0A B1 37 0F	GAMMA-N
reg(0x00EF) 00 04 2A 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 81 00 00	Interface Control

[image]

prenticedavid, 2019-09-08:
You definitely have a genuine ILI9340.
`reg(0x00D3) 00 00 93 40 ILI9341, ILI9488
`

I do not have a Nucleo-L152 board.    I would be fairly confident that I have the timing ok.    The L152 is 48MHz.   I have a Nucleo-F072 which should be similar 48MHz.

The RM68090 is another matter.   It will be "slower" than the ILI9340.
I will compare an ILI9320 on my Nucleo-F072 board.

Thanks for your "Issue".
I want to improve the readID() reliability with fast MCUs.    It is 100% important to detect the correct ID. 

David.

prenticedavid, 2019-09-08:
Oops.   I have just checked.   The L152 is actually a M3 and not an M0 

All the same,  I would not expect any extra delays would be necessary.   After all,  it is only 32MHz.

Please go to mcufriend_shield.h and adjust the delays e.g. from

```
#elif defined(STM32L152xE)
#define WRITE_DELAY { } //32MHz M3
#define READ_DELAY  { RD_ACTIVE; }
#define GPIO_INIT()   { RCC->AHBENR |= RCC_AHBENR_GPIOAEN | RCC_AHBENR_GPIOBEN | RCC_AHBENR_GPIOCEN; }
#define PIN_OUTPUT(port, pin) PIN_MODE2((port)->MODER, pin, 0x1)

```
to
```

#elif defined(STM32L152xE)
#define WRITE_DELAY { WR_ACTIVE; } //32MHz M3
#define READ_DELAY  { RD_ACTIVE2; }
#define GPIO_INIT()   { RCC->AHBENR |= RCC_AHBENR_GPIOAEN | RCC_AHBENR_GPIOBEN | RCC_AHBENR_GPIOCEN; }
#define PIN_OUTPUT(port, pin) PIN_MODE2((port)->MODER, pin, 0x1)

```

Please let me know how you get on.   
1.  it must read the ID correctly.
2.  all the Adafruit Tests should be 100% without glitches.
3.  you can easily add extra delays.

David.

shreyask21, 2019-09-09:
I have made changes as you suggested. It still does not work. Btw I have tried both 2.8" (RM68090) and 2.4" (ILI9340) shields. These work perfectly file in arduino UNO but not on Nucleo.
I have uploaded the graphicstest_kbv to Nucleo via arduino ide and ST STM32 core.

The sketch outputs following in serial monitor:

**2.8" RM98090:**
```
Serial took 0ms to start
ID = 0x6808
```
**2.4" ILI9340:**
```
Serial took 0ms to start
ID = 0x0
```

I apologize if I have done anything stupid 😛

prenticedavid, 2019-09-09:
I don't have a Nucleo-L152 but I do have several other Nucleos.
Do you have any other shields or any other Nucleos? 

If ILI9340 and RM68090 work 100% on a Uno,  I would expect them to work exactly the same on the Nucleo.

I presume that you made the LCD_ID_readreg report(s) on the Nucleo.
And have compared the Nucleo output with the Uno output.

This is a 9340 report from Arduino user gregted

```
reg(0x0000) 00 00    ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00 Manufacturer ID
reg(0x0009) 00 00 61 00 00  Status Register
reg(0x000A) 00 08   Get Powsr Mode
reg(0x000C) 00 06   Get Pixel Format
reg(0x0030) 00 00 00 00 00  PTLAR
reg(0x0033) 00 00 00 00 00 00 00    VSCRLDEF
reg(0x0061) 00 00   RDID1 HX8347-G
reg(0x0062) 00 00   RDID2 HX8347-G
reg(0x0063) 00 00   RDID3 HX8347-G
reg(0x0064) 00 00   RDID1 HX8347-A
reg(0x0065) 00 00   RDID2 HX8347-A
reg(0x0066) 00 00   RDID3 HX8347-A
reg(0x0067) 00 00   RDID Himax HX8347-A
reg(0x0070) 00 00   Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00  RD_DDB SSD1963
reg(0x00B0) 00 00   RGB Interface Signal Control
reg(0x00B3) 00 00 1B 1B 1B  Frame Memory
reg(0x00B4) 00 02   Frame Mode
reg(0x00B6) 00 0A 82 27 04  Display Control
reg(0x00B7) 00 07   Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00   ILI9481, HX8357-B
reg(0x00C0) 00 21 00 00 00 00 00 00 00  Panel Control
reg(0x00C1) 00 10 10 10 Display Timing
reg(0x00C5) 00 31   Frame Rate
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00  GAMMA
reg(0x00CC) 00 30   Panel Control
reg(0x00D0) 00 00 00 00 Power Control
reg(0x00D1) 00 00 00 00 VCOM Control
reg(0x00D2) 00 00 00    Power Normal
reg(0x00D3) 00 00 93 40 ILI9341, ILI9488
reg(0x00D4) 00 00 00 00 Novatek
reg(0x00DA) 00 00   RDID1
reg(0x00DB) 00 00   RDID2
reg(0x00DC) 00 00   RDID3
reg(0x00E0) 00 08 0E 12 05 03 09 47 86 2B 0B 04 00 00 00 00 GAMMA-P
reg(0x00E1) 00 08 1A 20 07 0E 05 3A 8A 40 04 18 0F 3F 3F 0F GAMMA-N
reg(0x00EF) 00 00 00 00 00 00   ILI9327
reg(0x00F2) 00 02 02 02 02 02 02 02 02 02 02 02 Adjust Control 2
reg(0x00F6) 00 01 00 00 Interface Control

```

It is quite possible that your 9340 is not made by Iliitek.    There are several differences in the registers.    However you say that your 9340 works fine on the Uno.

The Indian market is often "different" to the rest of the world.   But your shields look Chinese.  And the Nucleo is definitely ST.

Note that current Mcufriend pcb have an empty TSOP-8 footprint and omit the AMS1117 regulator.
This is very BAD for a Uno's wimpy 3.3V pin but should be fine for the Nucleo's 3.3V pin.

David.

shreyask21, 2019-09-09:
I bought these shields from Banggood, not from India as they are overpriced here. I have 2 of each displays, ILI9340 and RM68090. And yes it is a genuine Nucleo from ST, I have not used it for anything before (No chance of blown GPIOs). I have uploaded **same** sketches to both UNO and nucleo. On UNO all 4 displays work fine, graphics test, readid all. But on nucleo they do not work, readid gives false values, garbage display or no display.

shreyask21, 2019-09-17:
Yes checked the L152 Datasheet. I have done that workaround pinMode before the readID. The ID is still not read correctly, nor the display works after forcing the correct ID.

I have brought Nucleo L476, the diaplay test works out of the box.

shreyask21, 2019-09-17:
Geez sorry. Accidentally clicked close and comment. My bad.

Is there anything that I can do to help or test?

Thanks for your support,
Shreyas.

prenticedavid, 2019-09-17:
I am sure that the L152 board is 100%.
You can try running regular Arduino sketches or MBED programs.

Since the LCD_ID_readreg.ino sketch works ok,  I can assume that all the Port pins are in working order.
Obviously I am doing something wrong.   e.g. not configuring PB3 correctly.   Or not enabling a peripheral clock somewhere.

I don't own an L152 board.   You are the first and only person to raise this issue.
I suggest that you experiment with the L476 board.     It runs very efficiently at 80MHz.    If you want to minimise power,  you can run at 4MHz and Sleep as often as possible.

Perhaps you or I will think of a solution for the L152.
If I get really frustrated I will buy an L152 for myself.     But I will wait for a while.

Did pinMode(3, OUTPUT) make any difference ?

I am intrigued.    Have you lived in the UK or US ?

David.

shreyask21, 2019-09-17:
Yes I have tried `pinMode(3, OUTPUT)` and it did not make any difference. I think I will stick to the L476 for now. I am impressed by the speed of L476. It is as not fast as an ESP32, but pretty low power. It also has 1M flash, which is handy for storing large bitmaps and other stuff instead of 512K on L152.

I might try fixing the L152, but its a rabbit hole. But for now I am not bothered. 
I will try later.

No I have not lived in UK or US.

Shreyas.

prenticedavid, 2019-09-17:
> Geez sorry.  ... . My bad.

did not sound like Indian English (to me)

BMP images are very wasteful.
You can store several JPEG images in the L476 1MB Flash memory.
Much easier than using the microSD card.

David.

shreyask21, 2019-09-17:
> did not sound like Indian English (to me)

Hmm, interesting.  I ponder why.

> BMP images are very wasteful.
> You can store several JPEG images in the L476 1MB Flash memory.

Yeah true. I will store jpegs instead. Not to mention slower speed when loading from SD.

Thank you for your fantastic library.
Shreyas.

prenticedavid, 2019-09-18:
I have put a very simple MBED program https://os.mbed.com/users/davidprentice/code/Nucleo_dir_L152/

It should obviously run on your L476 board
Please try it on the L152 board

It should show the ID on a Serial Terminal (9600 baud)

David.

prenticedavid, 2019-09-25:
Please could you try the MBED program.

I would like to resolve the problem.

David.

shreyask21, 2019-09-26:
Apologies for the delay.

[image]
Umm, the program does not compile. I have imported your program in Mbed compiler. I am inexperienced in Mbed. Am I doing something wrong?

prenticedavid, 2019-09-26:
Ah-ha.   0x6808 is not a valid ID.   I presume that you have a Raydium RM68090 with ID = 0x6809.
What ID did you get for the Ilitek ILI9340 ?

It looks like a timing problem.   The ILI9340 should work much faster than the RM68090.

The MBED program  should work the same on L152 and on L476 boards. 

David.

shreyask21, 2019-09-27:
Yes I have the RM68090.

I tried ILI9340 on L152, and it works! 😄

It prints ID correctly and display works too. I think you are correct. It may be a timing problem.

[image]

shreyask21, 2019-09-27:
The RM68090 works on L476. 

[image]

prenticedavid, 2019-09-27:
Ok.   Everything works fine on L476.  Arduino and MBED.
L152 fails on Arduino for RM68090 and ILI9341
L152 reads wrong ID on MBED for RM68090

Please force ID = 0x6809 on MBED.   Does MBED display with `tft.begin(0x6809);`

David.

shreyask21, 2019-09-27:
> Please force ID = 0x6809 on MBED. Does MBED display with tft.begin(0x6809);

Yep, that did it. Display is working. Forced ID in setup() -> `tft.begin(0x6809);`.

closed 2021-08-24 by prenticedavid