prenticedavid/MCUFRIEND_kbv #111: TouchScreen_Calibr_native not working

Zeref010, 2019-09-18:
Hello !

I recently bought a 2.8" TFT LCD Shield from Wish  similar to this one 
http://www.mcufriend.com/index.php?main_page=product_info&cPath=4&products_id=18
and am determined to make it work with my Arduino Mega 2560.

So I know that the ID of the chip is 0x4747, I followed the procedure and uncommented the #define SUPPORT_8347D  in the cpp file.

Thank's to this, I successfully ran graphictest_kbv and differents codes to print informations on the screen without any problems, but now that I want to test the touch screen, I end up with the same white screen from before the #define.

I ran LCD_ID_read_reg and ended up with this : 

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 47 47	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 3F 3F 3F 3F 3F	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 01 01	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 80 80	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03	GAMMA-P
reg(0x00E1) 49 49 49 49 49 49 49 49 49 49 49 49 49 49 49 49	GAMMA-N
reg(0x00EF) 49 49 49 49 49 49	ILI9327
reg(0x00F2) 10 10 10 10 10 10 10 10 10 10 10 10	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

1) Can you tell me a way to know for sure if the touchscreen isn't working at all ? 

2) If it's not broken, what would be the issue according to you ?

I'm using arduino IDE app for Win10 1.8.9 and  (normal) arduino IDE 1.8.10 in case the first one was the issue. 

Thank's for your time,

prenticedavid, 2019-09-18:
You have a Himax HX8347-D controller.    
You have defined SUPPORT_8347D.

So all of the examples should work 100%.
Including TouchScreen_Calibr_native.ino

Please copy-paste the output from the Serial Terminal.

I use Win10-64 with IDE v1.8.9 and this works fine with v2.9.8 Release (as installed by IDE Library Manager).

I do not advise using the v2.9.9-Beta.    There are a lot of changes but they are mostly for STM32.

David.

Zeref010, 2019-09-18:
Thank's for your quick response ! 
[image]
I had nothing on the monitor .... But then i remembered I modified the code and didn't put it back afterward, now i'm getting the "broken touchscreen" in white with red background and the monitor printed me this :

19:30:07.797 -> TouchScreen.h GFX Calibration
19:30:07.797 -> Making all control and bus pins INPUT_PULLUP
19:30:07.865 -> Typical 30k Analog pullup with corresponding pin
19:30:07.900 -> would read low when digital is written LOW
19:30:07.968 -> e.g. reads ~25 for 300R X direction
19:30:08.002 -> e.g. reads ~30 for 500R Y direction
19:30:08.035 -> 
19:30:08.035 -> BROKEN TOUCHSCREEN
19:30:08.272 -> ID = 0x4747

Thank's for the advice, I usually get the last stable version of  a software to be safe :)

prenticedavid, 2019-09-18:
The photo in your link shows a Touch Panel.  i.e. 4 gold contacts at bottom left of
screen.

Your report suggests that there is no Touch Panel.   Or it is broken in both directions.    You should see a cracked glass easily in your photo.

Use a DMM if you don't believe me.

David.

Zeref010, 2019-09-18:
Ok so I guess if there is no cracked glass the touch panel isn't a touch panel but just a screen, not that this surprises me,  from something that cheap I should have wondered.

Thank's anyway, at least I can use it as a 2.8" screen, so no big deal. :)

closed 2019-09-18 by Zeref010

prenticedavid, 2019-09-18:
Please can you confirm whether the screen on your desk matches the photos in your link.
i.e. Touch Panel
i.e. Blue pcb
i.e. same components,  same copper traces on the pcb.

David.

Zeref010, 2019-09-18:
Well it does look like it, except some details you can see in the picture, same number of pin, same copper traces, same blue pcb.

For the components I can't  tell sadly..

[image]

prenticedavid, 2019-09-18:
Your screen has a Touch Panel.    It even has five printed icons.
I would expect a 500R resistance between LCD_D0 and LCD_RS,  and 500R resistace between LCD_D1 and LCD_WR.

Ok,  the Mcufriend site does not show the pcb.
Does your pcb have two 74HC245 and one AMS1117 i.c. ?

David.

Zeref010, 2019-09-18:
[image]
So that's the back, the 2 dark components are the same and have this written on it :

SUM74HC245T
D1412A4PJ1

prenticedavid, 2019-09-18:
The older pcb had an AMS1117 LDO regulator.
Current Mcufriend pcbs omit the regulator.

This is probably ok for 2.4 inch displays.
But backlight draws too much current from an Uno or Mega 3.3V pin on 2.8 inch, 3.5 inch.

David.

Zeref010, 2019-09-18:
Oh I see, so design issues... Not really a good sign, and with an external power input, would it work ? It's like witchcraft at this point, I could end up with anything..

Thank's for your time, I'll give it an update if I find anything.