prenticedavid/MCUFRIEND_kbv #114: maple (f103re) compile error

clear-sky, 2019-10-13:
please support stm32f103ret6 (maple)
i edited mcufriend_shield.h but got tons of errors!

prenticedavid, 2019-10-13:
mcufriend_shield.h supports the ARDUINO_MAPLE_REV3 which is a Uno format board with standard Arduino headers.

If you have a commercially available board or adapter with Arduino headers,   I will support it.

If you only have hand-made connections it needs to be a SPECIAL.

I suggest that you look at some of the BLUEPILL specials.
If you post your wiring,  I can help you write the SPECIAL.
Please test the #defines in the LCD_ID_readreg sketch.    Post the #defines when you have verified the wiring.

David.

clear-sky, 2019-10-14:
hi dear david
thanks for your answer

first of all my board is handmade 
this is my config
d0-d7 > pin c0 to c7
rd > pin c8
wr > pin c9
cs > pin c10
rs > pin c11
rst > pin c12

this is my lcd
https://www.raspberrypi.org/forums/viewtopic.php?t=180064
i am 99% sure it has ili9488 chip
in this module rd is connected to vcc but i modified and rd is now available

and i use stm32duino 
https://github.com/stm32duino/Arduino_Core_STM32
this is my programmer
http://electrobist.com/product/st-link-v2-st-link-v2-stlink-v2-programmer/
thanks for your notice

prenticedavid, 2019-10-14:
Ah-ha.   The RPi link shows the pinout of the display pcb.    

So you can make the appropriate defines in LCD_ID_readreg
And verify your wiring.

I will write the SPECIAL for you when you copy-paste the defines from the sketch.
And copy-paste the output from the Serial Terminal.   reg(0x00D3) line is sufficient.

David.

clear-sky, 2019-10-15:
hi dear david and very thanks for your notice
this is my defines
//-- Arduino UNO or Mega 2560 Plugged as shield
#define LCD_RST PC12 //A1
#define LCD_CS PC10
#define LCD_RS PC11
#define LCD_WR PC9
#define LCD_RD PC8

#define LCD_D0 PC0
#define LCD_D1 PC1
#define LCD_D2 PC2 //10
#define LCD_D3 PC3 //11
#define LCD_D4 PC4
#define LCD_D5 PC5 //13
#define LCD_D6 PC6
#define LCD_D7 PC7
 
and this is my serial monitor output :

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66	Manufacturer ID
reg(0x0009) 08 08 08 08 08	Status Register
reg(0x000A) 08 08	Get Powsr Mode
reg(0x000C) 08 5C	Get Pixel Format
reg(0x0030) 00 00 00 00 00	PTLAR
reg(0x0033) 00 00 00 00 00 00 00	VSCRLDEF
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B3) 00 00 11 11 11	Frame Memory
reg(0x00B4) 00 02	Frame Mode
reg(0x00B6) 00 02 02 33 33	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 08 0E 0E 0E 0E 0E	ILI9481, HX8357-B
reg(0x00C0) 00 06 06 06 06 06 06 06 06	Panel Control
reg(0x00C1) 00 B3 B3 B3	Display Timing
reg(0x00C5) 00 43	Frame Rate
reg(0x00C8) 08 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E	GAMMA
reg(0x00CC) 08 0B	Panel Control
reg(0x00D0) 00 00 00 00	Power Control
reg(0x00D1) 00 00 00 00	VCOM Control
reg(0x00D2) 00 00 00	Power Normal
reg(0x00D3) 00 00 94 80	ILI9341, ILI9488
reg(0x00D4) 00 66 00 00	Novatek
reg(0x00DA) 08 08	RDID1
reg(0x00DB) 08 08	RDID2
reg(0x00DC) 08 9F	RDID3
reg(0x00E0) 00 00 07 04 05 13 01 36 A2 46 01 10 05 12 16 07	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 08 4C 4C 4C 4C 4C	ILI9327
reg(0x00F2) 00 00 00 00 20 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2019-10-15:
Your readreg does not look quite correct.   e.g.

```
reg(0x0004) 00 54 80 66 Manufacturer ID
reg(0x00D3) 00 00 94 80 ILI9341, ILI9488
reg(0x00DA) 08 08 RDID1
reg(0x00DB) 08 08 RDID2
reg(0x00DC) 08 9F RDID3

```
I would expect different values in 0xD3, DA, DB, DC e.g.

```
reg(0x0004) 00 54 80 66    Manufacturer ID
reg(0x00D3) 00 00 94 88 RDID4
reg(0x00DA) 00 54   RDID1
reg(0x00DB) 00 80   RDID2
reg(0x00DC) 00 66   RDID3

```

Please double check your wiring.    It is "almost" ok.

I will post a SPECIAL later this evening.

David.

prenticedavid, 2019-10-15:
I am not too sure which Core or board you are using.
The official STMicroelectronics Core does not have a specific RET6 target
The weird Roger Clark Core does have ARDUINO_MAPLE_RET6

Untested.   I can build but not run.
If your readreg is inconsistent,   the SPECIAL is unlikely to work.

Add  #define USE_CLEAR_SKY and this block to mcufriend_special.h
Add USE_SPECIAL to mcufriend_shield.h
```
//############################### CLEAR_SKY MAPLEMINI or MAPLE_RET6 ###########################################
#elif defined(USE_CLEAR_SKY) && (defined(ARDUINO_MAPLEMINI_F103CB) || defined(ARDUINO_MAPLE_MINI) || defined(ARDUINO_MAPLE_RET6))
#warning USE_CLEAR_SKY on MAPLEMINI

//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS  |CS  |RST | |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PC7|PC6|PC5|PC4|PC3|PC2|PC1|PC0| |PC8|PC9|PC11|PC10|PC12| 

#if defined(ARDUINO_ARCH_STM32)   //regular CMSIS libraries
#define REGS(x) x
#define GPIO_INIT()   { RCC->APB2ENR |= RCC_APB2ENR_IOPAEN | RCC_APB2ENR_IOPBEN | RCC_APB2ENR_IOPCEN | RCC_APB2ENR_IOPDEN | RCC_APB2ENR_AFIOEN; \
        AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_1;}
#else                                                                  //weird Maple libraries
#define REGS(x) regs->x
#endif

#define WRITE_DELAY { }
#define READ_DELAY  { RD_ACTIVE; }
#define GROUP_MODE(port, reg, mask, val)  {port->REGS(reg) = (port->REGS(reg) & ~(mask)) | ((mask)&(val)); }
#define GP_OUT(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x33333333)
#define GP_INP(port, reg, mask)           GROUP_MODE(port, reg, mask, 0x44444444)
#define PIN_OUTPUT(port, pin) {\
        if (pin < 8) {GP_OUT(port, CRL, 0xF<<((pin)<<2));} \
        else {GP_OUT(port, CRH, 0xF<<((pin&7)<<2));} \
    }
#define PIN_INPUT(port, pin) { \
        if (pin < 8) { GP_INP(port, CRL, 0xF<<((pin)<<2)); } \
        else { GP_INP(port, CRH, 0xF<<((pin&7)<<2)); } \
    }
#define PIN_HIGH(port, pin)   (port)-> REGS(BSRR) = (1<<(pin))
#define PIN_LOW(port, pin)    (port)-> REGS(BSRR) = (1<<((pin)+16))

#define RD_PORT GPIOC
#define RD_PIN  8
#define WR_PORT GPIOC
#define WR_PIN  9
#define CD_PORT GPIOC
#define CD_PIN  11
#define CS_PORT GPIOC
#define CS_PIN  10
#define RESET_PORT GPIOC
#define RESET_PIN  12

// configure macros for the data pins
#define write_8(d)    { GPIOC->REGS(BSRR) = 0x00FF << 16; GPIOC->REGS(BSRR) = (d) & 0xFF; }
#define read_8()      (GPIOC->REGS(IDR) & 0xFF)
//                                         PA7 ..PA0
#define setWriteDir() {GP_OUT(GPIOC, CRL, 0xFFFFFFFF); }
#define setReadDir()  {GP_INP(GPIOC, CRL, 0xFFFFFFFF); }

#define write8(x)     { write_8(x); WRITE_DELAY; WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; READ_DELAY; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }

```

prenticedavid, 2019-10-16:
Bump.   I don't think that you would have been notified when I edit an existing message.

clear-sky, 2019-10-18:
hi dear david -thanks for your notice
you are very kind
i rechecked wiring several times but it was true 
i am using stm32cores ver 1.7.0
your first code give me tons of errors but second with no errors
i run glue demo 480x320 but nothing happens on screen
after that i used this commands instead of "myGLCD.InitLCD();"
 myGLCD.reset();
 myGLCD.begin(0x9488);
 myGLCD.setRotation(3);
after that i can see a vertical column in the right of my screen it goes darker then lighter (repeats)
thats all
thanks for your notice again

prenticedavid, 2019-10-18:
You are using a good Core i.e. v1.7.0
I do not have a stm32f103ret6.   But I would be 99.99% confident that Frederic from ST has written an accurate "variant"
Please post a link to the actual RET6 board that you have bought.

Run the readreg sketch again.   Report back.   Especially if you get the values that I "expected".

Never start with the Glue Demo.
Run the graphictest_kbv or diagnose_TFT_support examples first.

David.

clear-sky, 2019-10-20:
hi dear david thank you for the time you spend for me
my f103re board link
https://eshop.eca.ir/%D9%85%D8%AD%D8%B5%D9%88%D9%84%D8%A7%D8%AA-eca/1325-stm32f103ret6-header-board.html
and the schematic
http://server4.eca.ir/eshop/000/header-STM32-ZET/documents/ECA_Schematic_Rev%20B.pdf

now i am sure my lcd is damaged
will try another lcd with ili9486(this nude lcd is much better than chinese lcd mudules)
lcd link
http://www.aitendo.com/product/15953
datasheet
http://aitendo3.sakura.ne.jp/aitendo_data/product_img/lcd/tft2/Z350IT002/Z350IT002_aitendo.pdf
i will make my own handmade module with 8 bit interface
my question is your driver has any problem with 8 bit wiring?
thank you

prenticedavid, 2019-10-20:
Your first link does not point me to a board.
The schematic looks fine
The bare TFT module will be fine.   Just need custom pcb to receive the ribbon.
A hand-made module is fine with your 3.3V STM32.   You do not need buffer chips for the GPIO.   You do need an external transistor if you want to control the backlight.

Library code should be fine.    It can do 8080-8 or 8080-16 parallel interface.

Please note that wiring up bare modules to a fast MCU board requires good pcb layout and good local capacitors.    Breadboard or Dupont wires are unwise.

Remember that I do not have your hardware.    I try to use ready-made Shields with headers mounted on either commercial Arduino boards or hand soldered Protoboard.  i.e. no floating wires.

David.

clear-sky, 2019-11-23:
hi dear david 
i made my shield with ili9486 in 8 bit mode
this is LCD_ID_readnew output
⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66	Manufacturer ID
reg(0x0009) 08 08 08 08 08	Status Register
reg(0x000A) 08 08	Get Powsr Mode
reg(0x000C) 08 5C	Get Pixel Format
reg(0x0030) 00 00 00 00 00	PTLAR
reg(0x0033) 00 00 00 00 00 00 00	VSCRLDEF
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B3) 00 00 00 00 00	Frame Memory
reg(0x00B4) 00 00	Frame Mode
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 08 08 08 08 08 08	ILI9481, HX8357-B
reg(0x00C0) 00 06 06 06 06 06 06 06 06	Panel Control
reg(0x00C1) 00 B3 B3 B3	Display Timing
reg(0x00C5) 00 43	Frame Rate
reg(0x00C8) 08 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E 0E	GAMMA
reg(0x00CC) 08 0C	Panel Control
reg(0x00D0) 00 00 00 00	Power Control
reg(0x00D1) 00 00 00 00	VCOM Control
reg(0x00D2) 00 00 00	Power Normal
reg(0x00D3) 00 00 94 86	ILI9341, ILI9488
reg(0x00D4) 00 66 00 00	Novatek
reg(0x00DA) 08 08	RDID1
reg(0x00DB) 08 88	RDID2
reg(0x00DC) 08 6E	RDID3
reg(0x00E0) 00 07 04 35 05 10 00 67 47 51 07 05 00 07 35 05	GAMMA-P
reg(0x00E1) 00 02 34 25 02 13 05 11 52 52 06 10 05 16 24 02	GAMMA-N
reg(0x00EF) 08 88 08 18 68 48	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 01	Interface Control

this is graphictest_kbv output:
Serial took 0ms to start
ID = 0x9486

everything looks good but nothing happens in screen!

prenticedavid, 2019-11-23:
Your readreg report is different to what I would expect.
It is even different to your report from Oct 15.

I suspect that you have poor or intermittent wiring.
If and when you resolve the wiring,  I would expect everything to work.

David.

clear-sky, 2019-11-24:
hi dear david, thanks for your notice
>>Your readreg report is different to what I would expect
which differece?

>>It is even different to your report from Oct 15.
that was ili9488 but new one is ili9486 in 8 bit mode
wiring has no problem
i am thinking about setting 8 bit mode for 9486 but could not find where can i find it
i have set r8 instead of r16 base on end of page 7 on this datasheet
http://aitendo3.sakura.ne.jp/aitendo_data/product_img/lcd/tft2/Z350IT002/Z350IT002_aitendo.pdf

prenticedavid, 2019-11-24:
I can't understand Japanese.   I suppose I could put it through Google Translate.
You normally select 8080-8 vs 8080-16 by moving an 0R SMD resistor mounted on the ribbon.

Personally,  I would select 8080-8 interface.   It uses less GPIO pins.
If you want to use 8080-16 interface,  select an appropriate SPECIAL.   I have used 8080-16 on a BluePill but it uses up most of your GPIO pins.

The ILI9846 always uses 8-bit commands regardless of the width of data bus.
The "comment" in the readreg report refers to the way an ILI9320 receives commands (and arguments)  versus a modern chip like ILI9486.

David.

clear-sky, 2019-12-02:
problem solved
dont know why but my maple had problems
replaced the board and new one works fine
big thanks for your notice and your nice code
you are great

closed 2019-12-02 by clear-sky