prenticedavid/MCUFRIEND_kbv #115: Display White Screen

Sagarlambat, 2019-10-17:
Diagnose whether this controller is supported
There are FAQs in extras/mcufriend_how_to.txt

tft.readID() finds: ID = 0x6090

MCUFRIEND_kbv version: 2.9.9

This ID is not supported
look up ID in extras/mcufriend_how_to.txt
you may need to edit MCUFRIEND_kbv.cpp
to enable support for this ID
e.g. #define SUPPORT_8347D

New controllers appear on Ebay often
If your ID is not supported
run LCD_ID_readreg.ino from examples/
Copy-Paste the output from the Serial Terminal
to a message in Displays topic on Arduino Forum
or to Issues on GitHub

Note that OPEN-SMART boards have diff pinout
Edit the pin defines in LCD_ID_readreg to match
Edit mcufiend_shield.h for USE_SPECIAL
Edit mcufiend_special.h for USE_OPENSMART_SHIELD_PINOUT

prenticedavid, 2019-10-17:
I had tea, toast and marmalade for breakfast.

Sagarlambat, 2019-10-17:
While interfacing tft display with arduino i got this output on my serial monitor.
It says that the device id 0x6090 isn't supported.

Diagnose whether this controller is supported
There are FAQs in extras/mcufriend_how_to.txt

tft.readID() finds: ID = 0x6090

MCUFRIEND_kbv version: 2.9.9

This ID is not supported
look up ID in extras/mcufriend_how_to.txt
you may need to edit MCUFRIEND_kbv.cpp
to enable support for this ID
e.g. #define SUPPORT_8347D

New controllers appear on Ebay often
If your ID is not supported
run LCD_ID_readreg.ino from examples/
Copy-Paste the output from the Serial Terminal
to a message in Displays topic on Arduino Forum
or to Issues on GitHub

Note that OPEN-SMART boards have diff pinout
Edit the pin defines in LCD_ID_readreg to match
Edit mcufiend_shield.h for USE_SPECIAL
Edit mcufiend_special.h for USE_OPENSMART_SHIELD_PINOUT

Kindly help!!

prenticedavid, 2019-10-17:
Go on.   Do you want to know what I had for lunch?

The diagnostic sketch suggests:

> If your ID is not supported
> run LCD_ID_readreg.ino from examples/
> Copy-Paste the output from the Serial Terminal
> to a message in Displays topic on Arduino Forum
> or to Issues on GitHub
> 

My usual advice on Arduino Forum is:

> Either post a link to the actual shield that you have bought.  e.g. Ebay sale
> Or post a photo of the pcb side of the display.
> 

> Post a photo of the shield plugged into the Arduino board.

If you are a beginner with Arduino,  it is probably better to ask questions on the Arduino Displays Forum
If you are experienced with programming,  ask questions on GitHub.

David.

Sagarlambat, 2019-10-18:
I runned LCD_ID_readreg.ino and got the following output.

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 60 90	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 02 00 02 00 02 00 02 00 02	Panel Control
reg(0x00C8) 00 FF 00 FF 00 FF 00 FF 00 FF 00 FF 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2019-10-18:
That was easy.   Thanks for your readreg report.

I do not recognise this controller.   
I do not know if you have a Mcufriend shield.   
I do not even know what colour the pcb is.  
Or if there is anything printed on the pcb. 
Or whether you know how to plug a shield into an Arduino.

David.

Sagarlambat, 2019-10-18:
Here it is....
2.8 inch tftlcd for arduino(ft1509)
(www.mcufriend.com)

[image]
[image]
[image]
[image]
[image]
[image]

Please Help!!

prenticedavid, 2019-10-18:
Your photo looks like a 240x400 screen
Most 2.4 inch screens are 240x320
Please measure the actual dimensions of the white area on your screen.   A 240x320 will be 44mm x 58mm.   240x400 will be a different shape.

It also looks like an ILI9320 (240x320) or ILI9326 (240x400) style interface i.e. 16-bit command with 16-bit data.

I can only suggest that you edit setup() in graphictest_kbv.ino
And try forcing different IDs e.g. `tft.begin(0x9320);` or `tft.begin(0x9326);`

Note that you must edit MCUFRIEND_kbv.cpp to enable support for ILI9326 i.e. SUPPORT_9326_5420
extras/mcufriend_how_to.txt shows supported controllers.

David.

arslan437, 2019-12-06:
I am also having the same problem here is the report 

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 99 99 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Powsr Mode
reg(0x000C) 00 66	Get Pixel Format
reg(0x0030) 00 00 00 01 3F	PTLAR
reg(0x0033) 00 00 00 01 40 00 00	VSCRLDEF
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B3) 00 00 00 00 00	Frame Memory
reg(0x00B4) 00 00	Frame Mode
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C1) 00 00 00 00	Display Timing
reg(0x00C5) 00 00	Frame Rate
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00 00	Power Control
reg(0x00D1) 00 00 00 00	VCOM Control
reg(0x00D2) 00 00 00	Power Normal
reg(0x00D3) 00 00 32 29	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek
reg(0x00DA) 00 99	RDID1
reg(0x00DB) 00 99	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control
```

prenticedavid, 2019-12-06:
It looks like a 240x320 ILI9329-style controller.   But I have no idea who made it or its part number.

I suggest that you force the ID e.g. `tft.begin(0x9329);`  and report back.
If you get some response,  either take paper notes for the graphictest_kbv example and type in your message.
Or post a video or photos of the Penguin screens..

David.

amitsingh78, 2020-01-07:
It doesn't work only 1 row works rest is colored lines. I already asked in the Arduino forum

closed 2021-08-24 by prenticedavid