prenticedavid/MCUFRIEND_kbv #116: Teensy 3.6 with 3.5 Touchscreen works only if using blue board display, Red Board works on Due, but not Teensy 3.6

badonj001, 2019-10-20:
I have several 3.5 inch MCUFRIEND touch Screens with the Red Board and are labeled "www.mcufriend.com 3.5" TFT LCD for arduino uno". Using this library the Red board touchscreens work on the Due and the Mega 2560, but on the Teensy 3.6 they do not work. 

This is the Red Board: https://www.amazon.com/gp/product/B07F2XM5HW/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
[image]

Using the Blue Board, this library works with Due, Mega 2560, and the Teensy 3.6 (Thank you for this change that allows me to use the Blue Board with the Teensy 3.6, Lightning speed graphics), 

This is on of the branded BLue Boards I am using: https://www.amazon.com/gp/product/B075FP83V5/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
[image]

On Due and Mega the get ID on the Red Board Comes back as Unrecognized and uses TFT.begin as set to use 0x9486 by default.  This does not work on Teensy 3.6.,

On Due, Mega, and Teensy 3.6, the get ID on the Blue Board Comes back as found "0x9486" and uses TFT.begin as set to use 0x9486. This works without issue.

Please note this is the pin configuration for LANDSCAPE on the Blue Board with the Teensy 3.6: "const int XP=9,XM=A3,YP=A2,YM=8;"

I have a case of the Red boards and would love to use them with the Teensy 3.6" if you can reconfigure the library to work with them.  I would be willing to send you a Red Board and a Blue Board for free to keep if this would be helpful with debugging. This is the least I can do for all of your help you are providing to us. Very, Very grateful for being at least able to use with the BLue Board.

Additional information:
On the Red Board, U2 and U3 are SUM74HC245T, (Tri-state output signal Octal Bidirectional Transceiver IC)

On the Blue Board, U2 and U3 are LVC245A, (Tri-state output signal OCTAL BUS TRANSCEIVER
WITH 3-STATE OUTPUTS)

As to reiterate:
The Red board using the SUM74HC245T on U2 and U3 does not work on the Teensy 3.6 only, but works on the Due and Mega.

The Blue board using the LVC245A on U2 and U3 works on the Due, Mega and Teensy 3.6.

This is the original library code provided by MCUFRIEND when I Purchased the my first www.mcufriend.com Red Board, and my first KUMAN Blue Board. To my knowledge they are the same, but the transceiver chips are different and the Red Board apparently does not pass the "the 0xD3D3 test" in the MCUFRIEND_kbv.cpp, from what I understand thus far with this type of coding. I am still learning.

" uint16_t identifier = tft.readID();
  Serial.println(F("TFT LCD test"));
  tft.reset();

  tft.print(F("ID = 0x"));
  tft.println(identifier, HEX);

  
  if(identifier == 0x9325) {
    Serial.println(F("Found ILI9325 LCD driver"));
  } else if(identifier == 0x9328) {
    Serial.println(F("Found ILI9328 LCD driver"));
  } else if(identifier == 0x4535) {
    Serial.println(F("Found LGDP4535 LCD driver"));
  }else if(identifier == 0x7575) {
    Serial.println(F("Found HX8347G LCD driver"));
  } else if(identifier == 0x9341) {
    Serial.println(F("Found ILI9341 LCD driver"));
  }else if(identifier == 0x7783) {
    Serial.println(F("Found ST7781 LCD driver"));
  }else if(identifier == 0x8230) {
    Serial.println(F("Found UC8230 LCD driver"));  
  }
  else if(identifier == 0x8357) {
    Serial.println(F("Found HX8357D LCD driver"));
  } else if(identifier==0x0101)
  {     
      identifier=0x9341;
       Serial.println(F("Found 0x9341 LCD driver"));
  }else if(identifier==0x9481)
  {     
       Serial.println(F("Found 0x9481 LCD driver"));
  }
  else if(identifier==0x9486)
  {     
       Serial.println(F("Found 0x9486 LCD driver"));
  }
  else {
    Serial.print(F("Unknown LCD driver chip: "));
    Serial.println(identifier, HEX);
    Serial.println(F("If using the Adafruit 2.8\" TFT Arduino shield, the line:"));
    Serial.println(F("  #define USE_ADAFRUIT_SHIELD_PINOUT"));
    Serial.println(F("should appear in the library header (Adafruit_TFT.h)."));
    Serial.println(F("If using the breakout board, it should NOT be #defined!"));
    Serial.println(F("Also if using the breakout, double-check that all wiring"));
    Serial.println(F("matches the tutorial."));
    identifier=0x9486;
    
  }

  tft.begin(identifier);

  Serial.print("TFT size is "); Serial.print(tft.width()); Serial.print("x"); Serial.println(tft.height());"

Let me know if you would like me to send you the boards, and again, I really appreciate the update that at least allows me to use my 3.5" Blue Board screens with and without touch.

Best Regards,
Badonj001

prenticedavid, 2019-10-20:
First off.   Check your current library version e.g. via IDE Library Manager.

Update to current 2.9.8 Release via Library Manager.
Update Adafruit_GFX

If you have installed from ZIP or some old Beta:  Leave IDE.   Delete any obsolete library directory.    Restart IDE.    Install via Manager.

Don't use the current 2.9.9 Beta unless I specifically ask you.

It is important to identify your Red Shields by running on Uno or Mega.
Run LCD_ID_readreg from examples.   Copy-Paste from Serial Terminal.

The Blue Shield backlight is powered by the 5V pin via the on-shield AMS1117.
The Red Shield backlight is powered by the 3.3V pin from Uno or Teensy.

The Red Shield design is **WRONG**.
Ok if your Teensy has a 200mA 3.3V pin.   The Uno or Mega is only permitted 50mA from its 3.3V pin.
If you have a DMM:   Connect GND,  3V3 measure the current.   Does backlight light?

The Blue Shield is much better.   It has a big regulator.   It uses LVC245 instead of HC245.
HC245 is fine for your Teensy.    Bad for a Uno.

It should be easy to resolve your Red Shield problem remotely.   (I have boxes full of TFTs)
Let's see if it is an unusual controller.

David.

badonj001, 2019-10-22:
David,

Thank you for your quick response, I really appreciate it! Before I
messaged you I was thinking maybe I could just replace the chips on the red
shield with the chips on the blue shield, but I also noticed the regulator
on the blue shield that the Red Shield does not have. I will follow your
instructions hopefully tonight and get back with you. Ultimately, if I have
to buy Blue Shield displays to use with my Teensy 3.6 projects for the
speed, flash size, and eeprom space that would not be a problem. Over a
year ago I purchased 1x Red Shield Display and it worked with the mega.
They were extremely  cheap ($7.50 vs. $17.50) and I purchased a box of
them  shipped from china. I do give away projects for others from time to
time and it is nice to provide them with a large 3.5" touch display. I
noticed the Missing voltage regulator a year ago on the Red Shield and
wondered why it didn't have one and the Blue Shield did. My first shield
was the Blue KUMAN.

So first, I will take some time and follow your instructions 100% and get
back with you and provide you with my results. Again, thank you so much for
everything.

I am building an RFID control system that is actually secure in comparison
to the really nice looking ones you can buy. The ones you buy can be
hacked, it you call the simple process of ripping off the control panel and
jumpering the voltage wire over to the door open signal wire hacking. The
connectors on the controllers board is labeled and you have zero guessing
to activate the door lock and gain access. I purchased several them and
returned them because every last one I received was not secure at all.
Ultimately, Once I am done with my design, I plan to post it open source
for all to download and use freely. I will be providing links to components
needed and the Blue Shield will be the display I recommend. I am mostly
done with the EEPROM user access code, and have been spending weeks
creating the graphics in gimp and using the library's quick graphics to
speed the graphics loading as fast as possible with still looking really
nice.  It has been a fun learning expriance.

Best Regard and I will be in touch soon,
Jason

prenticedavid, 2019-10-23:
Please paste a LCD_ID_readreg report for your Red shield.
I would like to see what controller you have.

Regarding backlight current.   If the screen feels hot,  it is taking too much current.    It is very difficult to insert a series resistor in the backlight circuit of the Red pcb.

David.

closed 2021-08-24 by prenticedavid