prenticedavid/MCUFRIEND_kbv #117: Strange behaviour on STM32 Core, F10

bittmanv, 2019-11-07:


prenticedavid, 2019-11-07:
The Maple Core and ST Core start up differently.   Hence I enable the APB2 clocks for PORTS and remap SWJ from PB4:
```
#if defined(ARDUINO_ARCH_STM32)   //regular CMSIS libraries
#define REGS(x) x
#define GPIO_INIT()   { RCC->APB2ENR |= RCC_APB2ENR_IOPAEN | RCC_APB2ENR_IOPBEN | RCC_APB2ENR_IOPCEN | RCC_APB2ENR_IOPDEN | RCC_APB2ENR_AFIOEN; \
        AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_1;}
#else                                                                  //weird Maple libraries
#define REGS(x) regs->x
#endif
```

The Maple Core enables the PORTS and remaps SWJ by default.

I don't have a STM32F103RET6 chip.   So this RET6 SPECIAL is Untested.
However I have tested BluePill and Nucleo with both Maple and ST Cores.

reset() initialises GPIO.
readID() calls reset() if required.
begin() calls reset() always.
User code never needs to call reset()

Do you have a problem with any other controller?
I possess a Renesas R61520 board but I seem to remember programming the ID so it appears in reg(0x00A1)

Please can you run LCD_ID_readreg.ino
Observe reg(0xA1), reg(0xB0), reg(0xBF)

David.

David.

bittmanv, 2019-11-08:
Hi David,

thanks for quick response. 

I started readId on UNO to ensure default setup 👍 {using different library folder...)

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 FF FF 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 01 22 15 20 02	ILI9481, HX8357-B
reg(0x00C0) 00 00 4F 00 00 02 00 01 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 07 C3	Power Control
reg(0x00D2) 00 63 24 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 63 24 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control
```

I tried to avouid tft.reset() and I confirm, that is not needed. 

Same result I got on STM32 chip, and I confirm, that rest of graphical demo also works (after my manual PC0 initialization.

Vaclav

prenticedavid, 2019-11-08:
Your readreg is exactly the same as mine.   The readreg sketch is guaranteed to work on any platform.   It uses digitalWrite().

My memory is shot.   It was the R61526 that I programmed the ID.   It now appears in reg(0x0004) on a 1526 as well as the factory reg(0x00BF).

I will dig out a 1520 and try it on a 216MHz STM32F767 Nucleo.    I will see how the readID() performs.
All the same,  your F103RET6 is only 72MHz.    So I doubt if speed would cause a problem.

I will report back.  Probably tomorrow.

David.

prenticedavid, 2019-11-11:
I am confused.   The R61520 does not seem to recognise the unlock sequence from an STM32 that I issue at reset().   (So I can read reg(0x00BF) )
It works reliably with a Uno.

I do not understand why the AVR and the STM32 would be different.

I have re-enabled the test for R61520 using reg(0x00A1).
I originally introduced this before Release v2.9.5 in 2017
It produced some false IDs.   Obviously ILI9327.   I can't remember which other controllers were falsely identified.  

I have also added some delays to the read16bits() function used by readReg()
This should not affect performance.  It works with a 216MHz STM32F767   

David.

bittmanv, 2019-11-11:
Hi David, 
I downloaded your last change (only MCUFRIEND_kbv.cpp was changed, right ) and magic happened. Display started to work without my magic PC0 activation. Many thanks. 

Vaclav

closed 2019-11-11 by bittmanv