prenticedavid/MCUFRIEND_kbv #119: 3.5" display on STM32

lukewpr, 2019-12-14:
I'm having trouble with a 9486 display. The display works using the library on a Due, but won't have any of it on a bluepill. My 2.4" display works fine on both.

The diagnose support sketch just returns an ID of 0x0. I am sure I had this display working previously on the bluepill, and most likely this library but I can't quite remember.

prenticedavid, 2019-12-14:
I use Adapter shields.   So a Shield can be unplugged from a Due.   And plugged into the BluePill Adapter.

This means that a 2.4" is tested on both Due and BluePill.   Edit the LCD_ID_readreg defines to match your wiring.

Run the 3.5" on the Due.   Record the readreg report.
Run the 3.5" on the BluePill.  Record the readreg report.

I would expect the 3.5" to work the same.   It is unlikely to have a speed or timing issue.
Post your results for 2.4" and 3.5"
Please say which Core,  which library Release (or date of the Beta)
Which Adafruit_GFX version.   Which IDE version.

David.

lukewpr, 2019-12-14:
IDE is version 1.8.8, running on Windows if that makes any difference at all.
Adafruit_GFX is 1.5.4, 2.9.8 of your library. Thats as much as I can give you as I now can't compile anything.

I have 2 STM32 cores available, one says "stm32duino.com" and the other was installed from the boards manager.

The STM32duino one can't find series/i2c.h and wont compile, and the other can't find stddef.h.

Funnily enough the Windows store IDE won't even launch and I think that's what I used originally to get the display working

lukewpr, 2019-12-15:
Ok I got somewhere, probably going to have to hose the Arduino IDE and start again but I have code compiling with latest STM32duino core from github.

This is the 3.5" display on the STM32 when running readreg

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 02 02 3B 3B	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 0E 0E 0E 0E 0E 0E 0E 0E	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 04	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 94 86	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 54	RDID1
reg(0x00DB) 00 80	RDID2
reg(0x00DC) 00 66	RDID3
reg(0x00E0) 00 0A 00 08 00 08 0F 5C 72 5C 02 1D 09 13 2B 06	GAMMA-P
reg(0x00E1) 00 07 3E 03 05 0F 0B 23 3F 0B 0D 17 0A 3D 2F 01	GAMMA-N
reg(0x00EF) 00 80 00 10 60 40	ILI9327
reg(0x00F2) 00 18 A3 12 02 B2 12 FF 10 00 00 00	Adjust Control 2
reg(0x00F6) 00 54 80 66	Interface Control

Here is is on the Due

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 02 02 3B 3B	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 0E 0E 0E 0E 0E 0E 0E 0E	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 04	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 94 86	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 54	RDID1
reg(0x00DB) 00 80	RDID2
reg(0x00DC) 00 66	RDID3
reg(0x00E0) 00 08 00 08 08 08 0F 4D 72 5C 02 1D 09 13 21 02	GAMMA-P
reg(0x00E1) 00 03 1C 03 0F 0B 0F 23 BF 0B 05 17 0A 3F 25 01	GAMMA-N
reg(0x00EF) 00 80 00 10 60 40	ILI9327
reg(0x00F2) 00 18 A3 12 02 B2 12 FF 10 00 00 00	Adjust Control 2
reg(0x00F6) 00 54 80 66	Interface Control

I'll post the 2.4" display seperately :)

lukewpr, 2019-12-15:
Here's the 2.4" on the STM

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 0A 82 27 04	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 21 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 71	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 03 03	NVM Read
reg(0x00D3) 00 00 93 41	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 0F 16 14 0A 0D 06 43 75 33 06 0E 00 0C 09 08	GAMMA-P
reg(0x00E1) 00 08 2B 2D 04 10 04 3E 24 4E 04 0F 0E 35 38 0F	GAMMA-N
reg(0x00EF) 00 03 80 02 02 02	ILI9327
reg(0x00F2) 00 02 02 02 02 02 02 02 02 02 02 02	Adjust Control 2
reg(0x00F6) 00 01 00 00	Interface Control

And on the Due

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 0A 82 27 04	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 21 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 71	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 03 03	NVM Read
reg(0x00D3) 00 00 93 41	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 0F 16 14 0A 0D 06 43 75 33 06 0E 00 0C 09 08	GAMMA-P
reg(0x00E1) 00 08 2B 2D 04 10 04 3E 24 4E 04 0F 0E 35 38 0F	GAMMA-N
reg(0x00EF) 00 03 80 02 02 02	ILI9327
reg(0x00F2) 00 02 02 02 02 02 02 02 02 02 02 02	Adjust Control 2
reg(0x00F6) 00 01 00 00	Interface Control

prenticedavid, 2019-12-15:
I ran v2.9.8 Release on the weird RogerClark "stm32duino.com" Core (but dated 2018.5.11) 

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 02 02 3B 3B	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 0E 0E 0E 0E 0E 0E 0E 0E	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 04	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 94 86	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 54	RDID1
reg(0x00DB) 00 80	RDID2
reg(0x00DC) 00 66	RDID3
reg(0x00E0) 00 0F 21 1C 0B 0E 08 49 98 38 09 11 03 14 10 00	GAMMA-P
reg(0x00E1) 00 0F 2F 2B 0C 0E 06 47 76 37 07 11 04 23 1E 00	GAMMA-N
reg(0x00EF) 00 80 00 10 60 40	ILI9327
reg(0x00F2) 00 18 A3 12 02 B2 12 FF 10 00 00 00	Adjust Control 2
reg(0x00F6) 00 54 80 66	Interface Control

```

I strongly advise using the official Core from STMicroelectronics i.e. via the Boards Manager.
I tend to use a STLinkV2 dongle for uploading code.   ($3 from Ebay)

Note that your "9486" produces different readreg report to me.   And it is different for Due and BluePill.    So I suspect that you do not have a genuine Ilitek ILI9486.
What happens if you just force the ID e.g. with `tft.begin(0x9486);`

David.

lukewpr, 2019-12-15:
I'm pretty sure it is the official Core that I'm using, added the board manager URL from github.com/stm32duino. There's nothing in boards manager apart from that one.

I tried forcing the ID and it's just a white screen again.

I'll copy the sketch over to another machine and try that. Fresh Ubuntu install that's never had Arduino installed .

Thanks for your help :)

prenticedavid, 2019-12-15:
This is the proper Core https://github.com/stm32duino/Arduino_Core_STM32

stm32duino.com is no longer a valid web address.   It used to be the "RogerClark" MapleCore website. 

I would expect both  to work fine.

If you have a problem,  please quote the actual website,   Response from IDE Board Manager, IDE Library Manager.

David.

lukewpr, 2019-12-15:
You may not believe this, but it was down to the touch panel of all things! I noticed a crack on the panel, as it's not going to work with damage, I cut the ribbon to the panel and it started displaying when I powered it back up.

The "software scroll" text in the graphic text corrupts slightly, some of the pixels go blue and the text has a horizontal stagger to it.

I guess it was adding a resistance to the shared pins and it was enough to corrupt the data, where as the Due was fine as it is slower.

prenticedavid, 2019-12-16:
I would feel much happier if the readreg report matched the Due.   And even happier if it matches my report.

If you had run the Calibration sketch on a Uno,  it would have reported a BROKEN TOUCHSCREEN
Alternatively you could have just measured the Resistance with a DMM.

David.

closed 2021-08-24 by prenticedavid