prenticedavid/MCUFRIEND_kbv #121: Flipped horizontal Screen in 2.4 TFT LCD

wawinfo, 2019-12-30:
Please fix the issue with flipped horizontal , Some where I have read you fix that in new beta release. Where can i get that?
I have no lock with all suggestion in the web to fix that?

prenticedavid, 2019-12-30:
Which controller?
Which Arduino?
Did you install library **via the IDE Library Manager?**

Have you run **all** the library examples?
Most examples report the ID to Serial Terminal.   Copy-paste from the Seriak Terminal to your message.

David.

wawinfo, 2019-12-30:
On uno and Mega, both by IDE and bye copy directly, no issue about serial report, Evey thing work well, but flipped horizontal

An hour ago I do some changes in cpp file and it does work like a magic.
There is a line that has function WriteCmdData, and there is a mask number 2700, I changed it to A700 (I read somewhere), and after one week of challenge it solved.

wawinfo, 2019-12-30:
The ID was 0x5408

prenticedavid, 2019-12-30:
Looking at Release 2.9.8
```

    case 0x5408:
        _lcd_capable = 0 | REV_SCREEN | READ_BGR; //Red 2.4" thanks jorgenv, Ardlab_Gent
//        _lcd_capable = 0 | REV_SCREEN | READ_BGR | INVERT_GS; //Blue 2.8" might be different
        goto common_9320;

```
You can choose one or the other set of attributes.    It is better to set the attributes than to fiddle with `WriteCmdData(0x60, GS | 0x2700);`

Please note that panels vary.   It is not unusual for a batch from one manufacturer to be be different.

David.

closed 2021-08-24 by prenticedavid