prenticedavid/MCUFRIEND_kbv #123: Grand Central M4 - white display

camdey, 2020-01-07:
Hey @prenticedavid , thanks for the quick work on this last night. Not getting much success so far, here's what I know.

Screen stays white, regardless of which example I run. It is connected directly to the Grand Central M4 as a shield

I commented out `#define SUPPORT_9342` in MCUFRIEND_kbv.cpp after it was identified as such by diagnose_tft_support. It now prints in that Serial:
```
Diagnose whether this controller is supported
There are FAQs in extras/mcufriend_how_to.txt

tft.readID() finds: ID = 0x9342

MCUFRIEND_kbv version: 2.9.9

PORTRAIT is 320 x 240

Run the examples/graphictest_kbv sketch
All colours, text, directions, rotations, scrolls
should work.  If there is a problem,  make notes on paper
Post accurate description of problem to Forum
Or post a link to a video (or photos)

I rely on good information from remote users
```

graphictest_kbv prints to Serial (and nothing else) with a white screen:
```
Serial took 5000ms to start
ID = 0x9342
```

diagnose_touchpins prints:
```
Making all control and bus pins INPUT_PULLUP
Typical 30k Analog pullup with corresponding pin
would read low when digital is written LOW
e.g. reads ~25 for 300R X direction
e.g. reads ~30 for 500R Y direction

Testing : (A2, D8) = 31
Testing : (A3, D9) = 4
Diagnosing as:-
YP,YM:  (A2, D8) = 31
XM,XP:  (A3, D9) = 4
```

At this point it's worth me mentioning some issues I'm having with this board in the Arduino IDE. It correctly picks up the board and on the COM port but sometimes during uploading it fails at the end. Adafruit suggests:

> If you ever get in a 'weird' spot with the bootloader, or you have uploaded code that crashes and doesn't auto-reboot into the bootloader, click the RST button twice (like a double-click)to get back into the bootloader.
> The red LED will pulse, so you know that its in bootloader mode.
> Once it is in bootloader mode, you can select the newly created COM/Serial port and re-try uploading

This works but ends up creating a separate COM port. I have to select that COM port for uploading a sketch and then the original COM port for reading over Serial. Reading over Serial doesn't always seem to work either. For example, for the diagnose_touchpins sketch it wouldn't print what was within the void setup() function, I had to copy those Serial.prints over to the void loop() and added a 500ms delay and then it successfully prints the diagnosis.

I hope this can be fixed because it's a nightmare. I have to keep putting it in bootloader mode each time I want to upload a sketch.

Back to diagnosis, sometimes the Serial.prints do work fine. Here's LCD_ID_readreg:
```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 02	Inversion Control
reg(0x00B6) 00 0A 82 27 04	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 21 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 73	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 01 01	NVM Read
reg(0x00D3) 00 00 93 41	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 0F 16 14 0A 0D 06 43 75 33 06 0E 00 0C 09 08	GAMMA-P
reg(0x00E1) 00 08 2B 2D 04 10 04 3E 24 4E 04 0F 0E 35 38 0F	GAMMA-N
reg(0x00EF) 00 03 80 02 02 02	ILI9327
reg(0x00F2) 00 02 02 02 02 02 02 02 02 02 02 02	Adjust Control 2
reg(0x00F6) 00 01 00 00	Interface Control
```

camdey, 2020-01-07:
TFT display I'm using: https://www.velleman.eu/products/view/?id=435582

camdey, 2020-01-07:
As Velleman claims this is a 9341 controller, I tried forcing that ID graphictest_kbv, but no luck. White screen remains:
```
ID = 0x9341;                             // force ID
    tft.begin(ID);
```

I also have a feeling this TFT was identified as a 9341 when used on my Mega as I had this in my project sketch (used successfully for the last 12 months):
```
	uint16_t identifier = tft.readID();
	if (identifier == 0x9341) {
		Serial.println(F("Found ILI9341 LCD driver"));
	} else {
		Serial.print(F("No driver found"));
	}
```

prenticedavid, 2020-01-07:
The library is obviously reading the ID.   Even if it got 0x9342 instead of 0x9341.
LCD_ID_readreg does not use the library.   It relies on digitalWrite(), pinMode(), ... etc.
Your readreg report looks fine to me.   I presume that you ran this on the Grand Central.

You can force the ID to 0x9341 in every sketch.
Your shield is designed correctly i.e. 74LV245 buffers,  AMS1117 regulator.   i.e. ok for Uno, Mega. 

I have adjusted the delay macros.    It might correct the readID().   I hope that it corrects the regular display writing. 

David.

camdey, 2020-01-07:
Thanks for the help. I've merged your recent changes with my local file but still no luck (I quit the IDE after saving changes to the header file too). White screen for graphictest sketch and ID'd as 9341 in the diagnose_TFT sketch. Forcing 9341 or leaving as 9342 doesn't seem to make a difference.

prenticedavid, 2020-01-07:
I have added a GPIO_INIT() macro to configure all  the pins as digital.

If it still does not work,   I will have to do some thinking.   But it is late.    I will have to wait till tomorrow.

David.

camdey, 2020-01-07:
Unfortunately still no luck. No rush on this! If it's any help, I also have a Due on the way, which is much more similar to the M4 than a Mega, and if it works on that I'm not sure if that will help rule out any potential issues for you (though I guess it's unlikely it will).

If the bootloader issue with the M4 can't be fixed, I'll be sticking with the Due though.

camdey, 2020-01-08:
Looks like the bootloader issues are a Win10 thing, working smoother on my Mac. Still no luck with the TFT screen but the Serial printouts are better, looks like the optimal Serial delay is about 1500ms for the Grand Central.

Let me know if there's anything you'd like me to try but I'm happy to wait if you're busy this week.

prenticedavid, 2020-01-12:
Sorry.   I had forgotten about you.

I printed the GRAND CENTRAL M4 on paper.   Then double-checked against variant.cpp

Please try the new shield.h file.   And let me know.

David.

camdey, 2020-01-12:
Hey David, thanks that has done the trick! I've just run graphictest_kbv and it looks to be running correctly. It completes in 2.11s and all the scrolling and rotation seems to work. 

Here's a video I took if you'd like the check. It's still processing but should be available soon:
https://photos.app.goo.gl/EmpgQbah2iwfw9qC9

prenticedavid, 2020-01-12:
Your ILI9341 will work much faster.    The video looks ok.   I presume that LANDSCAPE_REV and INVERT SCREEN work 100%.

Please experiment with the DELAY macros
```
#define WRITE_DELAY { WR_ACTIVE8; }
#define IDLE_DELAY  { WR_IDLE4; }
#define READ_DELAY  { RD_ACTIVE16;}
```

Observe the Adafruit tests for glitches.
Note the FillScreen and Lines times in the Adafruit Tests Report.    FillScreen shows the fillRect() efficiency.   Lines shows the drawPixel() efficiency.
READ_DELAY affects readID() and readGRAM()
The Total time is useful number but recording FillScreen, Lines times shows the effects better.

I would expect that this will work with your 120MHz M4:
```
#define WRITE_DELAY { WR_ACTIVE4; }
#define IDLE_DELAY  { WR_IDLE2; }
#define READ_DELAY  { RD_ACTIVE8;}
```

If the data port was on consecutive port bits,   it is possible to reduce the "Total Time" to about 0.65 sec.    I would be happy with 1.00sec but 2.11sec is not unreasonable. 

Your ILI9341 works faster than other controllers.    You can possibly reduce the delays even further.    But you might get glitches with older controllers.     

David.

camdey, 2020-01-12:
> I presume that LANDSCAPE_REV and INVERT SCREEN work 100%.

Yeah they seem to work, I can video the entire test if you would like.

I notice in the scrolling examples there's often a [white line](https://i.imgur.com/lrT7d5e.jpg) at one end of the screen. I assume this is normal behaviour as it appears in all examples I tested below.

Thanks for the tips, here's some benchmark results:

**delays(8,4,16)**
FillScreen: 262,878
Text: 24,156 
Lines: 228,454
Total: 2.11 sec

**delays(4,2,8)**
FillScreen: 185,965
Text: 19,884
Lines: 186,249
Total: 1.55 sec

**delays(2,2,4)**
FillScreen: 160,372
Text: 18,471
Lines: 172,206
Total: 1.36 sec

**delays(2,2,2)**
FillScreen: 160,326
Text: 18,468
Lines: 172,211
Total: 1.36 sec

> If the data port was on consecutive port bits, it is possible to reduce the "Total Time" to about 0.65 sec. I would be happy with 1.00sec

Yeah it's a shame about that, same issue plagues the Mega2560 right? At least this one is faster. I haven't benchmarked my Mega yet but I'd suspect it's several times slower than this.

prenticedavid, 2020-01-12:
I would be "happy enough" with 4,2,8 and 1.55 sec.
Did you not get any glitches with 0,0,0 ?

There is little point in recording the Text times.   It only reflects the drawPixel() efficiency.

The White Line is intentional.   It shows whether the screen panel is aligned properly.   The Touch icons (house, phone, etc) on some panels obscure it. 

Yes,   running a Uno shield on a Mega2560 is heart breaking.
Running a Mega Shield on a Mega works nicely.   But most Mega shields are write-only.

Your SAMD51 can display JPEG images from PROGMEM or SD card.   The M4 cpu is fast.   An M7 cpu is much faster.

David.

camdey, 2020-01-12:
> Did you not get any glitches with 0,0,0 ?

Not that I noticed but you know the test better than I so [here's a full-length video](https://photos.app.goo.gl/M64KqqpgFdY56RCP7) of it set at 0,0,0.

I'm mostly drawing small bitmap icons so I think PROGMEM is better for this than the SD card, but will keep it in mind now that I can actually display things pretty quick!

BTW I played with some of the [M4 performance options](https://learn.adafruit.com/adafruit-grand-central?view=all#m4-performance-options-7-20) and got even faster benchmarks. Without glitches from what I could tell but if you'd like to see it, I'll show you the video.

**delays(0,0,0) + 200MHz overclock + "Fastest(-OFast)" optimization setting**
FillScreen: 82,679
Text: 9.418
Lines: 88,581
Total: 0.70 sec

Let me know if you'd like me to do any further testing. I'm pretty happy with sub-2sec times so will probably dial things back a bit to make sure it's stable. I'll leave it up to you to pick the best delay settings for shield.h. Will you merge this into master in the next week or so, or it will be awhile before that happens?

If you'd like any other displays tested on the Grand Central, just let me know and I can buy one as a favour to you for supporting the board in this library. I'm in the EU as well so easy for us to communicate at similar times of day.

prenticedavid, 2020-01-12:
I did not see any glitches in the video.    But the video does work very fast.

One further test.   Please comment the GPIO_INIT() macro.

If everything works ok,   I will add the 4,2,8 timing to Master branch.   

Job done.   Until someone asks for the other Adafruit boards.

David.

camdey, 2020-01-12:
> One further test. Please comment the GPIO_INIT() macro.

Commented out and everything still working as expected 👍 

Thanks for all the help David, I really appreciate it!

Cam

closed 2020-01-12 by camdey