prenticedavid/MCUFRIEND_kbv #124: 3.2" Open Smart LCD not working

sima997, 2020-01-10:
Hello,
I have an issue with this LCD to make in works. It only shows white screen and blinks sometimes. Also ID that is outputed in serial monitor doesn´t match with listed ones.

I am using Arduino Mega and i´ve already made changes in files, that are decribed in tutorial.

Thank you for reply.
[image]
[image]
[image]

Here is ID.
Serial took 0ms to start
ID = 0xE1E1

Here is output from readreg

`Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) C1 C1	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) C5 D5 D7 D7	Manufacturer ID
reg(0x0009) C9 D9 DB DB DB	Status Register
reg(0x000A) CB DB	Get Power Mode
reg(0x000C) CD DD	Get Pixel Format
reg(0x0061) E1 F1	RDID1 HX8347-G
reg(0x0062) E3 F3	RDID2 HX8347-G
reg(0x0063) E3 F3	RDID3 HX8347-G
reg(0x0064) E5 F5	RDID1 HX8347-A
reg(0x0065) E5 F5	RDID2 HX8347-A
reg(0x0066) E7 F7	RDID3 HX8347-A
reg(0x0067) E7 F7	RDID Himax HX8347-A
reg(0x0070) F1 F1	Panel Himax HX8347-A
reg(0x00A1) E1 F1 F3 F3 F3	RD_DDB SSD1963
reg(0x00B0) F1 F1	RGB Interface Signal Control
reg(0x00B4) F5 F5	Inversion Control
reg(0x00B6) F7 F7 F7 F7 F7	Display Control
reg(0x00B7) F7 F7	Entry Mode Set
reg(0x00BF) FF FF FF FF FF FF	ILI9481, HX8357-B
reg(0x00C0) C1 D1 D3 D3 D3 D3 D3 D3 D3	Panel Control
reg(0x00C8) C9 D9 DB DB DB DB DB DB DB DB DB DB DB	GAMMA
reg(0x00CC) CD DD	Panel Control
reg(0x00D0) D0 D1 D3	Power Control
reg(0x00D2) D3 D3 D3 D3 D3	NVM Read
reg(0x00D3) D3 D3 D3 D3	ILI9341, ILI9488
reg(0x00D4) D4 D5 D7 D7	Novatek ID
reg(0x00DA) DB DB	RDID1
reg(0x00DB) DB DB	RDID2
reg(0x00DC) DC DD	RDID3
reg(0x00E0) E1 F1 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3	GAMMA-P
reg(0x00E1) E1 F1 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3	GAMMA-N
reg(0x00EF) EF FF FF FF FF FF	ILI9327
reg(0x00F2) F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3 F3	Adjust Control 2
reg(0x00F6) F7 F7 F7 F7	Interface Control
`

prenticedavid, 2020-01-11:
Yes,  you have a HX8352-B pcb.   Yes,  you have plugged the Shield into your Mega2560.

You should follow the instructions e.g. from extras/mcufriend_how_to.txt:

```
19. OPEN-SMART Shields have different wiring to regular Uno Shields:
    Edit utility/mcufriend_shield.h:  #define USE_SPECIAL
    Edit utility/mcufriend_special.h: #define USE_OPENSMART_SHIELD_PINOUT
    Edit MCUFRIEND_kbv.cpp:           #define SUPPORT_8352B
```

And post the lines from your **three** edited files.    i.e. file name + location and the single edited line.

The readreg sketch needs #defines to match the wiring on your Open-Smart board.

Let me know how you get on.    I own the same shield.    It works for me.
If you are using the Beta,  there are separate defines in mcufriend_special.h e.g.
`#define USE_OPENSMART_SHIELD_PINOUT_MEGA
`

David.

sima997, 2020-01-11:
Thank you for your reply.

Here are screenshots of my edited files.
[image]
[image]
[image]
[image]

prenticedavid, 2020-01-11:
I asked for file + single line e.g.
```
C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\utility\mcufriend_special.h
#define USE_OPENSMART_SHIELD_PINOUT_MEGA
```
Your images are a nightmare to read.    But it "looks" as if you have edited them correctly.

I suggest that you re-install the library from scratch.   Then do the edits again.
And please copy-paste text instead of posting unreadable images.

David.

sima997, 2020-01-11:
Sorry,

My path is C:\Users\kalom\OneDrive\Dokumenty\Arduino\libraries\MCUFRIEND_kbv-master\utility\mcufriend_special.h

I´ve already try to reinstal, but always the same.

prenticedavid, 2020-01-11:
I have used the current Beta.

```
C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\utility\mcufriend_special.h
#define USE_OPENSMART_SHIELD_PINOUT_MEGA

C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\utility\mcufriend_shield.h
#define USE_SPECIAL             //check for custom drivers

C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\MCUFRIEND_kbv.cpp
#define SUPPORT_8352B             //HX8352B

```
And the graphictest_kbv example worked fine with my OpenSmart-8352B on a Mega2560.
```

Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino-1.8.9\hardware\arduino\avr\libraries\SPI 
Using library Adafruit_GFX_Library at version 1.7.3 in folder: C:\Users\David Prentice\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Mcufriend_kbv at version 2.9.9-Beta in folder: C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv 
"C:\\Program Files (x86)\\Arduino-1.8.9\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\DAVIDP~1\\AppData\\Local\\Temp\\arduino_build_425669/graphictest_kbv.ino.elf"
Sketch uses 33246 bytes (13%) of program storage space. Maximum is 253952 bytes.
Global variables use 1601 bytes (19%) of dynamic memory, leaving 6591 bytes for local variables. Maximum is 8192 bytes.
```

Note that you can avoid all the "edits".   Just build the "graphictest_opensmart" example.

I also notice that your library path seems to be on `OneDrive\Dokumenty` instead of mine which is `Documents`
Please can you post your "verbose build report".   i.e. copy-paste only the equivalent lines that I have pasted.

David.

prenticedavid, 2020-01-11:
As a general rule I expect developers to Fork the library from GitHub.   (and use Issues or Pull Requests)

I expect Arduino "users" to install the v2.9.8 Release **via the IDE Library Manager.**   (and use Arduino.cc Forum messages)

Only "Download ZIP" if you know what you are doing.    You _might_ get multiple libraries.

David.

sima997, 2020-01-11:
Here is my build report:
 ```
Using library SPI at version 1.0 in folder: D:\Software\Arduino\hardware\arduino\avr\libraries\SPI 
Using library Adafruit_GFX_Library at version 1.7.4 in folder: D:\Software\Arduino_Project\libraries\Adafruit_GFX_Library 
Using library MCUFRIEND_kbv-master at version 2.9.9-Beta in folder: D:\Software\Arduino_Project\libraries\MCUFRIEND_kbv-master 
"D:\\Software\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\kalom\\AppData\\Local\\Temp\\arduino_build_515268/graphictest_kbv.ino.elf"
Sketch uses 32748 bytes (12%) of program storage space. Maximum is 253952 bytes.
Global variables use 1600 bytes (19%) of dynamic memory, leaving 6592 bytes for local variables. Maximum is 8192 bytes.
```

My instalation was from github.

prenticedavid, 2020-01-11:
The build is Using library MCUFRIEND_kbv-master at version 2.9.9-Beta in folder: D:\Software\Arduino_Project\libraries\MCUFRIEND_kbv-master 

Your edits were from C:\Users\kalom\OneDrive\Dokumenty\Arduino\libraries\MCUFRIEND_kbv-master\MCUFRIEND_kbv.cpp

I suggest that you:
1.  leave IDE.
2.  delete all MCUFRIEND_kbv directories.  
3.  start IDE
4.  install MCUFRIEND_kbv via the IDE Library Manager.
5.  edit the three files.
6.  run graphictest_kbv example.

If you still have a problem,  post your edit "report" and your "build report"

David.

sima997, 2020-01-12:
Path is different because I´ve try to relocate arduino files from one drive.

I made all the stuffs, that you suggested to me, but always the same.

Here is my edited files.
```
D:\Software\Arduino_Project\libraries\MCUFRIEND_kbv\utility\mcufriend_shield.h
#define USE_SPECIAL  
```

```
D:\Software\Arduino_Project\libraries\MCUFRIEND_kbv\utility\mcufriend_special.h
#define USE_OPENSMART_SHIELD_PINOUT //thanks Michel53
```

```
D:\Software\Arduino_Project\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp
#define SUPPORT_8352B             //HX8352B
```

My build report
```
Multiple libraries were found for "MCUFRIEND_kbv.h"
 Used: D:\Software\Arduino_Project\libraries\MCUFRIEND_kbv
Multiple libraries were found for "SPI.h"
 Used: D:\Software\Arduino\hardware\arduino\avr\libraries\SPI
Multiple libraries were found for "Adafruit_GFX.h"
 Used: D:\Software\Arduino_Project\libraries\Adafruit_GFX_Library
Using library SPI at version 1.0 in folder: D:\Software\Arduino\hardware\arduino\avr\libraries\SPI 
Using library Adafruit_GFX_Library at version 1.7.4 in folder: D:\Software\Arduino_Project\libraries\Adafruit_GFX_Library 
Using library MCUFRIEND_kbv at version 2.9.8 in folder: D:\Software\Arduino_Project\libraries\MCUFRIEND_kbv 
"D:\\Software\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\kalom\\AppData\\Local\\Temp\\arduino_build_232014/graphictest_kbv.ino.elf"
Sketch uses 32152 bytes (12%) of program storage space. Maximum is 253952 bytes.
Global variables use 1600 bytes (19%) of dynamic memory, leaving 6592 bytes for local variables. Maximum is 8192 bytes.
```
I found in build report "Multiple libraries were found..." It is normal or not ?

prenticedavid, 2020-01-12:
You seem to be editing and building from the correct library(s)

If "Multiple libraries were found",   copy-paste the message(s).
Or by copying the whole build report to a error.txt file and attaching the file.

David.

sima997, 2020-01-12:
Here is my whole build report.
[error.txt](https://github.com/prenticedavid/MCUFRIEND_kbv/files/4050447/error.txt)

prenticedavid, 2020-01-12:
I am using IDE v1.8.9
My build report just says:
```
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino-1.8.9\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino-1.8.9\hardware\arduino\avr
Detecting libraries used...
```

Did you leave IDE,  delete folders, start IDE ?

I can only suggest that you re-start your PC.   Start IDE.   Build the sketch.   Post the new error.txt report.

Please run the LCD_ID_readreg example (after editing the defines to match your Open-Smart pinout)

I am sorry that you are having all these problems.

David.

closed 2021-08-24 by prenticedavid