prenticedavid/MCUFRIEND_kbv #125: 3.97" TFT on Nucleo64 / F303RET board -- problems

alfsch1, 2020-02-09:
Hello David, 
great work. I put Mcufriend 3.97" TFT on Nucleo board (was F302 , but i changed chip to F303RET ) , so is STM32F303 now;  compiled (arduino) ...but didnt run. After some search with DSO, i found write signals with about 10ns ! so just to check: reduced clock (PLL mult. x 5) to 40 MHz...and TFT runnning !
fine...so far.
now my problem: to get some fast data from ADC, i need to control/adjust ADC , DMA... but how to do ??? 
A: get the HAL calls working in arduino --- no success now.
B: get the (your) TFT-lib running in STM/cube (workspace in cube-IDE ) -- no success until now.

so: what should i do? and how...i think, both ways would work, but i am to silly...so please help me.

best regards
Alfred

prenticedavid, 2020-02-09:
The current Beta supports the F303

```
#elif defined(STM32F303xE)
#define WRITE_DELAY { }
#define READ_DELAY  { RD_ACTIVE8; }  //thanks MasterT
#define GPIO_INIT()   { RCC->AHBENR |= RCC_AHBENR_GPIOAEN | RCC_AHBENR_GPIOBEN | RCC_AHBENR_GPIOCEN; \
                      /* AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_1; */ }
#define PIN_OUTPUT(port, pin) PIN_MODE2((port)->MODER, pin, 0x1) //thanks fpiSTM

```

From memory,  the v2.9.8Release that is installed by the IDE Library Manager has bad timing for the F303.    (it assumed it was Cortex-M3 and not M4)   

I don't have a F303.   I rely on users' feedback.

Please use the Beta.    And let me know if _any_ library examples do not work properly.

David.

alfsch1, 2020-02-09:
ok, i try. and thx for your fast response.
 ...and yes, delay should be about same, because it is F4 core at 72 MHz (standard max. speed).
but just to say : i was very surprised, to see a some 10..14 ns long write puls , coming from a 72MHz CPU. and btw , coming from a curious mix of shifts and ORs...to a port register over the internal AHB bus. seems, it can access ports in just one clock cycle.
+
can you give me some advice, how to get your lib running in STM32cubeIDE ?

Alfred

prenticedavid, 2020-02-09:
I suggest that you start with Arduino.   The official STM32 Core supports your Nucleo-F303.    I would expect the shield to plug into the Nucleo.   Select Nucleo64->F303.   Run library examples.

Get the Arduino projects running first.
I can build in Keil or Rowley.   Which both support all ARM targets.

I have no plans for Cube.

Yes,  M4 has impressive GPIO.   M3 is quite good.   M0 is crap.
The reason for the somewhat "pointless" WR_ACTIVE macro is that it produces clock-dependent delays because the Compiler must always obey a volatile access.

David.

alfsch1, 2020-02-09:
so...i just delete the mcufriend directory in the library folder and loaded from zip : ..test_1503.
hope, this is, what you say...date: 26.1.2020 ; 
and no...nothing on TFT with original setting (pll x 9 = 72 M), just with x 5 = 40 M its running.
pll x 6 = 48 MHz shows some graphics, with errors (missing parts), but text is away.
so 40 M is maximum with this timing.

+
cubeIDE is just another IDE...similar Keil. (original was Atollic TrueStudio , STM buyed them..)
so : howto import to ..Keil? (should be about same )

Alfred

alfsch1, 2020-02-09:
[image]
...see: how to do a correct ckock tree in arduino ? (i could just copy the generated code from cube to arduino, but i dont know, where to place it - without a lot of error messages.)
Alfred

prenticedavid, 2020-02-09:
The current Beta is always "master".
Ignore other branches unless specifically advised.

You don't do anything in Arduino.
Install Arduino IDE.
Tools->Board->Board Manager:  install STM32 
Tools->Board: select Nucleo64, select F303

File->examples->MCUFRIEND_kbv: select graphictest_kbv

Seriously.   Just run code from "out of the box".

If you have a problem with Arduino,   take notes on paper.
Then write all the steps from your notes.   Number each step.

I have NO intention of even looking at your Cube images until you have run Arduino.

David.

alfsch1, 2020-02-09:
sorry, i forgot to say: this was for shure the first, i did.
and it didnt do any output...until slowing down the master PLL.
but...now:
so...i do again, to show you the output:
>>
     |                                            ^
In file included from /home/alfsch/.arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/wiring.h:34,
                 from /home/alfsch/.arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/Arduino.h:32,
                 from /home/alfsch/Arduino/libraries/Adafruit_GFX_Library/Adafruit_GFX.h:5,
                 from /home/alfsch/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.h:16,
                 from /home/alfsch/Arduino/libraries/MCUFRIEND_kbv/MCUFRIEND_kbv.cpp:24:
/home/alfsch/.arduino15/packages/STM32/hardware/stm32/1.8.0/cores/arduino/wiring_constants.h:110:14: note: declared here
  110 | typedef bool boolean __attribute__((deprecated));
      |              ^~~~~~~
Bibliothek SPI in Version 1.0 im Ordner: /home/alfsch/.arduino15/packages/STM32/hardware/stm32/1.8.0/libraries/SPI  wird verwendet
Bibliothek Adafruit_GFX_Library in Version 1.7.0 im Ordner: /home/alfsch/Arduino/libraries/Adafruit_GFX_Library  wird verwendet
Bibliothek MCUFRIEND_kbv in Version 2.9.8 im Ordner: /home/alfsch/Arduino/libraries/MCUFRIEND_kbv  wird verwendet
Bibliothek SrcWrapper in Version 1.0.1 im Ordner: /home/alfsch/.arduino15/packages/STM32/hardware/stm32/1.8.0/libraries/SrcWrapper  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Nucleo-64.
<<
-- dont know, what is wrong now. i did, as you told me.
(btw. >The current Beta is always "master". < i didnt know. but it was anyway my first try. )

 graphictest_kbv was running fine, with 40 MHz clock.
but no touch...i had to find out the wiring for the touch. then it was working fine also.
then...tried the ADC. working...but useless, i need much more speed than the arduino setup makes.
and i need ADC3 , which needs setting up the clock tree - and i have no idea , how to do with standard arduino; my idea was, just to add some HAL generated code, to do this, and this should work, but i could not find some example, where to put these code and not disturb all the other settings. thats, why i asked...

Alfred

prenticedavid, 2020-02-10:
Yes,  the STM32 Compiler whinges about _boolean_
Ok,  you reduced clock speed.   It would be better to just change WRITE_DELAY in mcufriend_shield.h e.g.
```

#elif defined(STM32F303xE)
#define WRITE_DELAY { WR_ACTIVE2; }  //for alfsch1
#define READ_DELAY  { RD_ACTIVE8; }  //thanks MasterT
#define GPIO_INIT()   { RCC->AHBENR |= RCC_AHBENR_GPIOAEN | RCC_AHBENR_GPIOBEN | RCC_AHBENR_GPIOCEN; \
                      /* AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_1; */ }
#define PIN_OUTPUT(port, pin) PIN_MODE2((port)->MODER, pin, 0x1) //thanks fpiSTM
```

I would appreciate a clear explanation for any changes that you make.   e.g. edit a file.
It is important to know whether you are running at 72MHz.

Life is much simpler when you Fork the GitHub project.   Then we can see each other's code.

Touch is fairly easy to calibrate.
ADC can be configured at a later date.

David.

alfsch1, 2020-02-10:
now: i delete (old) libs and put master branch to libs folder. (and switch off warnings.)
put back pll (x9) to original value. so all should be "new and clean".
compiler runs, upload - but no picture.
so i made the suggested change:
#elif defined(STM32F303xE)
#define WRITE_DELAY { WR_ACTIVE2; }  //for alfsch1      was : #define WRITE_DELAY { }
#define READ_DELAY  { RD_ACTIVE8; }  //thanks MasterT
#define GPIO_INIT()   { RCC->AHBENR |= RCC_AHBENR_GPIOAEN | RCC_AHBENR_GPIOBEN | RCC_AHBENR_GPIOCEN; \
                      /* AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_1; */ }

> and now:
[image]

(protective film is still on screen)

> great. seems good timing now, for F303 at original 72MHz.     thx a lot.  :)

so next : how to make all the settings for adc and opamps... uuuh.

Alfred

prenticedavid, 2020-02-10:
First off.   Study the graphictest_kbv output carefully.
I have never seen an ST7793.   It can not Band Scroll.  Do you see a white area during vertical scroll?   Or random pixels?   
Software scroll and Invert Display should work.

Then try with `#define WRITE_DELAY { WR_ACTIVE; }` 
Do you see any glitches?

Regarding ADC.   The Arduino just provides a simple Analog_read() function.
You can always write your own ADC functions in any sketch.   Either with bare-metal or with HAL functions.

I suspect that you want to use a regular IDE.   Which one?
I would still familiarise yourself with Arduino first.

David.

alfsch1, 2020-02-10:
yes...{ WR_ACTIVE; } also running;  :) little bit faster, says: 2.48 sec.

the TFT (from ali ) is: 3,97-zoll TFT LCD , controller ST7793 , 400*240 pix.
i reduced VCC with 2r2 to 3.0V , because back LEDs (maybe without resistors) getting hot;
now, at 3.0V , it takes 200mA from USB, still good brightness, and temp is 33.1° at LED side (at 23.2° room), seem ok for me.

on graphictest_kbv all seem running fine, also with { WR_ACTIVE; } (one delay, i think); no glich;
vert. scroll smooth, just about 8mm heigh random pixel area, moving with the picture; 
maybe this is just a wrong size definition somewhere, because 400 pix is a bit unusual.
after scroll, sometime the random pix area is still on one side visible, but same area on other side is outside screen; so just looks like: scroll ok, but wrong size of scroll-area.

and about IDE : ( at work i use...what i have to use. ) 
but for me, just for fun, i use what is giving me best result in less time, so for most of my funny things i use arduino (with AVR , nano etc.), what is really great . 
just for the bigger cpu, STM32xxx , i use the STM32CubeIDE , because in arduino i see nothing about the settings of such a complex chip and where i can change them; 
on contrast, in STM32CubeIDE , i can see the ckock tree, adjust it, set all peripheral to the possible pins and set them to the desired state; then : "generate code" ...and voila, a C program frame is generated, with all the desired settings; now just write the program...very easy and almost same, as in arduino.
So, if can make the setup for ADC (and so on) as i want, i would use the arduino IDE - no doubt.

Alfred

alfsch1, 2020-02-10:
ok, i tried now 2 hours long, to a sample from adc1 , with full speed. data always = 0 .
so , now i can say: better , to use the TFT lib in STM32CubeIDE , with debugger and view of all peripheral registers . 

this was my experiment:
`
ADC1->SQR1 = 0x000180;                                                // adc1 , ch6 = 1. seq. regular in ; 
ADC1->SMPR1 = 0x00010;                                                // adc1 , sample time 4.5 cyc
ADC1->CR = 0x01;                                                      // AD enable

  for( j=0 ; j<NN+1 ; j++ )
  { 
loops=0;
ADC1->CR = 0x05;                                                      // adc start
while(  (ADC1->CR & ADC_CR_ADSTART ) )
{ loops++; if(loops == 1000) break;   }                               // adc ready ?

adval = ADC1->DR ;
   `
...something wrong, maybe wrong clock ...

-- so can you help me, howto get it to STM32CubeIDE ?

Alfred

alfsch1, 2020-02-14:
now got ADC running in arduino !
with some help in stm32duino forum...call to HAL lib working, and direct register access also;
can get about 4 Ms now , with software start+polling. thats ok.
but the touch lib no more working - because  uses (i suppose) standard analogRead(x) - but this hangs up now. (i had to remove the standard init for analog , this was the trick to get hi speed. )
i could post here, howto....if you like.

Alfred

prenticedavid, 2020-02-14:
Yes,  TouchScreen.h will call the regular analogRead()
You can  copy TouchScren.h and TouchScreen.cpp to local tabs in your Sketch.
And access the local copy with #include "TouchScren.h"

Then you can edit the local TouchScreen.cpp to use your local ADC function.    Remember that the library code probably expects 10-bit ADC results.   

If you are writing bare-metal code,  you simply read and save the "Arduino" values of the ADC registers.   Call your fast ADC.   Restore ADC registers.

The ADC does not seem relevant to the mcufriend_how_to.txt file.
I suggest that you leave the discussion on Arduino or stm32duino Forum.

David.

alfsch1, 2020-02-14:
thx , David, for your help.
i will try with a copy and edit "touch..." , if i get it working, i have all i need in arduino...and can make the program now.... (and then close discussion ).

thx
Alfred

alfsch1, 2020-02-14:
ok, got it working.  was a bit hard fight...init ADC1 , 2, and 3 to hi speed 
and then mod the touch lib, to read ADC /analog in 10bits and 20x oversampling (otherwise ADC too fast for "touch"... )

thx for your fast help and (maybe) see you, with next "problem TFT " ... :)

best regards
Alfred

prenticedavid, 2020-02-14:
Good luck.     I am sure that stm32duino Forum can help with any problems.

David.

closed 2020-02-21 by alfsch1