prenticedavid/MCUFRIEND_kbv #130: Touch Calibration DEMO

badonj001, 2020-04-20:
This is a guided Touch Screen Calibration for the MCUFRIEND_kbv library for arduino and a modified version of the included example TouchScreen_Calibr_native. Works in all orientations, you just need to make use of the collected data. 

https://youtu.be/odUhmjiWc8w

prenticedavid, 2020-04-20:
Have you looked at the Calibration sketch in the current Beta (i.e. master Branch) ? 
It has several improvements over the v2.9.8 Release

The Calibration Example has always worked in all rotations.
You get the same Portrait Calibration result.  i.e. Touch pins, TS_LEFT, TS_RIGHT, ... **const** values

When you use Touch in your sketch,  you map() the results to suit the rotation.
Look at the TouchShield example.

The Calibration sketch was developed from the UTFT / URTouch sketch.   Hence the cross-hair navigation order.    A clockwise navigation would be more natural but would "feel different" to existing users. 

David.

badonj001, 2020-04-21:
I haven't looked at the new releases yet, I just wanted to share in the
event it was useful t someone. It was just something I did for fun one day.

closed 2021-08-24 by prenticedavid