prenticedavid/MCUFRIEND_kbv #131: ESP32 with R61581

David-Schiedek, 2020-04-21:
Hello,

ESP32 with this library works correctly with 3.5" LCD (R61581 driver) only with downsizing ESP32 to 80 MHz. When I flashing with 240MHz, the display after "begin" go to whitescreen. It is any settings in this library to correct it? I'm trying some changes with `#define WRITE_DELAY {  }
#define READ_DELAY  {  }`
by STM32 settings, but without success.

When ESP32 is flashing on 80MHz, it is OK, but if is switch on BLE -> whitescreen. It frustrate me ;-o

prenticedavid, 2020-04-21:
Do some experimenting.

You "know" the ID.   So force `tft.begin(0x1581);` in your experiments.
Then adjust WRITE_DELAY() until you get glitches in graphictest_kbv graphics.
Observe if ID value fails
Observe if Software Scroll fails.
Keep notes.

I would hope that the current Beta is safe for reading ID.   I suspect that readID() is failing before any write operation. 

Bodmer supports R61581 shields in his TFT_eSPI library.
Compare behaviour.

David.

David-Schiedek, 2020-04-21:
With TFT_eSPI library it was same. I tried everything, change some LCDs and at last I changed ESP32 to another one and voilá ;-)
The origin ESP was some old item. Thank you David for your time and great job!

tarikcosovic, 2020-04-23:
I just want to thank David for his amazing work, this library is super useful!

closed 2021-08-24 by prenticedavid