prenticedavid/MCUFRIEND_kbv #133: Problems with many icons in flash memory

teHwHoly, 2020-05-08:
Hi!

I'm not quiet sure, but I'm getting errors in the pictures, when using pushColors() with arrays in Flash (PROGMEM). This occurs regularly with the same grain pattern, when too many arrays are defined. In the function I can see:
```
static void pushColors_any(uint16_t cmd, uint8_t * block, int16_t n, bool first, uint8_t flags)
{
    uint16_t color;
    uint8_t h, l;
	bool isconst = flags & 1;
	bool isbigend = (flags & 2) != 0;
    CS_ACTIVE;
    if (first) {
        WriteCmd(cmd);
    }
    while (n-- > 0) {
        if (isconst) {
            h = pgm_read_byte(block++);
            l = pgm_read_byte(block++);
[..]
```
May the pgm_read_byte() function have problems with big adresses? Could it be replaced by pgm_read_byte_far() to access areas at the end of the flash memory in the arduino mega?
Or is that probably another problem?
First problems occur in the image at address 57938 with a size of 16200 bytes, beginning at the lower half of the image.

prenticedavid, 2020-05-08:
Yes,   it assumes your arrays live in PROGMEM.   And never exceeds 64kB.

If you have large numbers of images,  store them in FAR memory.   Then use memcpy_PF() to copy to SRAM for pushColors()

I have posted sketches that display JPEG images stored in Flash memory.    However you need >4kB SRAM to decode the JPEG e.g. mega128, xmega128, mega2560, ...

David.

teHwHoly, 2020-05-08:
Wow, thank you so much for the fast response! I'll check if I will modify the library for my case or copy the data to SRAM, depending on the time it takes.

Thanks!

closed 2021-08-24 by prenticedavid