prenticedavid/MCUFRIEND_kbv #134: how to display a random image using showBMP_kbv_Uno

RedRhino4153, 2020-05-14:
I am currently using the showBMP_kbv_Uno ino and it works great to display the images stored on the MicroSD card but it displays them in the order they are saved on the card (ie... numerically or alphabetically).  Is there any way to have them get displayed randomly?  Is there a variable in the code I can activate to make this happen?
Thanks for the help.

prenticedavid, 2020-05-14:
Look at the code.   It reads the directory.   It displays any filenames that match the pattern e.g.  .bmp
It could use a proper wildcard match but the current strstr() does it good enough.

Write a function to read through the directory looking for matches. Store the index numbers in a global integer array.
Write a function to find a filename by index.
Then your loop() just uses random() to generate an index.  Display that filename.

You can afford a small integer array on a Uno.   Larger if you remove support for Palette format .BMPs
A Mega or Due could use store an array of filenames instead of integer indexes.

For an example,  it is wise to display in a fixed order.
YMMV.

David.

RedRhino4153, 2020-05-14:
Thanks.  I have no idea how to do that.  I have a group of 100 images (ie... 001.bmp, 002.bmp etc) on the SD card that I want to randomly display.
I will do some homework and see what I can come up with.
I realize that file size is an issue on an Uno, but that it what I am using due to form factor.
If you have any sample code for creating that array, that is optimized to preserve file size I would be grateful.

Ryan

prenticedavid, 2020-05-14:
Seriously.  If your filenames are 001.bmp to 999.bmp you don't need to do any directory scanning.
You just do

```
  int n = random(1000, 1999);
  char buf[10];
  itoa(buf, n, 10);
  strcat(buf, ".bmp");  //e.g. "1987.bmp"
  showBMP(buf + 1, x, y);  //e.g. "987.bmp"
```

For simple things like this,  the Arduino Displays Forum is probably a better place than posting on GitHub.

David.

RedRhino4153, 2020-05-14:
David,
Seriously? Thanks for the hint.  I am sorry I am a total noob and an using your great code to make my project work.  I bought the board and screen combos and your code is in the examples folder.
So I am going to be a pain and ask another noob question as I have been trying unsuccessfully to plug that section of code into the rest of it and make it work. Where the hell do I put it/in the place of what do I put it?
Thanks and sorry,
Ryan

prenticedavid, 2020-05-14:
First off.   Install from the IDE Library Manager.   Do not install from some random CD or website.

I will post an adaptation of the example in the v2.9.8 Release.   Probably tomorrow.

David.

prenticedavid, 2020-05-15:
Change the loop() in the example sketch to this.
```
void loop()
{
    char *nm = namebuf + pathlen;
    uint8_t ret;
    uint32_t start;
    {
        strlwr(nm);
        int n = random(1001, 1200);
        char buf[10];
        itoa(n, buf, 10);   // 1001 - 1200
        strcpy(nm, buf+1);  // 001 - 200
        strcat(nm, ".bmp"); // 123.bmp
        if (strstr(nm, ".bmp") != NULL && strstr(nm, NAMEMATCH) != NULL) {
            Serial.print(namebuf);
            Serial.print(F(" - "));
            tft.fillScreen(0);
            start = millis();
            ret = showBMP(namebuf, 5, 5);
            switch (ret) {
                case 0:
                    Serial.print(millis() - start);
                    Serial.println(F("ms"));
                    delay(5000);
                    break;
                case 1:
                    Serial.println(F("bad position"));
                    break;
                case 2:
                    Serial.println(F("bad BMP ID"));
                    break;
                case 3:
                    Serial.println(F("wrong number of planes"));
                    break;
                case 4:
                    Serial.println(F("unsupported BMP format"));
                    break;
                case 5:
                    Serial.println(F("unsupported palette"));
                    break;
                default:
                    Serial.println(F("unknown"));
                    break;
            }
        }
    }
}
```
but you could just as easily use sprintf() e.g.

```
        char filepath[32];
        sprintf(filepath, "/my_directory/%03d.bmp", random(1, 123));
        uint8_t ret = showBMP(filepath, x, y);

```

David.

closed 2021-08-24 by prenticedavid