prenticedavid/MCUFRIEND_kbv #135: 'A2' was not declared in this scope.

ghosttk, 2020-05-23:
I have a wemos UNO style Esp32 board (already hard wired).
Pluging with my paralell shield display, it passed the [diagnose_TFT_support.ino](https://github.com/prenticedavid/MCUFRIEND_kbv/blob/master/examples/diagnose_TFT_support/diagnose_TFT_support.ino)  in arduino IDE , and displayed the informations on the display, which showed the display ID was 0XB509 (strange thing is , there was not any info printed on Serial port,).
But running other examples  resulted in errors like:
'A2' was not declared in this scope.

Does it needs to change pin defines somewhere or  change some new defines?
But I didn't find the right place.

ghosttk, 2020-05-23:
Running diagnose_Touchpins, I got these outputs:
Making all control and bus pins INPUT_PULLUP
Typical 30k Analog pullup with corresponding pin
would read low when digital is written LOW
e.g. reads ~25 for 300R X direction
e.g. reads ~30 for 500R Y direction

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40080f15  PS      : 0x00060f30  A0      : 0x800d0c20  A1      : 0x3ffb1f60  
A2      : 0x00000007  A3      : 0x00000005  A4      : 0x0800001c  A5      : 0x00000003  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x3f403f54  A9      : 0xffffffff  
A10     : 0xffffffff  A11     : 0x00000064  A12     : 0x08000000  A13     : 0x4a000000  
A14     : 0xffffffff  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xffffffff  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: ffffffffffffffffffffffffffffffff00000000000000000008ccceeeffffff

Backtrace: 0x40080f15:0x3ffb1f60 0x400d0c1d:0x3ffb1f80 0x400d1cda:0x3ffb1fb0 0x40087f85:0x3ffb1fd0

ghosttk, 2020-05-23:
I copied "int XP = 27, YP = 4, XM = 15, YM = 14; " from TouchScreen_Calibr_native.ino, the "A2 not declared" error dispeared.
But running TouchScreen_Calibr_native.ino there were only text on the display , no crosshair to touch. 
Serial port reported :
16:15:23.903 -> TouchScreen.h GFX Calibration
16:15:23.903 -> Not possible to diagnose Touch pins on ARM or ESP32
16:15:23.950 -> ID = 0xb509

prenticedavid, 2020-05-23:
You have a Renesas R61509 controller.
The Serial Port should work.   Set baud to 9600.

Analog pins A0-A5 will only be defined if there is a specific Wemos R32 board in boards.txt

Using the Touchscreen needs two Analog pins but not all Analog pins can be used as Digital.

It is not possible diagnose pins on an ARM or ESP32
The Calibration sketch will warn you about this but the "old" diagnose_Touchpins sketch does not tell you.

You need to use a DMM to find the Touch Pins.   Tell me the real answers.   

David.

ghosttk, 2020-05-23:
I can not find my old DMM,maybe need to buy a new one:-(
But I have uno board ,so I run Calbr with uno board on display, this time I can not open serial monitor from IDE(this happens in some projects but others do not)
I take a picture of the display.
[image]

prenticedavid, 2020-05-23:
From mnufriend_shield.h

```
#elif defined(ESP32)       //regular UNO shield on TTGO D1 R32 (ESP32)
#define LCD_RD  2  //LED
#define LCD_WR  4
#define LCD_RS 15  //hard-wired to A2 (GPIO35) 
#define LCD_CS 33  //hard-wired to A3 (GPIO34)
#define LCD_RST 32 //hard-wired to A4 (GPIO36)

#define LCD_D0 12
#define LCD_D1 13
#define LCD_D2 26
#define LCD_D3 25
#define LCD_D4 17
#define LCD_D5 16
#define LCD_D6 27
#define LCD_D7 14

```

So I would expect:

```

//const int XP=8,XM=A2,YP=A3,YM=9; //240x400 ID=0xB509
const int XP=12,XM=15,YP=33,YM=13; //240x400 ID=0xB509
const int TS_LEFT=916,TS_RT=102,TS_TOP=61,TS_BOT=932;
```

But before you go any further,  I would want the screen, Touch and Serial to work 100% on the Uno.
And to see your hardware mod to the Wemos R32 board.

The R32 board has CH340 USB-Serial bridge chip.   Windoze CH340 drivers should work fine.   99% of Chinese Uno clones use CH340.

David.

ghosttk, 2020-05-24:
After plugging in and pulling out cables between board and my laptop，the laptop‘s video card suddenly got broken :(
So I spend hours to find another PC，download IDE and libraries.
My wemos board uses CH340 drivers too, so I installed the driver.
After setting up all the stuffs, I connect the board and run button_simple.ino.
Then I get these errors:

```

Serial port COM6
Connecting........_____....._____....._____....._____....._
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 8c:aa:b5:86:3c:dc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content
```

Sometimes even ”connecting failed“

But running some other examples, every thing goes well.

I think there must be something wrong in esp32 board itself， because wether I use wemos or the original esp32 dev board， this always happens with some ino projects， but not with others。

ghosttk, 2020-05-24:
These are the boards
[image]

ghosttk, 2020-05-25:
Uno board running with no errors, this is the Calibr Serial output:

```
`TouchScreen.h GFX Calibration
Making all control and bus pins INPUT_PULLUP
Typical 30k Analog pullup with corresponding pin
would read low when digital is written LOW
e.g. reads ~25 for 300R X direction
e.g. reads ~30 for 500R Y direction

Testing : (A2, D8) = 20
Testing : (A3, D9) = 33
Diagnosing as:-
XM,XP:  (A2, D8) = 20
YP,YM:  (A3, D9) = 33
ID = 0xB509
TouchScreen.h GFX Calibration
Making all control and bus pins INPUT_PULLUP
Typical 30k Analog pullup with corresponding pin
would read low when digital is written LOW
e.g. reads ~25 for 300R X direction
e.g. reads ~30 for 500R Y direction

Testing : (A2, D8) = 26
Testing : (A3, D9) = 38
Diagnosing as:-
XM,XP:  (A2, D8) = 26
YP,YM:  (A3, D9) = 38
ID = 0xB509

cx=890 cy=71 cz=651 LEFT, TOP, Pressure
cx=884 cy=473 cz=580 LEFT, MIDH, Pressure
cx=877 cy=895 cz=459 LEFT, BOT, Pressure
cx=521 cy=79 cz=576 MIDW, TOP, Pressure
cx=526 cy=911 cz=304 MIDW, BOT, Pressure
cx=160 cy=84 cz=606 RT, TOP, Pressure
cx=157 cy=494 cz=478 RT, MIDH, Pressure
cx=140 cy=901 cz=231 RT, BOT, Pressure
MCUFRIEND_kbv ID=0xB509  240 x 400

const int XP=8,XM=A2,YP=A3,YM=9; //240x400 ID=0xB509
const int TS_LEFT=916,TS_RT=118,TS_TOP=56,TS_BOT=923;
PORTRAIT CALIBRATION     240 x 400
x = map(p.x, LEFT=916, RT=118, 0, 240)
y = map(p.y, TOP=56, BOT=923, 0, 400)
Touch Pin Wiring XP=8 XM=A2 YP=A3 YM=9
LANDSCAPE CALIBRATION    400 x 240
x = map(p.y, LEFT=56, RT=923, 0, 400)
y = map(p.x, TOP=118, BOT=916, 0, 240)
`
```

ghosttk, 2020-05-25:
But even on Uno board, copy the "const int TS_LEFT=916,TS_RT=118,TS_TOP=56,TS_BOT=923;" and run,  touching the two buttons displayed on tft still have no effects.

prenticedavid, 2020-05-25:
The Adafruit TouchScreen.h library expects 10-bit ADC resolution e.g. in TouchShield example.
Add this to setup()
`
    analogReadResolution(10);
`

This should make that example work.

The Calibration sketch "works".   "TouchScreen_kbv.h" converts from 12-bit ADC to 10-bit.
This seems to get bad values from my 3.95 inch ST7796S Shield.   But is fine for 240x320 ILI9320 Shield.

I will investigate a robust fix for ESP32.   Possibly tomorrow or Wednesday.

David.

ghosttk, 2020-05-25:
Adding `analogReadResolution(10)` to Touch_Shield_new.ino Setup(), I got this error:
```
C:\Users\ADMINI~1\AppData\Local\Temp\arduino_modified_sketch_759180\Touch_shield_new.ino:94:5: note: suggested alternative: 'analogRead'

     analogReadResolution(10);

     ^~~~~~~~~~~~~~~~~~~~

     analogRead

exit status 1
'analogReadResolution' was not declared in this scope
```
I could not find where analogReadResolution is declared, it was not in Arduino.h. neither could I find where the "TouchScreen_kbv.h" is, 

From Arduino.cc website , I found analogReadResolution  is not for Uno.
`analogReadResolution() is an extension of the Analog API for the Arduino Due, Arduino and Genuino Zero and MKR1000.`

closed 2021-08-24 by prenticedavid