prenticedavid/MCUFRIEND_kbv #137: help

mokey808, 2020-05-25:
an some one help me i dont know what this error is im trying to port this and i dont know what to do

Arduino: 1.8.1 (Windows 8.1), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\keanu\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\keanu\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\keanu\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path C:\Users\keanu\AppData\Local\Temp\arduino_build_917559 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\keanu\Desktop\tfterminal\tfterminal\tfterminal.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\keanu\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\keanu\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\keanu\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10801 -build-path C:\Users\keanu\AppData\Local\Temp\arduino_build_917559 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\keanu\Desktop\tfterminal\tfterminal\tfterminal.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\keanu\AppData\Local\Temp\arduino_build_917559\sketch\tfterminal.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\keanu\AppData\Local\Temp\arduino_build_917559\sketch\tfterminal.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\keanu\Documents\Arduino\libraries\Adafruit_GFX_Library" "C:\Users\keanu\AppData\Local\Temp\arduino_build_917559\sketch\tfterminal.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\keanu\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\keanu\Documents\Arduino\libraries\MCUFRIEND_kbv" "C:\Users\keanu\AppData\Local\Temp\arduino_build_917559\sketch\tfterminal.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\keanu\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\keanu\Documents\Arduino\libraries\MCUFRIEND_kbv" "C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src\SPI.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\keanu\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\keanu\Documents\Arduino\libraries\MCUFRIEND_kbv" "C:\Users\keanu\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_GFX.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\keanu\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\keanu\Documents\Arduino\libraries\MCUFRIEND_kbv" "C:\Users\keanu\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\keanu\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\keanu\Documents\Arduino\libraries\MCUFRIEND_kbv" "C:\Users\keanu\Documents\Arduino\libraries\Adafruit_GFX_Library\glcdfont.c" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "-IC:\Users\keanu\Documents\Arduino\libraries\Adafruit_GFX_Library" "-IC:\Users\keanu\Documents\Arduino\libraries\MCUFRIEND_kbv" "-IC:\Users\keanu\Documents\Arduino\libraries\MCUFRIEND_kbv\utility" "C:\Users\keanu\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUF [... cut ...]
C:\Users\keanu\Desktop\tfterminal\tfterminal\tfterminal.ino: In function 'void tftPutchar(char)':

tfterminal:72: error: 'x' was not declared in this scope

     x=0;
[... 154 lines cut ...]
  Tft.TFTinit();  

  // Display header and clear terminal
  tftDisplayHeader();
  tftClearTerminal();

  // Initialize USB serial
  Serial.begin(115200);
  
}

void loop()
{
  // If a new char is available on serial link, 
  // Display the char
  if (Serial.available())
  {
    char c=Serial.read();
    tftPutchar(c);
    
    // Uncomment for echoing
    //Serial.print(c);
    
  }
}

prenticedavid, 2020-05-25:
Please run ALL the examples that come with the library.   Report any examples by name if you have a problem.   There is no need to paste the error report unless I ask you to.

Your sketch has many errors.    You should always start with a working sketch that has the correct syntax.

The Arduino  Displays forum is probably a better place for beginner questions. 

David.

mokey808, 2020-05-25:
graphictest_kbv example im using the setup for the lcd on some code i found for a serial terminal using the ili9340 tft lcd im getting a error
 'INT16U' does not name a type i have no clue what it is every example for the screen works but i try to use it in other stuff its a no go

prenticedavid, 2020-05-25:
Go on.    Does the graphictest_kbv example from the library work ?

I don't think that I have ever used 'INT16U'.
If you have a link to working code,  I will show you how to use it with MCUFRIEND_kbv
i.e. post a link to the original code

The Arduino Displays forum is probably a better place for beginner questions.   We can probably point you to examples, tutorials, advice, ...

David.

closed 2021-08-24 by prenticedavid

mokey808, 2021-08-24:
Sorry gave up on it

prenticedavid, 2021-08-24:
It is fairly straightforward.   Beginners should post in Arduino.cc/Displays Forum
1. post a link to the actual display that you have bought.
2. paste, attach or post a link to the original **un-edited** project code
3. post a schematic or description of your wiring.

If the project uses common libraries and hardware,   Forum readers will help you to "port" the original code.

And you will probably learn in the process.

David.