prenticedavid/MCUFRIEND_kbv #139: Changing pins ESP32

fernandodaniele, 2020-06-04:
Hi, I need change the LCD_RS pin and LCD_RST, i tried with the mcufriend_shield.h but it didnt work.

The original configuration:
#define LCD_RS 15  //hard-wired to A2 (GPIO35) 
#define LCD_CS 33  //hard-wired to A3 (GPIO34)
#define LCD_RST 32 //hard-wired to A4 (GPIO36)

I need this configuration
 #define LCD_RS 32  //hard-wired to A2 (GPIO35) 
#define LCD_CS 33  //hard-wired to A3 (GPIO34)
#define LCD_RST 5 //hard-wired to A4 (GPIO36)

is it possible?

prenticedavid, 2020-06-04:
Well,   have you made the necessary hardware mod ?
Have you changed the defines in mcufriend_shield.h ?

David.

fernandodaniele, 2020-06-05:
I included mcufriend_shield.h, I changed the defines in mcufriend_shield.h, then I uploaded the program, but  LCD_RS still works on GPIO15 and LCD_RST still works on GPIO32
What do you mean with hardware mod ? I re-wired the pins according the defines, but its like the changes on mcufriend_shield.h have no effect

prenticedavid, 2020-06-05:
The Shield is designed to mate with the TTGO R32 board.   (after you have completed the recommended hardware mod)

You should be able to select other control pins (but they must be capable of Digital)

I do NOT advise Dupont wires.   The ESP32 will probably not be stable with random wires.

David.

hervema, 2020-12-01:
Hello, I have the same concern about another kind of ESP32 card (https://www.az-delivery.de/en/products/esp32-developmentboard).. Changing something in this file, does not work. What is the way to achieve a custom pins support, PLEASE. Thanks in advance.

prenticedavid, 2020-12-01:
I advise you to use the TTGO R32 board with the recommended hardware mod.

There is little point in using a Uno Display Shield unless you have Arduino header sockets to plug the Shield into.

David.

hervema, 2020-12-01:
This change my project :-( By the way, what do mean by "**hardware mod**" ?

Hervé

hervema, 2020-12-02:
Hello. 

To move on I got a TTGO R32 board, but nothing more works well :-(

I'm just trying to use the TouchScreen_Calibr_native.ino but nothing work as pre-configured.

how I can find the correct values for XP, YP, XM, YP ?

The reconfigured values are not fine 
// MCUFRIEND UNO shield shares pins with the TFT.
#if defined(ESP32)
**int XP = 27, YP = 4, XM = 15, YM = 14;  //most common configuration**
#else
//int XP = 6, YP = A1, XM = A2, YM = 7;  //most common configuration
int XP = 7, YP = A2, XM = A1, YM = 6;  //next common configuration

Same question/problem with this #define list in mcufriend_shield.h. How can I found the correct input ? :-(

#elif defined(ESP32)       //regular UNO shield on TTGO D1 R32 (ESP32)
#define LCD_RD  2  //LED
#define LCD_WR  4
#define LCD_RS 15  //hard-wired to A2 (GPIO35) 
#define LCD_CS 33  //hard-wired to A3 (GPIO34)
#define LCD_RST 32 //hard-wired to A4 (GPIO36)

#define LCD_D0 12
#define LCD_D1 13
#define LCD_D2 26
#define LCD_D3 25
#define LCD_D4 17
#define LCD_D5 16
#define LCD_D6 27
#define LCD_D7 14

Thanks for your help.

Regards,

Hervema

prenticedavid, 2020-12-02:
The hardware mod for the TTGO D1 R32 board is to solder GPIO15 to GPIO35 etc.  as described in those defines.

You can identify the Touch pins either by running the Calibration sketch on a Uno or by measuring resistance with a DMM.

I don't know what display you own.   3.5 inch Blue pcb and 2.8inch Blue pcb tend to use D8, D9 pins instead of D6, D7 pins.

David.

hervema, 2020-12-02:
Find attached for more info D1 R32 and screen I use :-)

Could you please point the link explaining the hardware modification to do, "as described in those defines" is not clear for me :-( and I don't really find any relevant clues to progress :-(

[image]

[image]

Thanks,

Hervema

prenticedavid, 2020-12-02:
You will see IO15, IO33,  IO32 near the 3V3 pin.
And GPIO35, GPIO34, GPIO36 (i.e. A2, A3, A4)

Turn the D1 R32 upside down.   You can solder the connections with some fine wires.

Regarding the Touch pins.   Run Calibration on a Uno, Mega, ...

David.

hervema, 2020-12-02:
I've found that on https://github.com/Bodmer/TFT_eSPI, that's what you are talking about hardware modification, right ?

[image]

prenticedavid, 2020-12-02:
Yes.   If you follow his instructions TFT_eSPI will work with several Parallel Displays.

What controller ID is mounted on your Display ?
e.g. run any examples on Uno, Mega, ...

David.

hervema, 2020-12-02:
TouchScreen.h GFX Calibration returns ID = 0x9486

I did hardware mod and I progressed a lot ! :-) 

[image]

The screen shows up ! 

[image]

Now next problem is "touch". Touch is working but only this part of screen is sensible / active ? Probably some incorrect pins number in code ? I use the default in your program : int XP = 27, YP = 4, XM = 15, YM = 14;  //most common configuration

[image]

Hervema

prenticedavid, 2020-12-02:
Try to touch the cross-hairs.   If nothing happens touch "near" the cross-hair.

ARM and ESP are not really equipped to drive Resistive Touch panels.
They "work" but they exceed the pin current ratings from the datasheet.

David.

hervema, 2020-12-02:
I can touch cross-hair or near but only on top left .. It's somewhat strange that only a part of screen is working, no ?

I attached a quick video :-)

[VID_20201202_222115.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/5632120/VID_20201202_222115.zip)

hervema

prenticedavid, 2020-12-02:
You did not try very hard.   I don't believe that the Touch only responds to the Top Left Hand Corner.

I don't believe that you have an ILI9486.   Your photo looks like a 2.8 inch Blue Mcufriend Shield that is normally 240x320.   The ILI9486 is 320x480.

Life is much easier with a Uno.   But once you have calibrated on the Uno,  most things should work on the ESP32.

David.

hervema, 2020-12-03:
I tried with success on a UNO (buy my project need wifi) and I didn't have to press hard to working well. 

The screen is this one : https://uk.banggood.com/2_8-Inch-TFT-LCD-Shield-Touch-Display-Screen-Module-Geekcreit-for-Arduino-products-that-work-with-official-Arduino-boards-p-989697.html?rmmds=search&cur_warehouse=CN and effectively it's a ILI9341. Is that change something ?

To which on screen are connected/linked XP, XM, YP, YM ?

const int XP = 27, XM = 4, YP = 15, YM = 14; //240x320 ID=0x9341
const int TS_LEFT = 909, TS_RT = 124, TS_TOP = 84, TS_BOT = 912;
TouchScreen ts = TouchScreen(XP, YP, XM, YM, 300);

Thanks

Hervema

prenticedavid, 2020-12-03:
I don't believe you.

```
const int XP = 27, XM = 4, YP = 15, YM = 14; //240x320 ID=0x9341
const int TS_LEFT = 909, TS_RT = 124, TS_TOP = 84, TS_BOT = 912;

```
does not look like a Uno result.
Please paste the actual result from the Uno Calibration.

Your link says:

> 4-wire resistive touchscreen 240 x 320 pixels with individual pixel control
No wiring, no soldering.
Simply plug it in and load library
On-board 3.3V 300mA LDO regulator

Does the pcb really look like the photo in the link?   i.e. with AMS1117 regulator and R1=10k
Does `tft.readID()` say 0x9341 ?

David.

hervema, 2020-12-03:
oups.. sorry a bad copy/paste among all tests I did !

Here is the actual result from the uno Calibration (and a video to believe me ;-) ) :

[VID_20201203_095230.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/5634932/VID_20201203_095230.zip)

TouchScreen.h GFX Calibration
Making all control and bus pins INPUT_PULLUP
Typical 30k Analog pullup with corresponding pin
would read low when digital is written LOW
e.g. reads ~25 for 300R X direction
e.g. reads ~30 for 500R Y direction

Testing : (A2, D8) = 25
Testing : (A3, D9) = 32
ID = 0x9341

cx=851 cy=142 cz=608  X, Y, Pressure
cx=853 cy=487 cz=481  X, Y, Pressure
cx=856 cy=866 cz=386  X, Y, Pressure
cx=519 cy=130 cz=555  X, Y, Pressure
cx=530 cy=876 cz=261  X, Y, Pressure
cx=158 cy=116 cz=517  X, Y, Pressure
cx=170 cy=475 cz=371  X, Y, Pressure
cx=168 cy=831 cz=214  X, Y, Pressure

*** COPY-PASTE from Serial Terminal:
const int XP=8,XM=A2,YP=A3,YM=9; //240x320 ID=0x9341
const int TS_LEFT=884,TS_RT=133,TS_TOP=104,TS_BOT=881;

PORTRAIT  CALIBRATION     240 x 320
x = map(p.x, LEFT=884, RT=133, 0, 240)
y = map(p.y, TOP=104, BOT=881, 0, 320)

LANDSCAPE CALIBRATION     320 x 240
x = map(p.y, LEFT=104, RT=881, 0, 320)
y = map(p.x, TOP=133, BOT=884, 0, 240)

[image]

and yes it's a same pcb with an AMS1117 and a R1==10k.

[image]

Hervema

prenticedavid, 2020-12-03:
As shown by the Uno Calibration,  your shield uses:
XP=8,XM=A2,YP=A3,YM=9

The equivalent for ESP32 would be:
XP=12,XM=15,YP=33,YM=13

Or if I swap XM, YP with XP,YM I can use:
XP=15,XM=12,YP=13,YM=33

The Calibration sketch worked fine on my ESP32 with a Blue 3.5 inch Shield (using the default local Touchscreen_kbv.h class)
```
const int XP=15,XM=12,YP=13,YM=33; //320x480 ID=0x7796
const int TS_LEFT=107,TS_RT=892,TS_TOP=935,TS_BOT=5;
```

Unfortunately the Adafruit Touchscreen.h does not work.   So this means you would need to use Touchscreen_kbv.h in your sketches.

Either store a local Touchscreen.h and Touchscreen_kbv.cpp in your Sketch tabs.
Or create a Public library.

I tested with:
```

#define USE_LOCAL_KBV 1
#if USE_LOCAL_KBV
#include "TouchScreen_kbv.h"         //my hacked version
#define TouchScreen TouchScreen_kbv
#define TSPoint     TSPoint_kbv
#else
#include <TouchScreen.h>         //Adafruit Library
#endif

```

Good news about your Blue Shield.   It contains AMS1117 and proper LVC245 buffers.

David.

hervema, 2020-12-03:
With these last information I able to run a project with all functionnalities ! Yes !!! Thanks for your support 👍 

A video showing the result, it's almost perfect, need some ajustements : 

[VID_20201203_194303.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/5638312/VID_20201203_194303.zip)

Regards,

hervema

hervema, 2020-12-05:
Hello.

For information.

I've move forward on my project and a new, big, problem occurred.

In fact, what I was not aware, the ESP32 (even on 8266), when you activate WIFI (I need it, and the reason I go to ESP 32), all ADC2 pins are not usable !! and this, of course, impact the touch feature of the screen ! :-(

If you have a look on community (https://github.com/espressif/arduino-esp32/issues/102), this problem is not new, and any of workarounds are not in fact, and none works for using a such screen.

I'm so disappointing. That's stop my project.

hervema, 2020-12-07:
Hello.

Finally I able to use touch and wifi, by using these approach :

// ADC2 restoring procedure
// This is a Workaround to use ADC2 Pins on ESP32 when Wifi or Bluetooth is on.
// (usually only ADC1 Pins are usable for analogRead() when Wifi or Bluetooth is on.)
// -- We save the ADC2 control register before WifiBegin() or BluetoothBegin()
// -- then restore its original value, then set a specific bit of the ADC2 control register   
//to avoid inverted readings: we do the latter two before every analogRead() cycle.
// -- This achieves ADC2 usability even when Wifi/Bluetooth are turned on!

#include "soc/sens_reg.h" // needed for manipulating ADC2 control register
uint64_t reg_b; // Used to store ADC2 control register
#define PIN xx // Substitute xx with your ADC2 Pin number
int value;

void setup() {
// Save ADC2 control register value : Do this before begin Wifi/Bluetooth
reg_b = READ_PERI_REG(SENS_SAR_READ_CTRL2_REG);
Wifi.Begin(); // or similar wifi init function or Bluetooth begin()
}

void loop() {
// ADC2 control register restoring
WRITE_PERI_REG(SENS_SAR_READ_CTRL2_REG, reg_b);
//VERY IMPORTANT: DO THIS TO NOT HAVE INVERTED VALUES!
SET_PERI_REG_MASK(SENS_SAR_READ_CTRL2_REG, SENS_SAR2_DATA_INV);
//We have to do the 2 previous instructions BEFORE EVERY analogRead() calling!
value = analogRead(PIN);
delay(1);
}

This is working correctly on my configuration ! Hope this help other.

hervema

hervema, 2020-12-07:
David, is it possible to use the SD card of the screen in my configuration ?

if yes, which pins I need to use ?

Thanks again,

regards,

Hervema

closed 2021-08-24 by prenticedavid