prenticedavid/MCUFRIEND_kbv #141: Increase screen refresh rate?

RedRhino4153, 2020-06-16:
I am using a 2.8" TFT on an Uno with the showBMP_kbv_Uno sketch and I want to increase the refresh rate of the screen so I do not see the black line swipe.  I would like the images to display in full on the screen or have the swipe be very fast.  I realize you cannot play a video file so I have broken one up into its elemental frames and want to display them quickly so as to give the illusion of an animation.

On line 79, I have changed delay to 0

                    Serial.print(millis() - start);
                    Serial.println(F("ms"));
                    delay(0);

Can I increase the buffer size or is there anything I can do to achieve this objective?

Thanks!

prenticedavid, 2020-06-16:
The example is designed to show BMP files that use different formats.

If you only use one format or only use known files there are several tricks.

1.  what size e.g. 240x320 ?
2.  full colour or monochrome ?
3.  how many images ?
4.  what speed ?

It would be better to ask this sort of question on the Arduino.cc Displays Forum.

David.

RedRhino4153, 2020-06-17:
David,
Yes, 24bit windows format bmp, 320x240, full color, about 200 images.
I have set the refresh to 0.

I'll check out the forums.  I was just wondering if there were any tweaks I could do to your sketch to speed things up.
Thanks,
Ryan

closed 2021-08-24 by prenticedavid