prenticedavid/MCUFRIEND_kbv #144: R61509V and Arduino DUE

ThoSanLinhHon, 2020-07-04:
Hi, David!
I use your library for DUE and HX8357 communication successfully, but when I switch to R61509V, DUE can't read ID (always show 0x0) and my LCD is always white, sometimes DUE read ID B509 and my LCD flashes on the color bands that I think are first frames then it is white.
I think my LCD has been suspended due to high communication speed.
Do you have any solution?
My LCD is 3.0 inch, using driver IC R61509V, connect pins by default.

prenticedavid, 2020-07-04:
I do not possess a R61509V shield any more.   So  I can not test R61509V on a Due.

Are you using the v2.9.8Release or the current Beta (i.e. master branch) ?

There are three main things to check in graphictest_kbv.ino :
1.   reliable readID().    Force tft.begin(0xB509).    Observe ID value on Serial.  Observe ID value on Adafruit Tests Report page.

2.  observe Adafruit Tests,  Penguin pages.   Any glitches ?  

3.  observe SOFTWARE SCROLL.   Any glitches ?

Report back with a clear description of where the glitch occurs e.g. filled rounded rectangle.  e.g. BAND SCROLL.

Note that Vertical Scroll only works on whole screen.   And you will see a White area come into view.

David.

ThoSanLinhHon, 2020-07-04:
I use the v2.9.7, and test LCD with graphictest_kbv.ino.
1. On Serial, it shows either 0x0 or 0xB509 every time I reset DUE
2. And when it shows 0xB509, my LCD will flash on (the color bands or the whole screen is black) and immediately, it turned off and the LCD turned white.
3. My lcd has never been displayed there.

prenticedavid, 2020-07-04:
Please install v2.9.8 via the Library Manager.

Report back.

ThoSanLinhHon, 2020-07-04:
it ran for longer but then still got a white screen, it even flickered

prenticedavid, 2020-07-04:
> My LCD is 3.0 inch, using driver IC R61509V, connect pins by default.

That does not sound like a "mcufriend shield"
Please post a link to the actual display that you bought.  e.g. Ebay sale page.
If it is not a shield,  please post a photo of your wiring.

David.

ThoSanLinhHon, 2020-07-04:
it is breakboard, i connect like mcufriend shield
https://aliexpress.ru/item/32897509943.html
LCD---------------------DUE
D0-----------------------8
D1-----------------------9
D2-----------------------2
D3-----------------------3
D4-----------------------4
D5-----------------------5
D6-----------------------6
D7-----------------------7

RST----------------------A4
RD----------------------A0
WR----------------------A1
RS----------------------A2
CS----------------------A3

prenticedavid, 2020-07-04:
The Open-Smart board should work fine on the Due (because the Due has 3.3V logic.)

I suggest that you use short wires.   I use a Protoshield for this type of board.   With a soldered header socket and wired routed to pins A0-A4, 8-9, 2-7.

Seriously.     A breadboard is a bad idea (tm).
Please post a photo.    Or use reliable wiring.

When the wiring is improved,   I can show you how to adjust timing.    There is no point in attempting any software until the hardware is reliable.

David.

ThoSanLinhHon, 2020-07-04:
Thank you!
I will try it later. Probably due to long wires (8 inches) and poor quality (no welding).
Thanks to you to remind me, I remembered that.
Thank you again!

prenticedavid, 2020-07-04:
Plug the display into your breadboard.
Then connect 100mm jumper cables to Due sockets: A0-A4, 2-9, GND, 3.3V

200mm wires are far too long.

Verify with LCD_ID_readreg sketch.
Then I will show you how to "slow down" the Due.

David.

ThoSanLinhHon, 2020-07-07:
It still doesn't work.  I used 10cm wires and soldered them together but the situation still happens.  I ordered 2 other LCDs and still failed

ThoSanLinhHon, 2020-07-07:
I will record a video and send you a link

prenticedavid, 2020-07-07:
The video shows that your R61509V is not working.

Please post a clear photo of your wiring.

I only have a 3.3V Open-Smart with ILI9225 controller.  
I do not have a R61509V shield now.

```
ILI9225  tWC =100ns  tWRH = 50ns  tRCFM = 300ns  tRC = 300ns
R61509V  tWC =75ns  tWRH = 25ns  tRCFM = 450ns  tRC = 450ns (Rev0.11 all figures TBD)
R61509V  tWC =110ns  tWRH = 30ns  tRCFM = 450ns  tRC = 450ns (Rev1.03)
SPFD5408 tWC =125ns  tWRH = 70ns  tRCFM = 450ns  tRC = 450ns

```

The ILI9225 display is running fine on my 84MHz Due.   It has a better timing spec than your R61509.
Only the SPFD5408 has a slower write cycle.

Note that the earlier Rev0.11 datasheet showed faster "estimated" cycle times. 

When I see your wiring I will show you how to adjust timing on your Due.    But quite honestly,   I would expect the R61509V to work if there is good wiring.

David.

ThoSanLinhHon, 2020-07-08:
[image]

ThoSanLinhHon, 2020-07-08:
I use a multimeter to check the continuity after welding. it's ok

prenticedavid, 2020-07-08:
It seems rather extreme to solder directly to the Due.    You will not get anything better !!

You have never said what library version you are using.
The IDE Library Manager will install v2.9.8 Release.  (until yesterday)
Now the IDE Library Manager will install v2.9.9 Release.

I suggest that you leave the IDE,  delete any existing library directory.
Start the IDE.   **Install v2.9.9 via the IDE Library Manager.**

If the R61509V still crashes,  edit utility/mcufriend_shield.h
from

```
//####################################### DUE ############################
#elif defined(__SAM3X8E__)      //regular UNO shield on DUE
#define WRITE_DELAY { WR_ACTIVE; }
#define IDLE_DELAY  { WR_IDLE; }
#define READ_DELAY  { RD_ACTIVE;}
```
to

```
//####################################### DUE ############################
#elif defined(__SAM3X8E__)      //regular UNO shield on DUE
#define WRITE_DELAY { WR_ACTIVE8; }
#define IDLE_DELAY  { WR_IDLE4; }
#define READ_DELAY  { RD_ACTIVE16;}
```

If graphictest_kbv.ino still crashes,   I conclude that your R61509V is faulty.

David.

ThoSanLinhHon, 2020-07-08:
sorry!! I installed version 2.9.8 from the IDE Library Manager.
I will update to v2.9.9 tonight and try again.
Thanks for your support!

prenticedavid, 2020-07-08:
If your v2.9.8 was installed by the IDE Library Manager it will Update happily to v2.9.9 i.e. without you deleting directories etc.

However,   the leave IDE, delete, start IDE, install procedure will ensure that you have a clean install.

Which HX8357 controller do you have?
What ID is reported? 0x8357, 0x9090, 0x0099

David.

ThoSanLinhHon, 2020-07-08:
https://www.aliexpress.com/i/32956967893.html
I ordered this model.
Run graphictest_kbv.ino with DUE, it displays ID 0x8357
It runs very smoothly, but when it comes to the scrolling screen, it flickers slightly.

prenticedavid, 2020-07-08:
Your link shows a pcb with a R61581 controller (HX8357 is not selected)

A genuine HX8357B will return ID = 0x8357

LCD_ID_readreg.ino should show:

`
reg(0x00BF) 00 01 62 83 57 FF   Device Code Read
`

If you have the R61581 I would be very interested.

David.

ThoSanLinhHon, 2020-07-08:
Depending on the shipment, I ordered 2 times and both times are HX8357
I asked the seller about ordering the product using R61581 IC but they reported that the stock was currently out of stock. I don't know when they will import new shipments

ThoSanLinhHon, 2020-07-08:
> It seems rather extreme to solder directly to the Due. You will not get anything better !!
>
> You have never said what library version you are using.
> The IDE Library Manager will install v2.9.8 [...]

Do i need to convert #if 0 to #if 1 in utility/mcufriend_shield.h

ThoSanLinhHon, 2020-07-08:
when I converted #if 0 to #if 1, IDE reported a lot of errors and couldn't compile

prenticedavid, 2020-07-08:
No,  you just use ctrl-F to search for `### DUE ###`

Then change the DELAY statements.
Save the file.

Build and run your sketches in the normal way.
The IDE will notice that the  `utility/mcufriend_shield.h` file has changed.   It will rebuild the library.

David.

ThoSanLinhHon, 2020-07-08:
My LCD is working. You are great.
However I think I should switch to using MEGA and adding voltage converter module.
Thanks for your help. I will close this issue.
Thank David!

closed 2020-07-08 by ThoSanLinhHon

prenticedavid, 2020-07-08:
It is thirteen level shifters.    More trouble than it is worth.

I would stick with Due, Zero, Teensy3.2, STM32, ... which are all 3.3V logic.

Please experiment with those delays.  Find what values cause glitches.

David.