prenticedavid/MCUFRIEND_kbv #146: Change data bus pins

erfanjahanshahlo, 2020-07-19:
Hello. I want to connect the data pin 1 that is connected to the digital pin 9 to the digital pin 0 or rx because I only need one pwm pin and I can not connect the whole d0-d7 to d0 to d7 Ardino because pcb I have it pre-designed and I just want to connect data bus 1 to pin 0 of the Arduino and I have to scratch part of the pcb. If a function was defined so that all data bus pins could be specified before begin, it would be very, very .... best. I'm in a hurry, please answer faster. Sorry for my bad english. Thank you in advance.

prenticedavid, 2020-07-20:
The shield on a Uno uses digital #8,#9, #2 - #7 for the data bus.
The Bootloader on a Uno uses Serial pins #0, #1 

If you try to use #0 for anything else it will make the Bootloader and Serial very difficult.

If you have a custom (non-Arduino) board please say so.
There are PWM pins on PORTD and PORTB on mega328 and mega324.
The SPI bus is on PORTB

So there are several choices if you are not using the microSD card.

David.

erfanjahanshahlo, 2020-07-20:
Thank you very much for your answer. Because I use a designed PCB, it is not possible to use the Arduino uno directly, so my board is not an Arduino and it is only a microcontroller, and I also used the sd, so there is no option. Thank you in advance.

prenticedavid, 2020-07-20:
Say which AVR.   Or better still,  post a schematic.

In reply to your comment on Arduino.cc.   The data bus and control pins are determined at compile-time.    The performance would be crippled by run-time configuration.

Yes,  other libraries let you change control pins in the constructor.
No libraries ever let you configure data bus at run-time.

Think about it.   If you use a shield,  all the wiring is fixed.   i.e. it is known at compile-time.
Likewise,   the wiring is fixed on any custom pcb.    If you tell me the wiring,  I will post the appropriate SPECIAL.

David.

erfanjahanshahlo, 2020-07-20:
Atmega328, i use [minicore](https://github.com/MCUdude/MiniCore) board package and wiring is quite similar to putting an LCD on an Arduino uno, except that it happens on a PCB. pcb photo is attached
[image]

.

erfanjahanshahlo, 2020-07-20:
It works perfectly on this pcb but I no longer have any pwm pins.

prenticedavid, 2020-07-20:
I can read a schematic.
I can read a list of pins.
Or just a list of requirements.
I am NOT going to study a pcb layout.

If using Serial, TFT, SD you could put one of the data bus pins on PC5.
If you are not using Serial,  you could put the whole data bus on PD0-PD7 (which is much faster than a regular Uno shield)

David.

erfanjahanshahlo, 2020-07-20:
Thank you very much for the idea you gave me. I connect the sd card ss pin to pin a5 and the 10 digital pin is released. Thank you very much for your time

closed 2020-07-20 by erfanjahanshahlo