prenticedavid/MCUFRIEND_kbv #147: "New" Banggood Screen? 2.8":

MBfromOK, 2020-07-20:
This display functions for scrolling (GraphicsTest_kbv), but not for static text and some drawing tests

I am trying to provide all the relevant details up front, my apologies if I missed something / provide extra "junk" data.

## Board is labeled: 
2.8" TFT LCD Shield
(RM68090)

## Purchase Link:
[UNO R3 Improved + 2.8TFT LCD Touch + 2.4TFT Touch](https://www.banggood.com/Geekcreit-UNO-R3-Improved-Version-2_8TFT-LCD-Touch-Screen-2_4TFT-Touch-Screen-Display-Module-Kit-p-1428291.html?cur_warehouse=USA)

## Drivers offered for download (2.8" model):
[68090](http://myosuploads3.banggood.com/products/20191212/201912120038082.8-driver-68090uno.rar)

- ReadReg:
```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 68 09	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 00 00 00	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 00 00	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

```

- GraphicTest_kb prints: ``ID = 0x6809``
but then most of the tests = include flashing white screen or text & artwork in just the top 1/8 or 1/16 of the screen... Strangely (to me) the scrolling demo shows fine from the initial draw through the conclusion (only issue I saw was the color grades not rendering).

- TouchShield_new: 
ID=0x6809
Screen is 240x320
Calibration is: 
LEFT = 907 RT  = 136
TOP  = 942 BOT = 139
Wiring is always PORTRAIT
YP=15 XM=16
YM=7 XP=6

MBfromOK, 2020-07-20:
FWIW: The version I received has 5 static "icons" outside the backlit area & opposite the Arduino's USB connector: Home, Book, Envelope, Camera, Monitor / Screen

Bonus points: include instructions / code for them to be usable when the calibration is complete.

MBfromOK, 2020-07-20:
Hopefully i can provide the data needed from [Arduino.cc forum post #34](https://forum.arduino.cc/index.php?action=post;quote=2560696;topic=366304.30;last_msg=4680628) (if not already provided).

prenticedavid, 2020-07-20:
It looks as if Banggood sell Uno + Red shield + Blue shield as a special deal.

Banggood comments imply that the Blue shield previously contained HX8347D controller.
I suspect that the Blue shield might have other controllers.
Does your Red shield contain ILI9341 ?
Does your Blue shield say 68090 on the pcb ?

Photos of Red and Blue pcbs in the link show "crap" pcb with no AMS1117 regulator and no backlight resistor.    And I suspect both have HC245 instead of LVC245 buffers.

I do not have your Blue RM68090 shield.
The library begin(0x6809) shares code with ILI9320.   A specific initialisation might work better.

But I would expect all my examples to work.   (note that this controller can not Band scroll)
Please describe problems _accurately_ in text.   Or better still in a video of graphictest_kbv 

David.

MBfromOK, 2020-07-20:
All data included above was specific to the blue shield, I had posted an [Arduino forum message](https://forum.arduino.cc/index.php?action=post;quote=4680628;topic=366304.3060;last_msg=4680797) on the red shield, but then discovered your GitHub and decided to post here to better the library for all (if possible).

As I have learned post purchase (reading your replies regarding various screens), these claim to be 5v logic compatible / ready, but this seems to be based solely on the 5R1 resistor soldered on the underside of the boards.

You are correct there is not a AMS1117 on either of my boards (screenshots below)

(1) The red shield:
- is definitely a ILI9341
- is a 6+6+8+8 screen shield
- your script(s) return a tft.readID() of  0xD3D3
- has the 5R1 resistor touted as allowing 5v logic
- contains 2x: SM245TS (CRAWRYAD42)s @ U2 & U3
- U5 = empty
- [ILI3491 ReadReg](https://pastebin.com/jXuZaYLL)

(2) The Blue shield:
- see data above
- says RM68090 (which I assume means the controller is 68090)
- your script(s) return tft.readID() of 0x6809
- is a 6+6+8+8 screen shield
- has the 5R1 resistor touted as allowing 5v logic
- contains 2x: SM245TS (CRAWRYAD42)s @ U2 & U3

[Board Screenshots](https://imgur.com/a/Cy2DCQX)

I will review this later and add any requested information still missing.

prenticedavid, 2020-07-20:
Where is this 5R1 resistor ?
I can't see a resistor on the Banggood photos.

I can't find a datasheet for SM245TS but a Chinese broker provides a 74HC245 datasheet.

Your Blue shield definitely has a Raydium RM68090 controller.   If you can provide photos, video or accurate written description of any problem,  I would like to solve it.

Your Red board has **NOT** got an Ilitek ILI9341 controller.
Please install the current v2.9.9 Release.    Then I can ask you to run some diagnostics that might identify your Red controller.

David.

MBfromOK, 2020-07-21:
Uh, did you see this: 5R1 = R1 on the Board Back Photo [Board Screenshots](https://imgur.com/a/Cy2DCQX) ?
and the register link I also included above: [Red Board: ReadReg ](https://pastebin.com/jXuZaYLL)

OK = Oklahoma :-)

Micah

prenticedavid, 2020-07-21:
Oops.   I saw the front sides but did not notice the pcb views below.   (it was off my screen)
The 5R1 will limit the LED backlight current.  e.g.  59mA if the forward voltage is 3.0V,  98mA if forward voltage is 2.8V

Either way,   the backlight current exceeds the specified limit for a Uno or Mega's 3.3V pin.
I suspect that your screen backlight draws about 55mA from the 3.3V pin (which is ok)
I don't worry about the 2.4 inch "Mcufriend pcb".    But the 3.5 inch "Mcufriend pcb" draw much higher currents from the 3.3V pin.

Regarding the readreg report:   Only these registers contain data.

```
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 06 06	Get Pixel Format
reg(0x00BF) FF FF FF FF FF FF	ILI9481, HX8357-B
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
```

Please run the LCD_ID_readnew.ino example.
You can read every register by un-commenting the  for() loop statement in setup()
You might see some real data in some other registers.

Then try again but with un-commenting the `unlock = unlock_1520;` statement

Then try again but with un-commenting the `unlock = unlock_1526;` statement

There are several "manufacturer lock" mechanisms.  With luck,  you might hit the jackpot.
Please paste the full output if you get something.

However,  you are a genuine BangGood customer.    You should be able to quote your invoice number and request information about your Red board.   i.e. part number of controller.

As you have noticed,   shop websites often provide wrong or misleading information.    Banggood do try to help their customers.   e.g. saying RM68090.

David.

MBfromOK, 2020-07-21:
Ok, I took some video of the operation of these two, just not sure where / how to post them :-p

Additionally, I have been in desktop computing for 20 years and never learned to read HEX / Binary... Am I looking for 00's eg:
```
reg(0x00DA) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x00DB) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x00DC) 00 00 00 00 00 00 00 00 00 00	f.k
```
or just column values values that change somewhere between column 0 and column 9?

```
      reg(0x0009) 00 00 61 00 00 00 00 00 00 00	f.k
      reg(0x002E) 00 FC FC FC 00 00 00 00 00 00	f.k
```

## Red Board Read_New:
```
Read Special Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

diagnose any controller
reg(0x0000) C0 C0	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 06 06	Get Pixel Format
reg(0x0030) F0 F0 F0 F0 F0	PTLAR
reg(0x0033) F3 F3 F3 F3 F3 F3 F3	VSCRLDEF
reg(0x0061) E1 E1	RDID1 HX8347-G
reg(0x0062) E2 E2	RDID2 HX8347-G
reg(0x0063) E3 E3	RDID3 HX8347-G
reg(0x0064) E4 E4	RDID1 HX8347-A
reg(0x0065) E5 E5	RDID2 HX8347-A
reg(0x0066) E6 E6	RDID3 HX8347-A
reg(0x0067) E7 E7	RDID Himax HX8347-A
reg(0x0070) F0 F0	Panel Himax HX8347-A
reg(0x00A1) E1 E1 E1 E1 E1	RD_DDB SSD1963
reg(0x00B0) F0 F0	RGB Interface Signal Control
reg(0x00B3) F3 F3 F3 F3 F3	Frame Memory
reg(0x00B4) F4 F4	Frame Mode
reg(0x00B6) F6 F6 F6 F6 F6	Display Control
reg(0x00B7) F7 F7	Entry Mode Set
reg(0x00BF) FF FF FF FF FF FF	ILI9481, HX8357-B
reg(0x00C0) C0 C0 C0 C0 C0 C0 C0 C0 C0	Panel Control
reg(0x00C1) C1 C1 C1 C1	Display Timing
reg(0x00C5) C5 C5	Frame Rate
reg(0x00C8) C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8	GAMMA
reg(0x00CC) CC CC	Panel Control
reg(0x00D0) D0 D0 D0 D0	Power Control
reg(0x00D1) D1 D1 D1 D1	VCOM Control
reg(0x00D2) D2 D2 D2	Power Normal
reg(0x00D3) D3 D3 D3 D3	ILI9341, ILI9488
reg(0x00D4) D4 D4 D4 D4	Novatek
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0	GAMMA-P
reg(0x00E1) E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1	GAMMA-N
reg(0x00EF) EF EF EF EF EF EF	ILI9327
reg(0x00F2) F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2	Adjust Control 2
reg(0x00F6) F6 F6 F6 F6	Interface Control
```

prenticedavid, 2020-07-21:
You are looking for something different.  e.g. different values in the same line.

If you look at the current non-values.   They are `reg | 0xC0` e.g. `reg(0x67)` is `F7 F7 ...`
The reason is simple.   Reading a non-existent register returns the register.   Since the Touch panel is on bit6 and bit7,  they will read high.   Hence `reg | 0xC0`

I suggested that you start with no "unlock" but display the whole 255 registers.
Then try some unlock keys.

But you could equally well start with the "regular register list" and try those two unlock keys.
Look for different reg(0xBF) or reg(0xD3)

It still needs a human input to look at the 255 garbage lines. at some stage.    Hey-ho.   You can just un-comment the for() statement and paste the results.    Then I can view all the garbage.

If you are not happy,  I can create "test_Micah" branch on GitHub with everything done for you.

David.

MBfromOK, 2020-07-21:
I don't mind doing the work at all, that said I don't want to waste your time by misreporting, or not reporting (simply because I can't be aware of what I don't know)

Based on the information above that would mean that:

- Not initialized:

```
reg(0x0000) C0 C0 C0 C0 C0 C0 C0 C0 C0 C0	f.k
```
- Data set (low)?
```
reg(0x0004) 00 00 00 00 00 00 00 00 00 00	f.k
```
- Data set (high)?
```
reg(0x0009) 00 00 61 00 00 00 00 00 00 00	f.k
```

If not worth the trouble here is:
## Full Registers: Red Shield - 
```
Read Special Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) C0 C0 C0 C0 C0 C0 C0 C0 C0 C0	f.k
reg(0x0001) C1 C1 C1 C1 C1 C1 C1 C1 C1 C1	f.k
reg(0x0002) C2 C2 C2 C2 C2 C2 C2 C2 C2 C2	f.k
reg(0x0003) C3 C3 C3 C3 C3 C3 C3 C3 C3 C3	f.k
reg(0x0004) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x0005) C5 C5 C5 C5 C5 C5 C5 C5 C5 C5	f.k
reg(0x0006) C6 C6 C6 C6 C6 C6 C6 C6 C6 C6	f.k
reg(0x0007) C7 C7 C7 C7 C7 C7 C7 C7 C7 C7	f.k
reg(0x0008) C8 C8 C8 C8 C8 C8 C8 C8 C8 C8	f.k
reg(0x0009) 00 00 61 00 00 00 00 00 00 00	f.k
reg(0x000A) 08 08 08 08 08 08 08 08 08 08	f.k
reg(0x000B) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x000C) 06 06 06 06 06 06 06 06 06 06	f.k
reg(0x000D) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x000E) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x000F) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x0010) D0 D0 D0 D0 D0 D0 D0 D0 D0 D0	f.k
reg(0x0011) D1 D1 D1 D1 D1 D1 D1 D1 D1 D1	f.k
reg(0x0012) D2 D2 D2 D2 D2 D2 D2 D2 D2 D2	f.k
reg(0x0013) D3 D3 D3 D3 D3 D3 D3 D3 D3 D3	f.k
reg(0x0014) D4 D4 D4 D4 D4 D4 D4 D4 D4 D4	f.k
reg(0x0015) D5 D5 D5 D5 D5 D5 D5 D5 D5 D5	f.k
reg(0x0016) D6 D6 D6 D6 D6 D6 D6 D6 D6 D6	f.k
reg(0x0017) D7 D7 D7 D7 D7 D7 D7 D7 D7 D7	f.k
reg(0x0018) D8 D8 D8 D8 D8 D8 D8 D8 D8 D8	f.k
reg(0x0019) D9 D9 D9 D9 D9 D9 D9 D9 D9 D9	f.k
reg(0x001A) DA DA DA DA DA DA DA DA DA DA	f.k
reg(0x001B) DB DB DB DB DB DB DB DB DB DB	f.k
reg(0x001C) DC DC DC DC DC DC DC DC DC DC	f.k
reg(0x001D) DD DD DD DD DD DD DD DD DD DD	f.k
reg(0x001E) DE DE DE DE DE DE DE DE DE DE	f.k
reg(0x001F) DF DF DF DF DF DF DF DF DF DF	f.k
reg(0x0020) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0	f.k
reg(0x0021) E1 E1 E1 E1 E1 E1 E1 E1 E1 E1	f.k
reg(0x0022) E2 E2 E2 E2 E2 E2 E2 E2 E2 E2	f.k
reg(0x0023) E3 E3 E3 E3 E3 E3 E3 E3 E3 E3	f.k
reg(0x0024) E4 E4 E4 E4 E4 E4 E4 E4 E4 E4	f.k
reg(0x0025) E5 E5 E5 E5 E5 E5 E5 E5 E5 E5	f.k
reg(0x0026) E6 E6 E6 E6 E6 E6 E6 E6 E6 E6	f.k
reg(0x0027) E7 E7 E7 E7 E7 E7 E7 E7 E7 E7	f.k
reg(0x0028) E8 E8 E8 E8 E8 E8 E8 E8 E8 E8	f.k
reg(0x0029) E9 E9 E9 E9 E9 E9 E9 E9 E9 E9	f.k
reg(0x002A) EA EA EA EA EA EA EA EA EA EA	f.k
[... 229 lines cut ...]
reg(0x00B0) F0 F0	RGB Interface Signal Control
reg(0x00B3) F3 F3 F3 F3 F3	Frame Memory
reg(0x00B4) F4 F4	Frame Mode
reg(0x00B6) F6 F6 F6 F6 F6	Display Control
reg(0x00B7) F7 F7	Entry Mode Set
reg(0x00BF) FF FF FF FF FF FF	ILI9481, HX8357-B
reg(0x00C0) C0 C0 C0 C0 C0 C0 C0 C0 C0	Panel Control
reg(0x00C1) C1 C1 C1 C1	Display Timing
reg(0x00C5) C5 C5	Frame Rate
reg(0x00C8) C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8	GAMMA
reg(0x00CC) CC CC	Panel Control
reg(0x00D0) D0 D0 D0 D0	Power Control
reg(0x00D1) D1 D1 D1 D1	VCOM Control
reg(0x00D2) D2 D2 D2	Power Normal
reg(0x00D3) D3 D3 D3 D3	ILI9341, ILI9488
reg(0x00D4) D4 D4 D4 D4	Novatek
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0	GAMMA-P
reg(0x00E1) E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1	GAMMA-N
reg(0x00EF) EF EF EF EF EF EF	ILI9327
reg(0x00F2) F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2	Adjust Control 2
reg(0x00F6) F6 F6 F6 F6	Interface Control
```

prenticedavid, 2020-07-21:
```
reg(0x0004) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x0009) 00 00 61 00 00 00 00 00 00 00	f.k
reg(0x000A) 08 08 08 08 08 08 08 08 08 08	f.k
reg(0x000B) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x000C) 06 06 06 06 06 06 06 06 06 06	f.k
reg(0x000D) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x000E) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x000F) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x002E) 00 FC FC FC 00 00 00 00 00 00	f.k
reg(0x0052) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x0054) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x0056) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x005F) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x0068) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x00BF) FF FF FF FF FF FF FF FF FF FF	f.k
reg(0x00DA) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x00DB) 00 00 00 00 00 00 00 00 00 00	f.k
reg(0x00DC) 00 00 00 00 00 00 00 00 00 00	f.k

```
That is the full set of User registers.   reg(0x2E) is GRAM contents.
I was hoping to see an unlocked ID register.

Now try again with some unlock statements.
See if you get any results.

David.

MBfromOK, 2020-07-21:
Sorry, working from home, doing this between work duties...

I ran these:
```
    //    unlock = unlock_1520;
    //    unlock = unlock_1526;
    //    unlock = unlock_3229;
    //    unlock = unlock_8357;
    //    unlock = unlock_5310;
    //    page_N = d5310_1_in;
```
Didn't see a difference in the for loop results... should I be looking at the readreg() results instead?

prenticedavid, 2020-07-21:
I would hope to see some of the garbage registers spring into life.    Perhaps by enabling the 0xB0 register e.g. unlock = unlock_1520 or unlock_1526.

unlock_3229 is just a magic value only known to a specific manufacturer.
unlock_8357 just refers to the 320x480 Himax controller HX8357.   I would have to guess ID values.
Likewise the unlock_5310 refers to Novatec controllers.

If manufacturers "played cricket",   the ID would be readable as a User register.  
Which would imply a certain Manufacturer.
And then you would read the datasheet to see the special unlock sequence that accesses the Manufacturer registers.

If nothing interesting has appeared it is time to email BangGood.
Quote your invoice number.   ASk them for the part number.    Give them my name as author of MCUFRIEND_kbv.

Since your Red board behaves like a regular MIPI-style controller most punters will be happy to just force tft.begin(0x9341)
However,   I am only happy when I know exactly what make and model of controller is mounted on your Red board.
And I like to get everything work 100%.

David.

MBfromOK, 2020-07-29:
Sorry, been a crazy week, have been losing my internet for weeks at a
time... Will have more updates shortly

Micah

closed 2021-08-24 by prenticedavid