prenticedavid/MCUFRIEND_kbv #149: Buydisplay 5 inch SSD1963 on Arduino Due not supported?

jponko, 2020-07-24:
I have a buydisplay.com (EastRising) 5 inch SSD1963 display. I know you've helped others who have displays which us the SSD1963 driver but I can't seem to get MCUFRIEND_kbv to work on it with those suggestions. I tried to use
#define USE_SPECIAL
#define USE_DUE_16BIT_SHIELD
tft.begin(0x1963); //The ID is not detected

but not even the backlight isn't turned on. I uncommented the USE_DUE_!6BIT_SHIELD in mcufriend_special.h so I'm at a loss on what to try next.

All the information on the display (ER-TFTM050-5(800x480 Pixels) with Resistive Touch Panel) is included in the attached zip along with my simple Arduino code (MuTX.ino). I bought it as a package - display, adapter and Arduino Due. The display works with the UTFT library but the lack of large and scalable fonts forced me to abandon using it.
[MuTX.zip](https://github.com/prenticedavid/MCUFRIEND_kbv/files/4974527/MuTX.zip)

Regards,
Jerry Ponko, AC9NM

prenticedavid, 2020-07-25:
The ER-AS-SSD1963 datasheet has the same wiring as the popular Mea2560 Adapter Shields (as used by UTFT)

```
#elif defined(__SAM3X8E__) && defined(USE_MEGA_16BIT_SHIELD)  //regular MEGA shield on DUE
#warning USE_MEGA_16BIT_SHIELD
#define USES_16BIT_BUS
// configure macros for the control pins
#define RD_PORT PIOA
#define RD_PIN  20     //D43
#define WR_PORT PIOC
#define WR_PIN  7      //D39
#define CD_PORT PIOC
#define CD_PIN  6      //D38
#define CS_PORT PIOC
#define CS_PIN  8      //D40
#define RESET_PORT PIOC
#define RESET_PIN  9   //D41
```

However these "Mega" Adapter Shields do not have a RD pin.    You must force `tft.begin(0x1963)`
Or force the appropriate ID for other displays. 

The "CTE Due Adapter" has different wiring (and no level shifters).    I make a simple hardware mod to connect RD to D24.   This makes the Adapter "readable".    i.e. I can read Ebay 40-pin displays with the "CTE Due Adapter"

I suggest that you use:

#define USE_SPECIAL
#define USE_MEGA_16BIT_SHIELD

Your screen should work on the Due.   Your screen should work on a Mega2560.   (the ER Adapter has level shifters)

It is  only necessary to provide links to ER documents.    Or regular datasheets.   Or regular libraries.
Obviously custom sketches need to be included in your ZIP.

Your ZIP does not seem to have the schematic for ER-AS-SSD1963.
I will look on BuyDisplay website.    It _might_ be possible to enable RD on the data bus buffers.

David.

jponko, 2020-07-25:
David,

I've uncommented  the #define USE_MEGA_16BIT_SHIELD in the
mcufriend_special.h file.

Here's my simple test sketch. The screen remains dark when run on either a
Due or Mega. Am I  missing something?

#include "Adafruit_GFX.h"
#include <MCUFRIEND_kbv.h>
MCUFRIEND_kbv tft;
#define USE_SPECIAL
#define USE_MEGA_16BIT_SHIELD

void setup() {
  tft.begin(0x1963);

  tft.print("HELLO");
}

void loop() {

}

___________________________
Jerry

prenticedavid, 2020-07-25:
Yes,   you have missed the plot.     

Those defines do **not** go into the sketch

You should edit utility/mcufriend_shield.h and utility/mcufriend_special.h
Then the compiler uses those MACROS when building the library

David.

jponko, 2020-07-25:
Still not working.
I removed the defines in my sketch.
I enabled USE_SPECIAL in utility/mcufriend_shield.
I enabled USE_MEGA_16BIT_SHIELD in utility/mcufriend_special.h

Jerry

prenticedavid, 2020-07-25:
I possess several "Mega2560 shields" i.e. 16-bit data bus on #22-#37.   control on #38-#41
e.g. SSD1289, ILI9481, ILI9486, RM68140, ...

These can run on Mega or Due.   Most have 10k resistor packs as serial resistors.    Some have HC245 buffers.
They work fine.    But obviously you need to force tft.begin(0x1289) ... or whatever.

I only have "CTE 40-pin  Adapter for Due".    This can run SSD1963 on a Due.
But I also run SSD1963 with a STM32 BluePill i.e. a home-made adapter from a piece of Protoboard.

I suggest that you start from scratch.   i.e. make a fresh install and edit.

David.

jponko, 2020-07-25:
I did a fresh install of MCUFriend before testing. The display has been out
for about 5 years I gather from the headers in the buydisplay provided UTF
examples. I'll try to find a font generator for the UTFT library.
Apparently not too many peaple have tried to use the Eastrising 5, 7, and 9
inch displays with your library. You may close the issue.

Thanks for trying to help David!

prenticedavid, 2020-07-25:
No,   I would like to get you working properly.

First off.   Run a UTFT example with the BuyDisplay provided UTFT library.

Then remove the BuyDisplay library.  Install and Run a UTFT example with the official RinkyDink UTFT library.

Quote example sketch by name.    Post links to installed library.    (actually I know it is UTFT_VERSION	283 )
And you have already included BuyDisplay stuff on your ZIP.

RinkyDink should work out of the box.     My SSD1963 is 5.0 inch

C:\Users\David Prentice\Documents\Arduino\libraries\UTFT\examples\Arduino (ARM) + Teensy\UTFT_Demo_800x480\UTFT_Demo_800x480.ino

`
UTFT myGLCD(ITDB50,38,39,40,41);
`

David.

jponko, 2020-07-25:
I used the "Resistive_Touch_ButtonTest.ino" sketch by buydisplay.com with
their libraries. They use UTouch instead of the newer URTouch.
Result: Works.
UTFT myGLCD(SSD1963_800480,38,39,40,41);  //(byte model, int RS, int WR,
int CS, int RST, int SER)
--------------------------------------------------------------------------------------------------------------------------------
Deleted Display.com libraries and downloaded/installed Rinky-Dink's UTFT
and URTouch libraries.
http://www.rinkydinkelectronics.com/download.php?f=UTFT.zip (v2.83
06-Aug-2018)

Copied my touch screen settings into the URTouchCD.h and changed the UTFT
display model
UTFT myGLCD(SSD1963_800,38,39,40,41);  //(byte model, int RS, int WR, int
CS, int RST, int SER)
Result: Works

The pin parameters to myGLCD are exactly the same as in your message but
your using a model ITDB50 instead.

Jerry

prenticedavid, 2020-07-25:
UTFT like using gobbledygook names e.g. ITDB50
`
#define ITDB50			13	// SSD1963	(16bit) 800x480
`

I would be pretty confident that MCUFRIEND_kbv would work just fine with:
_I enabled USE_SPECIAL in utility/mcufriend_shield.
I enabled USE_MEGA_16BIT_SHIELD in utility/mcufriend_special.h_
and `tft.begin(0x1963);`
The library build should report:
#warning USE_MEGA_16BIT_SHIELD

David.

jponko, 2020-07-25:
I was looking for the warning message in the output window but it wasn't
there. Could it have anything to do with the Arduino IDE v1.8.13?

Jerry

jponko, 2020-07-25:
These are the shield pinouts and schematics for the shield. The
ER-AS-RA8875.png is and exploded view of the 40-pin connector to the
display. It shows pin 5 as RD/CS. I don't know if you ever got the shield
schematics but here you go.

Jerry

prenticedavid, 2020-07-25:
I added the defines.    And compiled an example that forces ID=0x9486 :
```

Compiling library "Mcufriend_kbv"
"C:\\Users\\David Prentice\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -MMD -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10812 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\David Prentice\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\David Prentice\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\David Prentice\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\David Prentice\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\David Prentice\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "-IC:\\Users\\David Prentice\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\libraries\\SPI\\src" "-IC:\\Users\\David Prentice\\Documents\\Arduino\\libraries\\Mcufriend_kbv" "-IC:\\Users\\David Prentice\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "-IC:\\Users\\David Prentice\\Documents\\Arduino\\libraries\\Adafruit_BusIO" "-IC:\\Users\\David Prentice\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\libraries\\Wire\\src" "-IC:\\Users\\David Prentice\\Documents\\Arduino\\libraries\\Mcufriend_kbv\\utility" "C:\\Users\\David Prentice\\Documents\\Arduino\\libraries\\Mcufriend_kbv\\MCUFRIEND_kbv.cpp" -o "C:\\Users\\DAVIDP~1\\AppData\\Local\\Temp\\arduino_build_337923\\libraries\\Mcufriend_kbv\\MCUFRIEND_kbv.cpp.o"
In file included from C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\utility/mcufriend_shield.h:16:0,

                 from C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\MCUFRIEND_kbv.cpp:35:

C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\utility/mcufriend_special.h:741:2: warning: #warning USE_MEGA_16BIT_SHIELD [-Wcpp]

 #warning USE_MEGA_16BIT_SHIELD

  ^

In file included from C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\MCUFRIEND_kbv.cpp:35:0:

C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv\utility/mcufriend_shield.h:18:2: warning: #warning WE ARE USING A SPECIAL CUSTOM DRIVER [-Wcpp]

 #warning WE ARE USING A SPECIAL CUSTOM DRIVER

```

and the library, memory use lines say:

```
Using library SPI at version 1.0 in folder: C:\Users\David Prentice\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\libraries\SPI 
Using library Mcufriend_kbv at version 3.0.0-Beta in folder: C:\Users\David Prentice\Documents\Arduino\libraries\Mcufriend_kbv 
Using library Adafruit_GFX_Library at version 1.9.0 in folder: C:\Users\David Prentice\Documents\Arduino\libraries\Adafruit_GFX_Library 
Using library Adafruit_BusIO at version 1.3.2 in folder: C:\Users\David Prentice\Documents\Arduino\libraries\Adafruit_BusIO 
Using library Wire at version 1.0 in folder: C:\Users\David Prentice\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12\libraries\Wire 
"C:\\Users\\David Prentice\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-size" -A "C:\\Users\\DAVIDP~1\\AppData\\Local\\Temp\\arduino_build_337923/graphictest_glue.ino.elf"
Sketch uses 53992 bytes (10%) of program storage space. Maximum is 524288 bytes.

```

So I am 100% confident that ID=0x1963 should work for you.

The IDE installs v2.9.9 Release.    I have v3.0.0-Beta on my PC but the code is identical (except for version number)

David.

p.s.  I have the schematic.   It would not be simple to make a hardware mod.
But most importantly,   I have MCUFRIEND_kbv running on Due with a ILI9486 shield.    It will build and run UTFT examples for ILI9486 too.

jponko, 2020-07-25:
It's sort of working now. I had to enable the backlight with
    digitalWrite(8, HIGH);
I tried your Font_simple program but the text lines are all underscored the
width of the screen except for the set background line (photo attached).

I still don't see the warning message though. Progress!

jerry

prenticedavid, 2020-07-25:
UTFT_Demo_800x480.ino does not have any digitalWrite(8. HIGH) for a backlight.
Because most Mega Adapter shields have the backlight hard-wired.

Please just run **all** the examples.   Make notes if you don't understand something.

The underscore shows how fonts are rendered.   e.g. from topline or baseline.

You don't need to agree.    Just understand how to control printing how **you** want it to look.

I don't see any "photo attached".    But I know how that example looks on a small 176x220 or a large 800x480 screen.

Note that your screen has an XPT2046 controller with the SPI signals on random GPIO pins.    So I think you will have to use URTouch.h library.    (copy URTouch constructor from BuyDisplay examples)  

You will only see a Warning if the library is re-compiled.   e.g. Orange text on the verbose report.
Once the library is up to date,  subsequent sketch builds do not need to re-build the library.
If you want to force a re-build,  edit one of the library files e.g. add a space and remove it.

David.

jponko, 2020-07-25:
I tried the graphictest_kbv and text renders ok.

As for the backlight, if I don't add digitalWrite(8, HIGH); to your
sketches, the display backlight remains off.

I know I have to use the UTouch or URTouch lib for the touch controller.
I'm not really that stupid.

It seems to work OK so you can close the issue.

Thanks for your help David!

Jerry

jponko, 2020-07-26:
As a followup to the backlight control. From the pdf data sheet I sent you,
on page 10, table 3.8 JP8-Arduino Board Output Interface, it says:
1 BL_CONTROL BackLight control signal Arduino board digital IO 8

Thanks again,
Jerry

sk8rsdad, 2020-07-31:
I found this very helpful in getting the non-shield version to work. I had to comment out the | INVERT_RGB when setting _lcd_capable, select table8_ads = SSD1963_480_regvalues, and set the height (*p16 = 272) to make graphicstest_kbv.ino fully functional. 

The next step is to get an 8-bit parallel version to work.

Thanks David and Jerry for making things a little easier for some random stranger!

prenticedavid, 2020-07-31:
Yes,   if your screen is 480x272 you will need different init table to the 5 inch 800x480 or the 7 inch 800x480.

Please let me know if everything works 100%.   i.e. rotations, colours, directions,...

sk8rsdad, 2020-07-31:
I think things are working as expected. Portrait and Landscape labels are messed up since the display's default is landscape and graphicstest_kbv.ino assumes portrait. Scrolling works but direction always seems to be vertical relative to portrait. I don't know if that's expected behaviour.

I recorded a video but it can't be posted here. Perhaps these screen captures will help.

[image]
[image]
[image]
[image]
[image]

prenticedavid, 2020-08-01:
Your photos imply that the rotations are good.   And the Vertical Scroll is good too.

Note that your SSD1963 has got oodles of GRAM memory.  (enough for 384000 pixels)
Your panel only displays 130k pixels.

It should be possible to build an image off-screen and flip to it.
There is no library support for this but you should find users that have done it with other libraries.

David.

closed 2021-08-24 by prenticedavid