prenticedavid/MCUFRIEND_kbv #150: Compile error / install possible error

markotime, 2020-07-25:
MCUFRIEND_kbv.h was not installed along with rest of files via Arduino IDE.  Manual download of source and placement in example resulted in many errors of the form ...  undefined reference to  'MCUFRIEND.kvb::...'.    IDE is 1.8.13.  What am I doing wrong? thanks

prenticedavid, 2020-07-25:
Install IDE in the normal way.  e.g. downloading from Arduino.cc

Run the IDE in the normal way.  e.g. Tools->Board:Uno, Tools->Port:COM4 etc.

Install/Update any libraries **via IDE Library Manager** 
e.g. Tools->Manage Libraries: Adafruit_GFX, MCUFRIEND_kbv

If you have installed libraries manually in the past,   I **strongly** advise you to leave the IDE,  delete the specific diredtories, start the IDE,  install properly **via IDE Library Manager**

I know that this is GitHub but unless you are an experienced GitHub user it is better to work via the Arduino IDE.   And post questions on the Arduino.cc Displays Forum

Please let me know how you get on.

David.

markotime, 2020-07-25:
Thank yhou so much for the prompt response. while the screen gave no
indication that the .h file had been found or installed, it worked.
Ran the Touchscreen_calibrate_native code, displayed on serial monitor but
can't copy, but can enter manually;.   WHERE do I paste it?
I cannot see where in touchscreen.h (I assume?) to paste it.
Thank you again!

prenticedavid, 2020-07-25:
First off.   Run **all** the example sketches.

Read any messages on Serial Terminal.
Copy-Paste anything that you don't understand.

Regarding Touch.   You copy-paste the two "const int" lines to your sketch.   You don't alter the library code.

markotime, 2020-07-25:
Thank you so much.  Nowhere did I see "paste into sketch".  Problem solved.

prenticedavid, 2020-07-25:
What was the original problem ?

How did you install and from where?

markotime, 2020-07-25:
Original problem was linking errors involving mcufriend .h file.  Having
reinstalled your library, that problem went away.
The calibration sketch compiled and ran, I just didn't know where to enter
the cal parameters.  Now I know, thank you.

closed 2021-08-24 by prenticedavid