prenticedavid/MCUFRIEND_kbv #152: Conflict with touch on Adafruit Grand Central M4 board

camdey, 2020-08-03:
@prenticedavid you may remember I helped you test the implementation of the Grand Central support for your library a few months ago. At the time I was mainly testing writing to the screen, lately I've been working more on the touch aspect.

I'm encountering an issue related to touching the screen but it seems that it could be related to this library. If I touch a certain part of the screen while printing to it, the screen will crash to either black or white or have other issues. Sometimes fading to white slowly or going straight to white. The Arduino itself is functional and continues to output on the serial monitor. Whether it crashes or has issues seems related to how much is being written to the screen or how frequently.

I've reproduced this issue on two TFTs:

- Velleman VMA412 2.8" TFT (8/16 bit, ILI9341)
- Kuman 3.5 Inch Tft (8/16 bit ILI9481 or ILI9488)

I've not been able to reproduce the issue on an Arduino Mega or Arduino Due. For reproducing I've been able to use a custom sketch that print a simple bitmap and changes its colour every x hundred milliseconds as well as the MCUFRIEND graphictest_kbv sketch.

It seems to be related to touching the screen while printing to the screen. If I print a small bitmap symbol and that's it, I can touch the screen without issue. If I keep changing the colour of the bitmap while touching the screen, it crashes or has issues. Perhaps some conflict between the shared pins?

It seems to only occur in some parts of the screen. For the 3.5", it's the top third of the screen above the LCD digital pins. 

An issue that can occur if it doesn't crash is that it seems to pause/stop printing to the screen while the screen is being touched. The Arduino can still output to the serial monitor but any code calling tft.print will not print until the screen is not touched in that particular 1/3 segment of the display. Sometimes it will print a few pixels but in the wrong areas.

Very odd behaviour. Seems like the screen is picking up a touch in certain areas and interrupting writing to the screen - which either messes up what is being written, stops it from being written, or crashes the screen controller.

Any ideas? So far I've tried increasing the delays in the shield.h file but no luck. I've linked some videos below. The first uses the graphcstest_kbv example, the second is a small sketch of my own that essentially boils down to:
```
  if (millis() - prevButtonCheck >= 1850) {
    tft.setCursor(100, 100);
    tft.setTextColor(0xFFE0);
    tft.drawBitmap(100, 100, flashOn, 80, 80, colour);
    if (colour == RED) {
      colour = YELLOW;
    }
    else {colour = RED;}
    prevButtonCheck = millis();
    Serial.println("test");
  }
```
(no touchscreen library or function call etc exists in the custom sketch).

the videos may still be processing if they don't load
https://photos.app.goo.gl/o7CpnnrNReoWMn1e9
https://photos.app.goo.gl/Qmp2uNsXu27PSeLU8

prenticedavid, 2020-08-03:
I will have look tomorrow.

Either show a video of an **un-altered** example e.g. graphictest_kbv
Or paste / attach your custom sketch.   So that I can run on my hardware.

David.

camdey, 2020-08-04:
The first video is an unaltered example of graphictest_kbv. Only the pinout values were changed to match my shield.

prenticedavid, 2020-08-04:
It looks as if your Shield plugs straight into the Grand Central headers.
And you have some external electronics going to A14, A15, GND, D37, 5V

I would expect the Shield to work straight out of the box.   i.e. library v2.9.9-Release installed by the Library Manager.
You do not need to alter any library files.

I will try some Touch programs on a 180MHz M4 and a 216MHz M7 STM32.  And a Teensy4.0

Note that the Resistive Touch panel is a heavy load on ARM GPIO pins.   But I would not expect it to kill regular graphics.

There are other Issues with Grand Central e.g. https://github.com/prenticedavid/MCUFRIEND_kbv/issues/151

I don't really want to buy a GC board.   It will be good to resolve these problems remotely.

David.

camdey, 2020-08-04:
> And you have some external electronics going to A14, A15, GND, D37, 5V

Ah yes sorry about the distraction with those, they are connected but not implemented in either the graphictest_kbv sketch or the custom sketch I was running. They were part of the main sketch where I first noticed the problem and I just left them attached. It's just a basic Arduino joystick with X,Y,Z inputs connected and a voltage divider to bring the 5V down to 3.3V.

> I would expect the Shield to work straight out of the box. i.e. library v2.9.9-Release installed by the Library Manager.
> You do not need to alter any library files.

Yes you're right, I remember now it was just the touch calibration example I was also testing yesterday that required some changes.

I also saw #151 yesterday and contemplated posting this problem under that issue but figured the symptoms were different enough to warrant its own thread.

Happy to help resolve remotely. I wouldn't recommend the GC myself, has given me too many headaches, so save yourself the money! After we got the GC working with your library I had to switch to the Due because of the other problems (GC tends to get stuck in bootloader mode and requires the reset pin to be tapped twice - useless the board is in an enclosure).

If it's of any importance, the areas on the 2.8" TFT that when touched would cause the same problems were different to the areas on the 3.5" TFT. On the 3.5", it's only the top third (in landscape, USB port on the left) / right third (in portrait, USB port at top) that will cause this problem when touched. On the 2.8", it was several areas across the screen. I can post a video of the 2.8" if it would help.

Cheers

prenticedavid, 2020-08-05:
In mcufriend_shield.h

```
//####################################### GRAND CENTRAL M4 ############################
#elif defined(__SAMD51P20A__)      //regular UNO shield on GRAND CENTRAL M4
//LCD pins   |D7  |D6  |D5  |D4  |D3  |D2  |D1 |D0  | |RD |WR |RS |CS |RST|
//SAMD51 pin |PD21|PD20|PC21|PC20|PC19|PC18|PB2|PB18| |PA2|PA5|PB3|PC0|PC1|
...
#define WRMASK        ((0<<22) | (1<<28) | (1<<30)) //

```
alter WRMASK e.g.

`#define WRMASK        ((1<<22) | (1<<28) | (1<<30)) //
`

See if increasing the DRVSTR for the data bus makes a difference.
Untested.   I can only test on SAMD21

David.

camdey, 2020-08-05:
Unfortunately changing from:
`#define WRMASK ((0<<22) | (1<<28) | (1<<30)) //`
to
`#define WRMASK ((1<<22) | (1<<28) | (1<<30)) //`

Had no effect as far as I can tell. Tested on both the graphictest_kbv sketch and my custom sketch.

I use the custom sketch since it doesn't tend to crash the display, rather just interrupts the printing to the screen. Probably won't help on your end but if interested it's copied below:
```
#include "Arduino.h"
#include "MCUFRIEND_kbv.h"

const unsigned char flashOn [] PROGMEM = {
  0x00, 0x00, 0x00, 0x00, 0x7f, 0xfe, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0xff, 0xff,
  0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3f, 0xff, 0xff, 0xfc, 0x00, 0x00, 0x00, 0x00, 0x00,
  0x00, 0xff, 0x00, 0x00, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0xf0, 0x00, 0x00, 0x0f, 0xc0,
  0x00, 0x00, 0x00, 0x00, 0x0f, 0xc0, 0x07, 0xe0, 0x03, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x3f, 0x01,
  0xff, 0xff, 0x80, 0xfc, 0x00, 0x00, 0x00, 0x00, 0x7c, 0x07, 0xff, 0xff, 0xf0, 0x3e, 0x00, 0x00,
  0x00, 0x01, 0xf0, 0x07, 0xc0, 0x03, 0xfc, 0x0f, 0x80, 0x00, 0x00, 0x03, 0xe0, 0x00, 0x00, 0x00,
  0x3f, 0x07, 0xc0, 0x00, 0x00, 0x07, 0x83, 0x80, 0x00, 0x00, 0x0f, 0xc1, 0xe0, 0x00, 0x00, 0x0f, 
  0x0f, 0x80, 0x00, 0x00, 0x01, 0xf0, 0xf0, 0x00, 0x00, 0x1e, 0x1f, 0x00, 0x00, 0x00, 0x00, 0xf8,
  0x78, 0x00, 0x00, 0x3c, 0x3c, 0x00, 0x00, 0x00, 0x00, 0x3c, 0x3c, 0x00, 0x00, 0x78, 0x78, 0x00,
  0x00, 0x00, 0x00, 0x1e, 0x1e, 0x00, 0x00, 0xf1, 0xe0, 0x00, 0xff, 0xff, 0xf0, 0x07, 0x0f, 0x00,
  0x00, 0xe1, 0xc0, 0x00, 0xff, 0xff, 0xf0, 0x03, 0x87, 0x00, 0x01, 0xc3, 0x80, 0x01, 0xc0, 0x00,
  0x70, 0x01, 0xc3, 0x80, 0x03, 0xc7, 0x00, 0x01, 0xc0, 0x00, 0x70, 0x00, 0xe3, 0xc0, 0x03, 0x8e,
  0x00, 0x01, 0x80, 0x00, 0x70, 0x00, 0x71, 0xc0, 0x07, 0x1e, 0x00, 0x03, 0x80, 0x00, 0x60, 0x00,
  0x78, 0xe0, 0x07, 0x1c, 0x00, 0x03, 0x80, 0x00, 0xe0, 0x00, 0x38, 0xe0, 0x0e, 0x38, 0x00, 0x07,
  0x00, 0x00, 0xe0, 0x00, 0x1c, 0x70, 0x0e, 0x38, 0x00, 0x07, 0x00, 0x01, 0xc0, 0x00, 0x1c, 0x70,
  0x1c, 0x70, 0x00, 0x07, 0x00, 0x01, 0xc0, 0x00, 0x0e, 0x38, 0x1c, 0x60, 0x00, 0x06, 0x00, 0x03,
  0x80, 0x00, 0x06, 0x38, 0x38, 0xe0, 0x00, 0x0e, 0x00, 0x03, 0x80, 0x00, 0x07, 0x1c, 0x38, 0xe0,
  0x00, 0x0e, 0x00, 0x03, 0x80, 0x00, 0x07, 0x1c, 0x31, 0xc0, 0x00, 0x1c, 0x00, 0x07, 0x00, 0x00,
  0x03, 0x8c, 0x71, 0xc0, 0x00, 0x1c, 0x00, 0x07, 0x00, 0x00, 0x03, 0x8e, 0x71, 0x80, 0x00, 0x1c,
  0x00, 0x0e, 0x00, 0x00, 0x01, 0x8e, 0x73, 0x80, 0x00, 0x38, 0x00, 0x0e, 0x00, 0x00, 0x01, 0xce,
  0x63, 0x80, 0x00, 0x38, 0x00, 0x0e, 0x00, 0x00, 0x01, 0xc6, 0xe3, 0x80, 0x00, 0x38, 0x00, 0x1c,
  0x00, 0x00, 0x01, 0xc7, 0xe3, 0x80, 0x00, 0x30, 0x00, 0x1c, 0x00, 0x00, 0x00, 0xc7, 0xe3, 0x00,
  0x00, 0x70, 0x00, 0x1f, 0xff, 0x80, 0x00, 0xc7, 0xe3, 0x00, 0x00, 0x70, 0x00, 0x1f, 0xff, 0x80,
  0x00, 0xc7, 0xe7, 0x00, 0x00, 0xe0, 0x00, 0x00, 0x07, 0x80, 0x00, 0xe7, 0xe7, 0x00, 0x00, 0xe0,
  0x00, 0x00, 0x07, 0x00, 0x00, 0xe7, 0xe7, 0x00, 0x00, 0xe0, 0x00, 0x00, 0x0f, 0x00, 0x00, 0xe7,
  0xe7, 0x00, 0x00, 0xc0, 0x00, 0x00, 0x1e, 0x00, 0x00, 0xe7, 0xe7, 0x00, 0x01, 0xc0, 0x00, 0x00,
  0x3c, 0x00, 0x00, 0xe7, 0xe7, 0x00, 0x01, 0xc0, 0x00, 0x00, 0x78, 0x00, 0x00, 0xe7, 0xe3, 0x00,
  0x01, 0xff, 0xfc, 0x00, 0xf0, 0x00, 0x00, 0xc7, 0xe3, 0x00, 0x01, 0xff, 0xfc, 0x00, 0xe0, 0x00,
  0x00, 0xc7, 0xe3, 0x00, 0x00, 0x00, 0x1c, 0x01, 0xc0, 0x00, 0x00, 0xc7, 0xe3, 0x80, 0x00, 0x00,
  0x1c, 0x03, 0x80, 0x00, 0x01, 0xc7, 0x63, 0x80, 0x00, 0x00, 0x18, 0x07, 0x80, 0x00, 0x01, 0xc6,
  0x73, 0x80, 0x00, 0x00, 0x38, 0x0f, 0x00, 0x00, 0x01, 0xce, 0x71, 0x80, 0x00, 0x00, 0x38, 0x1e,
  0x00, 0x00, 0x01, 0x8e, 0x71, 0xc0, 0x00, 0x00, 0x30, 0x3c, 0x00, 0x00, 0x03, 0x8e, 0x31, 0xc0,
  0x00, 0x00, 0x70, 0x78, 0x00, 0x00, 0x03, 0x9c, 0x38, 0xe0, 0x00, 0x00, 0x70, 0x70, 0x00, 0x00,
  0x07, 0x1c, 0x38, 0xe0, 0x00, 0x00, 0x70, 0xe0, 0x00, 0x00, 0x07, 0x1c, 0x1c, 0x60, 0x00, 0x00,
  0x61, 0xc0, 0x00, 0x00, 0x06, 0x38, 0x1c, 0x70, 0x00, 0x00, 0xe3, 0x80, 0x00, 0x00, 0x0e, 0x38,
  0x0e, 0x38, 0x00, 0x00, 0xe7, 0x80, 0x00, 0x00, 0x1c, 0x70, 0x0e, 0x38, 0x00, 0x00, 0xcf, 0x00,
  0x00, 0x00, 0x1c, 0x70, 0x07, 0x1c, 0x00, 0x01, 0xde, 0x00, 0x00, 0x00, 0x38, 0xe0, 0x07, 0x1e,
  0x00, 0x01, 0xfc, 0x00, 0x00, 0x00, 0x78, 0xe0, 0x03, 0x8e, 0x00, 0x01, 0xf8, 0x00, 0x00, 0x00,
  0x71, 0xc0, 0x03, 0xc7, 0x00, 0x03, 0xf0, 0x00, 0x00, 0x00, 0xe3, 0xc0, 0x01, 0xc3, 0x80, 0x03,
  0xe0, 0x00, 0x00, 0x01, 0xc3, 0x80, 0x00, 0xe3, 0xc0, 0x03, 0xc0, 0x00, 0x00, 0x03, 0x87, 0x00,
  0x00, 0xf1, 0xe0, 0x03, 0xc0, 0x00, 0x00, 0x07, 0x8f, 0x00, 0x00, 0x78, 0xf8, 0x00, 0x00, 0x00,
  0x00, 0x1e, 0x1e, 0x00, 0x00, 0x3c, 0x3c, 0x00, 0x00, 0x00, 0x00, 0x3c, 0x3c, 0x00, 0x00, 0x1e,
  0x1f, 0x00, 0x00, 0x00, 0x00, 0xf8, 0x78, 0x00, 0x00, 0x0f, 0x0f, 0x80, 0x00, 0x00, 0x01, 0xf0,
  0xf0, 0x00, 0x00, 0x07, 0x83, 0xf0, 0x00, 0x00, 0x0f, 0xc1, 0xe0, 0x00, 0x00, 0x03, 0xe0, 0xfc,
  0x00, 0x00, 0x3f, 0x07, 0xc0, 0x00, 0x00, 0x01, 0xf0, 0x3f, 0xc0, 0x03, 0xfc, 0x0f, 0x80, 0x00,
  0x00, 0x00, 0x7c, 0x0f, 0xff, 0xff, 0xf0, 0x3e, 0x00, 0x00, 0x00, 0x00, 0x3f, 0x01, 0xff, 0xff,
  0x80, 0xfc, 0x00, 0x00, 0x00, 0x00, 0x0f, 0xc0, 0x07, 0xe0, 0x03, 0xf0, 0x00, 0x00, 0x00, 0x00,
  0x03, 0xf0, 0x00, 0x00, 0x1f, 0xc0, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0x00, 0x00, 0xff, 0x00,
  0x00, 0x00, 0x00, 0x00, 0x00, 0x3f, 0xff, 0xff, 0xfc, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07,
  0xff, 0xff, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x7f, 0xfe, 0x00, 0x00, 0x00, 0x00
};

MCUFRIEND_kbv   tft;

unsigned long prevButtonCheck = 0;
[... cut ...]
[... 1 lines cut ...]

void setup() {
  Serial.begin(250000);
  tft.reset();
  uint16_t ID = tft.readID(); //
  tft.begin(ID);
  tft.fillScreen(0x0000);
  tft.setRotation(1);
  colour = RED;
}

void loop() {
  if (millis() - prevButtonCheck >= 150) {
    tft.setCursor(100, 100);
    tft.setTextColor(0xFFE0);
    tft.drawBitmap(100, 100, flashOn, 80, 80, colour);
    if (colour == RED) {
      colour = YELLOW;
    }
    else {colour = RED;}
    prevButtonCheck = millis();
    Serial.println("test");
  }
}
```

camdey, 2020-08-05:
And just to be thorough: changing `tft.setRotation(1);` to `tft.setRotation(0);` also has no effect. The area on the screen that causes the issue when touched remains the same.

Not sure how these screen drivers work, do they break down the display into columns/rows where a group of these are controlled by a specific port on the driver? Not sure why else only parts of the screen would be affected.

prenticedavid, 2020-08-05:
Your first video certainly shows a crashed Arduino.
The custom sketch would just show an icon changing from Red to Yellow every 150ms.

```
void loop() {
  if (millis() - prevButtonCheck >= 150) {
    tft.drawBitmap(100, 100, flashOn, 80, 80, colour);
    if (colour == TFT_RED) {
      colour = TFT_YELLOW;
    }
    else {colour = TFT_RED;}
    prevButtonCheck = millis();
  }
}
```
the drawBitmap() is an inherited GFX method that plots each pixel with drawPixel() i.e. fairly slow.    But should complete inside 150ms.

I don't see what your second video is supposed to demonstrate. 

Does the graphictext_kbv example run 100% if you don't touch the screen ?

David.

camdey, 2020-08-05:
> The custom sketch would just show an icon changing from Red to Yellow every 150ms.
Correct

> Does the graphictext_kbv example run 100% if you don't touch the screen ?

Yup, about 3.2seconds or so if I remember correctly.

> But should complete inside 150ms [...] I don't see what your second video is supposed to demonstrate.

I've changed the timing of the loop at various points to test if it made a difference. In the second video the loop is set at 1850ms I believe.

The second video shows that the bitmap is not written to the screen when the screen is touched. Perhaps it's hard to see given the delay is almost two seconds, but you'll notice that while I'm touching the screen it won't change colour for much longer than 2 seconds, it won't change until my finger is released.

The serial print showed me that the loop _is_ executing while the screen is being touched, so the issue must be related to the writing of the bitmap to the screen (actually just writing in general - whether a bitmap or text or a line etc).

The difference between the first and second video is in the symptoms. The first video shows a crash, the second shows an interruption to the writing to the screen. My hypothesis is that the difference here is related either to the refresh rate/how frequently something is written to the screen or the amount of data being written to the screen. 

Perhaps if the refresh rate is very high, such as on the graphicstest_kbv sketch, the interruption from touching the screen causes it to completely crash. But if the refresh rate is slower (150ms+), it is able to handle it without crashing but not function properly.

prenticedavid, 2020-08-05:
I suggest that you write a sensible sketch.   And post the video that relates to it.

But first off.    I would run **all** the library examples without tapping the Touch screen.
Obviously the Touch examples need to use the Touch stylus in the regular way.

Report any problems.

I am more concerned with regular Graphics programs and any glitches.
If graphictest_kbv runs tests too fast,   add a longer delay after each test.

The photos from the #151 Issue show several specks.
Does altering DRVSTR make a difference?
What is the quality of soldering on your Shield and M4 board ?

Chinese manufacture is generally pretty good.
US manufacture might not be as good.

David.

camdey, 2020-08-05:
What is your definition of a sensible sketch? Displaying an icon is not unreasonable and the refresh rate seems to not matter. It's a simple sketch that is easy to test and reproduce the error. I can print text instead and the issue is still reproduced easily.

I've run a bunch of the library examples - the three graphictest ones, drawBitmap, button_simple, scroll_kbv, readpixel etc. I only identify an issue when I touch the screen in the aforementioned areas. They otherwise run normally as far as I can tell.

Another symptom worth mentioning is that when the display crashes I usually have to unplug the USB connected between board and PC before I'm able to upload another sketch. The device disappears from the COM ports. This is not an infrequent issue for the GC boards in general but in this case it does specifically occur whenever the display crashes.

Re: #151 I haven't seen that kind of issue before. My displays works fine except for the issue that is the subject of this thread.

**A new finding:** Using the stylus pen seems to be fine, I can only replicate the issue when using my finger. The stylus pen is one of those crappy white plastic ones supplied with TFT screens.

Two new videos for you, they may take some time to process so it's often quicker to download them from Photos first.
1. button_simple sketch: https://photos.app.goo.gl/GtFdFL36P9rdYHJ1A
**changes I made:** touch pins, left/right top/bottom screen calibration, screen rotated, buttons moved to "problem" area
**observations:** Works as expected with stylus. With finger touch, buttons are triggered less consistently and the display eventually crashes.
**wait until end of video for screen crash**

2. scroll_kbv sketch: https://photos.app.goo.gl/RRb8Cq4JzX86Gug98
**changes I made:** none
**observations:** Works as expected with stylus. With finger touch, screen crashes fairly quickly

I don't believe it's related to how hard I touch the screen, either with a stylus or my finger. I can press fairly hard with a stylus and it won't cause any issues.

TFT soldering: seems fine and I can't replicate this issue with the same screen on other boards (Mega, Due)
Grand Central soldering: looks perfect. Additionally, putting lateral pressure on the screen (and thus the board headers) does not cause any issue, so doesn't seem to be an issue with the soldering or the headers themselves. Can also rule this out due to pressure tests using finger vs stylus and when I put downward force on the screen by holding the sides.

camdey, 2020-08-05:
UPDATE:
I made this change and ran the scroll_kbv sketch again:
`#define WRMASK        ((1<<22) | (1<<28) | (1<<30))`

The stylus now causes the display to crash, practically immediately. I changed it back to the original setting `(0<<22)` and the stylus no longer causes crashes, only finger touches.

prenticedavid, 2020-08-05:
Sorry about the "sensible" comment.

A test sketch needs to be as simple as possible,   And distinguish between setup() and loop()
e.g. setup() prints "hello".   fills screen with a special colour.   waits for a second.  fills screen with regular background colour.
e.g. loop() draws a coloured box.   Changes colour every 100ms

This would show if the program is resetting.
And what areas of the screen are sensitive.

There is nothing wrong with drawing an icon.    It just means a lot of data lines that "clutter" the test sketch.

I still suspect a bad contact that is aggravated by your finger pressure.

Yes,   I would like to resolve any GC problems.
The behaviour on Mega, Due, GC should be identical.

I have numerous STM32 boards,  Zero, Due, Teensy 3.2, 4.0.
They all seem pretty reliable.

David.

camdey, 2020-08-05:
No worries David.

I can do some more testing this evening or tomorrow - maybe even do a quick test sketch as you suggest. In the mean time I don't think the contact pressure is the issue, though it would make sense.

With the scroll_kbv sketch, I can get the display to crash with moderate finger pressure where moderate is about the hardest you would normally press.

With the button sketch, I can push with my finger far beyond what would be normally reasonable and the display will not crash. It seems to really take awhile for it to crash on this sketch in general.

If it is a bad contact it must be on the GC given I can't replicate on the Due or Mega but I can replicate on another screen with the GC. Let me know if you'd like to see any videos that might illustrate the contact pressure I'm placing on the screen, or any other suggestions that might help point to this being the problem. Based on my experiences so far it doesn't appear to be related to the pressure.

I was able to cause the screen to crash with a finger-sized plastic object but this object can also perform readable touches on the display - perhaps it's an issue with the resistance or capacitance reading (can't recall which one this screen is) over an area larger than a stylus?

camdey, 2020-08-05:
Two more videos for you. Here's the new test sketch:
```
#include "Arduino.h"
#include "MCUFRIEND_kbv.h"

MCUFRIEND_kbv   tft;

unsigned long prevButtonCheck = 0;
#define YELLOW 0xFFE0
#define RED 0xF800
#define BLUE 0x001F
int colour = 0;

void setup() {
  Serial.begin(250000);
  tft.reset();
  uint16_t ID = tft.readID(); //
  tft.begin(ID);
  tft.fillScreen(BLUE);
  tft.setRotation(1);
  tft.setCursor(50, 50);
  tft.setTextColor(0xFFFF);
  tft.setTextSize(12);
  tft.print("HELLO WORLD");
  delay(1000);
  tft.fillScreen(0x0000);
  colour = RED;
}

void loop() {
  if (millis() - prevButtonCheck >= 150) {
    tft.fillRect(20, 20, 120, 60, colour);
    if (colour == RED) {
      colour = YELLOW;
    }
    else {colour = RED;}
    prevButtonCheck = millis();
    Serial.println("test");
  }
}
```
Sorry for the lighting, was a bit tricky for my phone to capture the bright screen relative to the brightness of my apartment.

https://photos.app.goo.gl/ctBzPMS2FVZQdyvWA
First video: shows the sketch running, a rectangle in the "problem" area changes colour every 150ms. I try to demonstrate how placing pressure on different parts of the screen shield (not the screen directly) doesn't seem to replicate the issue. I also try to show how much pressure I'm putting on the screen with the stylus compared to with my finger.

https://photos.app.goo.gl/WnNMhSHKV1DsaE8R8
Second video: shows the displacement of the screen relative to its shield. Here you can see the screen is moving at least an equal amount when I use the stylus compared to when I use my finger. I'd argue I put much less pressure on it with my finger, it's hard to tell but I really have to steady it with my hand when trying to force the stylus to cause the problem.

It seems the screen did crash once with the stylus but I really had to push hard, compared to with my finger where it's a much more normal touch pressure - what you'd normally use when interacting with these displays.

Just did a quick third video with the 2.8" display. The problem area for this one seems to be on the opposite side to the 3.5" display. Can't replace the bug when pressing on the same side as before at all.
https://photos.app.goo.gl/ejyuiBUsD3oS67UC9

The video is a bit hard to see what's going on due to the lighting, basically when the square flashes white it's yellow but just very bright. When I start pressing the other side you can see the drawing either stops or gets mixed in with the other colour (red).

I wouldn't 100% rule out a contact issue but I've really given it a good shake and move about, even slightly rocking the pins out of the headers (not completely, just a reasonable amount to resemble the leverage from the pressure I'd apply on a corner of the screen) and nothing. Can't get any glitches of any kind.

prenticedavid, 2020-08-05:
Looking at your side-view video.    The TFT panel is normally mounted on the Shield pcb with some double-sided foam tape.    Many just use plain double-sided sellotape i.e. no physical thickness.

I suspect that your brutal fingers are shorting the metal backing plate of the TFT panel against the pcb traces.
Buy a small roll of 2mm foam tape first.
Then carefully separate the TFT from the pcb by cutting through the existing double-sided tape with an X-Acto knife.
Re-assemble with the 2mm foam.

Resistive Touch Panels are designed for gentle touch with a stylus.   Or gentle touch with a finger.
Capacitive Touch Panels work with very little pressure.

Be realistic about your "Users".    Do they lose their temper?   Are they brutal ?
e.g. I would not trust you with a Resistive Panel.

David.

camdey, 2020-08-06:
I can take another look at this but this doesn't quite add up for several reasons. 

1. I **cannot** replicate this issue on the Due or on the Mega. 
2. I am pressing much harder with the stylus and can't get it to replicate the problem
3. I'm using a normal finger pressure, not "brutal", however it may look. 
4. There is a lot of resistance or padding under the screen when I try to push on it. There is a little "give" but normal amount in my experience with TFTs. It seems like it'd be foam but I can check. 
5. I'm able to replicate the issue when holding the sides of the screen (not the backing plate) - an attempt to minimise the movement between screen and plate. 
6. Changing the drvstr makes the screen very sensitive to this problem. I can tap relatively lightly with a stylus and reproduce the problem. 
7. The 2.8" TFT has the same problem and I've not been able to reproduce it on other boards. 

I think you are mistaking my attempts and trying to replicate the problem under certain conditions with my normal usage. I am purposefully using hard pressure with the stylus to show that the movement of the screen seems to be irrelevant. You can even see in that side video I didn't realise I triggered the problem with my finger because I was pressing so lightly. 

Happy to do some tests with different finger pressure to show you but I believe the above points are indicating that we should to focus our attention elsewhere.

prenticedavid, 2020-08-06:
`#define WRMASK ((1<<22) | (1<<28) | (1<<30))`

Which DRVSTR is more sensitive?   (1<<22) or (0<<22)

It is probably wise to wait for response from other GC owners e.g. #151 

It seems to be related to the GC board rather than Shield.
SAM51D will be nearest to the SAM3X on a Due.

I don't have a GC.   I suggest that you ask other GC owners e.g. on Adafruit Forum.    Since the GC is supported by the regular MCUFRIEND_kbv 2.9.9-Release it is easy for another GC owner to plug a Shield into the GC headers.

David.

camdey, 2020-08-06:
Hi David,

I'll reach out on the forums at some point to see if anyone can replicate. If you have a chance, may be worth asking the person with the issue in #151 when they next reply.

By the way, I checked this morning and my screen does have the 2mm black foam under the display.

> Which DRVSTR is more sensitive? (1<<22) or (0<<22)

`(1<<22)` was more sensitive when I tested it. More sensitive defined as being very easily triggered by the stylus when normally I cannot replicate the issue with the stylus.

Cheers

camdey, 2020-08-09:
Unfortunately no one replied to a thread I posted on the Arduino forum, hopefully someone else sees this eventually and can test.

In the mean time, I wedged a small bit of folded paper between the screen and its backing plate to prevent it from moving when touched. Didn't make a difference, I still have the problem. It's difficult to capture on video but I can have the screen detect touches with very little finger pressure and it's still enough to cause the issue.

At least I'm primarily using the Due these days so happy to give up on this problem for now.

prenticedavid, 2020-08-11:
I was suggesting Adafruit Forums.   They are more likely to have Adafruit "owners",

I frequent the Arduino.cc Displays Forum but seldom look at the other Arduino.cc forums.
I suspect that GC owners will only frequent the specific forum(s) that interest them.

Arduino.cc is "too big" for me.
On the other hand,  I see every new post title on avrfreaks.net but only read the threads that interest me.
I have never followed Adafruit Forum.

David.

camdey, 2020-08-11:
I've been posting on the Adafruit forum recently regarding my other GC issues and have found it generally a bit quiet. The Adafruit people haven't been particularly helpful either. My GC will disable my computer's peripheral devices occasionally when it uploads a sketch and they're practically ignoring me at this point on the forums.

(don't buy a Grand Central).

I'll post something and see if they reply anyway. At the very least it's good to advertise that your library can support these new boards as it seems a few people are unsure on what library to use for TFT displays with the GC.

closed 2021-08-24 by prenticedavid