prenticedavid/MCUFRIEND_kbv #153: How to compile code with stm32 and PlatformIO on Arduino framework?

Magix35, 2020-09-17:
Hello,
I don't understand how to compile the library in platformIO, with arduino framework with an STM32F103CBT6 bluepill.
I have set genericSTM32F103CB in platformio.ini.
I succeeded to test my LCD (2.4" TFT LCD SHIELD) with an arduino UNO and everything works fine.
When I compile (GLUE_Demo_320x240) for the stm32 I have more than 1000 errors.

Do i need to define some macros when i use STM32 ?

Thank for all reply.

prenticedavid, 2020-09-17:
In an ideal world you plug an Arduino Shield into header sockets on an Arduino board.   Build for that specific Arduino board e.g. Uno, Due, ...

The IDE will pass some defines  to the Compiler e.g. `-DARDUINO_AVR_UNO` and `-DARDUINO_ARCH_AVR`
e.g. `-DARDUINO_BLUEPILL_F103CB` and `-DARDUINO_ARCH_STM32`
The appropriate conditional block is selected in mcufriend_shield.h by the preprocessor.

I have never used PlatformIO.   Look at the command line e.g. for building an empty sketch.
Note all the `-Dxxxx` arguments and post them in your message.

The default wiring for a BLUEPILL is:
```
//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1**

```

David.

Magix35, 2020-09-18:
Thank to replied that fast, I couldn't reply before due to I'm french.

It seem that platfomIO doesn't define macros -DARDUINO_BLUEPILL_F103CB and -DARDUINO_ARCH_STM32. And we need to define them with build_flags zone of platformio.ini

[image]

With those macros the program compile and the LCD get data but there are only the top of the screen which print data the rest of the screen print random pixels.
I have test many of wiring and all give the same results. 

(For wiring I have made a shield to connect the LCD to the STM32 with
 //LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1** 
)

I have test my LCD and it work always fine.

Sacha Bellier.

prenticedavid, 2020-09-18:
First off.   Run the LCD_ID_readreg example.
Edit the defines to match the wiring that I posted for the BluePill.

If you do not get a proper readreg report then your wiring is wrong.
There is no point in trying any library until you get a good readreg report.

When I compile a sketch in the proper Arduino IDE I get a command line like this:
`"C:\\Users\\David Prentice\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino-1.8.12\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino-1.8.12\\hardware\\arduino\\avr\\variants\\standard" "-IC:\\Program Files (x86)\\Arduino-1.8.12\\hardware\\arduino\\avr\\libraries\\Wire\\src" "-IC:\\Program Files (x86)\\Arduino-1.8.12\\hardware\\arduino\\avr\\libraries\\SPI\\src" "-IC:\\Users\\David Prentice\\Documents\\Arduino\\libraries\\I2C_graphical_LCD_display-master" "C:\\Users\\DAVIDP~1\\AppData\\Local\\Temp\\arduino_build_775170\\sketch\\LCD_Demo_Using_I2C.ino.cpp" -o "C:\\Users\\DAVIDP~1\\AppData\\Local\\Temp\\arduino_build_775170\\sketch\\LCD_Demo_Using_I2C.ino.cpp.o"
`

If I look at the -Dxxx arguments 
`
-DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR`

Please post the default -Dxxx arguments from your command line.  (remove any that you have added)

I presume that you are using the official STM32 core from STMicroElectronics.

Why are you using PlatformIO instead of the proper Arduino IDE ?

David.

Magix35, 2020-09-18:
Thank you for yours replied.

I just found an error in my wiring (I don't know where but the system work fine with a new shield).

"Please post the default -Dxxx arguments from your command line."  
I don't found the same command line like you but I found this define block in the c_cpp_properties.json (I have remove all that I had in build_flags zone)

[image]

If I don't add -DARDUINO_BLUEPILL_F103CB in build_flags in platformio.ini the code doesn't compile because marco ARDUINO_BLUEPILL_F103CB is not defined. (macro ARDUINO_ARCH_STM32 is well defined in the c_cpp_properties.json)

Finaly, to compile the code my platformio.ini must be like that.

[image]

"Why are you using PlatformIO instead of the proper Arduino IDE ?": I prefer PlatformIO than the proper Arduino IDE because PlatformIO is on Visual Studio Code that give C++ IntelliSense. And I think that PlatformIO has a better library managment (at least when we doesn't have macros definition problems) because each project has a copy of the library (that is useful when we use library like your library MCUFRIEND_kbv because for each project we can have different settings (like LCD support)).

Thank for yours help and to replied that fast. And thank you for publishing your libraries.

Sacha Bellier

prenticedavid, 2020-09-18:
The good news is that `ARDUINO_ARCH_STM32` shows that you are using the official STM32 Core.

mcufriend_shield selects BluePill like this: 
```
//              weird MAPLE CORE                      STM32 CORE                        STM32CORE
#elif defined(ARDUINO_GENERIC_STM32F103C) || defined(ARDUINO_BLUEPILL_F103C8) || defined(ARDUINO_BLUEPILL_F103CB)
#warning Uno Shield on BLUEPILL

```

I had assumed that you would select BluePill as target in the IDE.   It looks as if you selected Generic F103CB instead of BluePill F103CB 

Perhaps I should check for `ARDUINO_GENERIC_STM32F103C8` and `ARDUINO_GENERIC_STM32F103CB` as well as the other defines.

Please run the LCD_ID_readreg.ino sketch with appropriate defines.

Oh,   if you want to post text,  please copy-paste as text instead of screenshot images.

David.

Magix35, 2020-09-18:
The LCD_ID_readreg.ino give me 0x4747 (for my 2.4" LCD Screen seem to be a HX8347 driver) with both arduino uno and STM32F103CBT6 with both Arduino IDE and PlatformIO. 

In PlatformIO there are not .ino file but only .cpp / .c / .h file. We have some change to make between Arduino IDE and PlatformIO.

"The good news is that ARDUINO_ARCH_STM32 shows that you are using the official STM32 Core." Yes I use the official STM32 Core. (It is stock on PlatformIO).

Sacha Bellier

prenticedavid, 2020-09-18:
Your LCD_ID_readreg is working ok.

`HX8347-D 240x320  ID=0x4747 #define SUPPORT_8347D
`

Your TFT is not one of the default controllers.  You need to enable the `#define SUPPORT_8347D`

If you had run the diagnose_TFT_controller.ino example it would have told you whether 0x4747 was not supported.

David.

Magix35, 2020-09-18:
"#define SUPPORT_8347D" Yes I did that all work fine I'm able to run the GLUE_Demo_320x240.

Thank for all of your help

Sacha Bellier

closed 2020-09-18 by Magix35