prenticedavid/MCUFRIEND_kbv #156: R32 with ILI9481

dmagyar, 2020-10-11:
Hi!

First of thanks for the awesome lib! I have a cheapo 480x320 3.5" module from ali, which I was able to get working with my Arduino Mega just fine.  Touchscreen pins were defined as the below:

const int XP = 8, XM = A2, YP = A3, YM = 9;

When I try the same screen with my Wemos R32 (after making the hw mods) and translating the pins above to the R32s PINs, the LCD screen works but the touch does not:

const int XP = 12, XM = 35, YP = 34, YM = 13;

I checked and these should be the right pins I can also measure ~300Ohms on X and ~600Ohms on the Y direction.
Not sure what to check at this point, any pointers are greatly appreciated.

prenticedavid, 2020-10-11:
Please post a link to the actual screen that you bought.

Uno shields are read-write.   And have raw Resistive panel that shares pins with the TFT.
If `const int XP = 8, XM = A2, YP = A3, YM = 9;` was the result from the Calibration sketch on the Mega.

`const int XP = 12, XM = 15, YP = 33, YM = 13;`

Put ` int XP = 12, XM = 15, YP = 33, YM = 13;` into the ESP32 Calibration sketch and you should get a "better" accuracy on the ESP32.

David.

dmagyar, 2020-10-11:
Unfortunately the Toucch native sketch still does not recognize me touching the screen :(
I bought the ILI9841 IPS with Touch version from here:

https://nl.aliexpress.com/item/1005001297986303.html?spm=a2g0s.9042311.0.0.27424c4dth7Dbt

prenticedavid, 2020-10-12:
That particular "Shield" will not work properly.
It uses series resistors instead of 74LVC245 level-shifter buffers.

The Touch panel will cripple any read function.
Mind you,  the Resistive Touch should work.

I am amazed by the lengths that the Chinese go to to save costs.   (without thinking of the consequences)

At the same time,  I would expect GitHub members to notice obvious differences in pcb before purchase.

I would be interested in how LCD_ID_readreg.ino behaves on a Uno, Mega, ...

I must have tried Touch on ESP32 at some time in the past:
```
MCUFRIEND_kbv ID=0x9488  320 x 480      Red 3.95 inch Seeeduino
const int XP=6,XM=A2,YP=A1,YM=7; //320x480 ID=0x9488
const int TS_LEFT=854,TS_RT=147,TS_TOP=748,TS_BOT=71;

MCUFRIEND_kbv ID=0x9488  320 x 480      Red 3.95 on ESP32
const int XP=27,XM=15,YP=4,YM=14; //320x480 ID=0x9488
const int TS_LEFT=858,TS_RT=105,TS_TOP=923,TS_BOT=13;

```

Note that this was a 6, A2, A1, 7 screen.   I can probably dig out a 8,A3,A2,9 screen if I have time.

David.

dmagyar, 2020-10-12:
Hi! It comes up with id 0x101 but after I force a 0x9481 ID the display comes up and behaves normally (it needs to be inverted but it's fine). Touch does not work on the R32. On the Mega everything works like a charm - button_simple, I can tap the buttons and it just works. Did not know about those resistors but given the fact they work with the Mega I thought I just have some mismatch with the pins. Output of LCD_ID_readreg.ino:

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 01 01       ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 01 01 01 01 Manufacturer ID
reg(0x0009) 01 01 01 01 01      Status Register
reg(0x000A) 01 09       Get Power Mode
reg(0x000C) 01 65       Get Pixel Format
reg(0x0061) 01 01       RDID1 HX8347-G
reg(0x0062) 01 01       RDID2 HX8347-G
reg(0x0063) 01 01       RDID3 HX8347-G
reg(0x0064) 01 01       RDID1 HX8347-A
reg(0x0065) 01 01       RDID2 HX8347-A
reg(0x0066) 01 01       RDID3 HX8347-A
reg(0x0067) 01 01       RDID Himax HX8347-A
reg(0x0070) 01 01       Panel Himax HX8347-A
reg(0x00A1) 01 01 01 01 01      RD_DDB SSD1963
reg(0x00B0) 01 01       RGB Interface Signal Control
reg(0x00B4) 01 01       Inversion Control
reg(0x00B6) 01 01 01 01 01      Display Control
reg(0x00B7) 01 01       Entry Mode Set
reg(0x00BF) 01 01 05 95 81 FD   ILI9481, HX8357-B
reg(0x00C0) 01 11 39 01 01 11 01 01 01  Panel Control
reg(0x00C8) 01 01 01 01 01 01 01 01 01 01 01 01 01      GAMMA
reg(0x00CC) 01 01       Panel Control
reg(0x00D0) 01 01 41    Power Control
reg(0x00D2) 01 01 21 01 01      NVM Read
reg(0x00D3) 01 01 21 01 ILI9341, ILI9488
reg(0x00D4) 01 01 21 01 Novatek ID
reg(0x00DA) 01 01       RDID1
reg(0x00DB) 01 01       RDID2
reg(0x00DC) 01 01       RDID3
reg(0x00E0) 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01     GAMMA-P
reg(0x00E1) 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01 01     GAMMA-N
reg(0x00EF) 01 01 01 01 01 01   ILI9327
reg(0x00F2) 01 01 31 01 01 01 01 01 01 01 01 01 Adjust Control 2
reg(0x00F6) 01 81 81 81 Interface Control
```

prenticedavid, 2020-10-12:
Yes, you can see that the Touch Panel on D8 is holding bit#0 high :

`reg(0x00BF) 01 01 05 95 81 FD   ILI9481, HX8357-B
`

So it makes all Read functions inoperable.
If you had bought the version without a Touch panel,  reads would probably be ok.

Hey-ho.   More apps will use Touch than want to read GRAM.   You are already resigned to readID() failure.

David.

dmagyar, 2020-10-12:
I'm not sure what this means - I can't read GRAM, but I can use the LCD as write-only all of the sketches work just fine. 
Why do I need the read feature? My main problem is with the touch. I can "sense" it with my ohmmeter, and my Mega works just fine. Sorry for the trouble but can you explain why I can't read the touch panel even with your suggested pins? 

I even validated the pins with dupont cables (yes I know, it's just a test :)) and I'm defining the right pins in the sketch.

dmagyar, 2020-10-13:
If you can point me to a shield that has better support (preferably SPI) I'll just go and buy new ones :(

prenticedavid, 2020-10-13:
Yes, you can see that the Touch Panel on D8 is holding bit#0 high :

`reg(0x00BF) 01 01 05 95 81 FD   ILI9481, HX8357-B
`

So it makes all Read functions inoperable.
If you had bought the version without a Touch panel,  reads would probably be ok.

Hey-ho.   More apps will use Touch than want to read GRAM.   You are already resigned to readID() failure.

David.

> I'm not sure what this means - I can't read GRAM, but I can use the LCD as write-only all of the sketches work just fine.
> Why do I need the read feature? My main problem is with the touch. I can [...]

The whole point of this library is that it identifies the TFT controller i.e. by tft.readID()

I suggest that you draw a schematic of the Touch Panel and TFT shield.
Then apply Kirchoff's Law.

This will show why Read fails.   And Touch should work on your Mega.

I  might try an 8,A3,A2,9 screen if I have time.

I strongly recommend that you use a Red SPI display (**with** Touch Controller).   Bodmer supports them with TFT_eSPI.
Now you have got lots of spare GPIO pins.

dmagyar, 2020-10-13:
Something like this? I'm trying to be careful with my next purchase ;)

https://nl.aliexpress.com/item/32954240862.html?spm=a2g0o.productlist.0.0.4bb0123brrlcLe&algo_pvid=c8538317-4ff9-42f1-9992-c72c93fed88c&algo_expid=c8538317-4ff9-42f1-9992-c72c93fed88c-0&btsid=0b0a050116026186007941722ea47b&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

prenticedavid, 2020-10-13:
Yes,  but avoid ILI9488.   Buy ST7796S if you can.

closed 2021-08-24 by prenticedavid