prenticedavid/MCUFRIEND_kbv #160: Unknown HX8347 with ID 0x8585

hlide, 2020-12-14:
### Object:
Unknown ID for HX8347: https://fr.aliexpress.com/item/32905829057.html

### Context: 
I used an Arduino MEGA and Soft SPI upon pins 11,12 and 13

### Result of diagnose_TFT_support
Diagnose whether this controller is supported
There are FAQs in extras/mcufriend_how_to.txt

tft.readID() finds: ID = 0x8585

MCUFRIEND_kbv version: 2.9.9

This ID is not supported
look up ID in extras/mcufriend_how_to.txt
you may need to edit MCUFRIEND_kbv.cpp
to enable support for this ID
e.g. #define SUPPORT_8347D

New controllers appear on Ebay often
If your ID is not supported
run LCD_ID_readreg.ino from examples/
Copy-Paste the output from the Serial Terminal
to a message in Displays topic on Arduino Forum
or to Issues on GitHub

Note that OPEN-SMART boards have diff pinout
Edit the pin defines in LCD_ID_readreg to match
Edit mcufiend_shield.h for USE_SPECIAL
Edit mcufiend_special.h for USE_OPENSMART_SHIELD_PINOUT

### Attempt
Undefined SUPPORT_8347D doesn't work.

### Result of LCD_ID_readreg.ino on UNO:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 95 95	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 3F 3F 3F 3F 3F	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 01 01	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03 03	GAMMA-P
reg(0x00E1) 06 06 06 06 06 06 06 06 06 06 06 06 06 06 06 06	GAMMA-N
reg(0x00EF) 14 14 14 14 14 14	ILI9327
reg(0x00F2) 08 08 08 08 08 08 08 08 08 08 08 08	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2020-12-14:
You clearly have HX8347-I i.e. 0x9595

`reg(0x0000) 95 95 ID: ILI9320, ILI9325, ILI9335, ...`

tft.readID() finds: ID = 0x8585

seems to show that readID() is not working properly.

Please edit setup() in graphictest_kbv.ino to `tft.begin(0x9595);`

David.

hlide, 2020-12-14:
I undefined SUPPORT_8347D and proceed as you said with an UNO without forcing ID. Success and ID is exactly 0x9595. Thank you for confirming LCD is good. If I remove the shield sitting on the MEGA, its program runs fine and I get 0x9595 as well. So, I believe that shield which plugged all pin sockets of MEGA should have one pin bent inner was the issue with the bad reading since replacing that shield with extreme cautions about their pins leads to success as well.

Thank you very much for your assistance.

closed 2020-12-14 by hlide

prenticedavid, 2020-12-14:
I don't understand.   Plug one display shield into the Mega2560.   It is obvious if there is a bent pin.

You should only have one shield plugged into the Arduino.    But you can still attach extra electronics to the 18x2 header.

hlide, 2020-12-15:
Between Mega and the LCD shield, there is a custom shield to access a 40-pin test socket: I couldn't see the bent pin because it was inside not outside the MEGA.

[image]