prenticedavid/MCUFRIEND_kbv #161: Help with an unknown processor set (Flipped and color-inverted TFT LCD)

Starwhip, 2020-12-31:
I've been struggling to diagnose issues with a 2.4" TFT LCD shield, the colors are inverted and the screen is printing text in reverse. Calling tft.invertDisplay(1); fixes white <-> black and blue <-> green inversions but there are still color errors. I have not been able to fix the screen mirror so far, in any 0-7 orientation. 

After updating to the most recent MCUFRIEND library, the screen remains a solid white instead of displaying with inversion issues.

Here's the output of the LCD_ID_readreg.ino program:

```
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 E3 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 06	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 80	Inversion Control
reg(0x00B6) 00 0A 82 27 04	Display Control
reg(0x00B7) 00 77	Entry Mode Set
reg(0x00BF) 00 02 02 02 02 02	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 20 31 42 13 04 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 06	Panel Control
reg(0x00D0) 00 A4 B2	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 E3	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control
```

Looking at other threads, it seems like reg(0x0000) should have some kind of ID besides 00 00. 
I have a second TFT LCD with a very similar build, which displays fine, but also has a 00 00 reading of its ID register. This one which does not work looks to have a missing chip (8 pin footprint) on the back, however, this appears to be part of the SD card reader traces.

prenticedavid, 2020-12-31:
You have an E300 controller.  I would dearly love to get a datasheet.    Then I would support this device officially.

Please ask your Vendor for the information e.g. controller part number or Manufacturer.   Ideally with a datasheet.   
It would take the Vendor 10 minutes to email their Wholesale Supplier.   And I would keep all their customers happy.
I suggest that you ask them politely for the information.   And ask for your money back if they do not provide it.

Meanwhile read message at https://forum.arduino.cc/index.php?topic=699834.msg4703138#msg4703138

David.

Starwhip, 2020-12-31:
Thanks, those fixes did the trick. The screen was ordered by someone else, so I don't have the vendor info at the moment, but I will try to get that information soon.

prenticedavid, 2020-12-31:
It would be so easy for the shop to email their Wholesale Supplier.
Unfortunately few shops can be bothered. 

If I was the shopkeeper,  I would sooner ask for the technical information than refund an unhappy customer.

David.

ESA-Angel, 2021-01-27:
I bought an Elegoo 2.8 "(320x240) TFT screen in amazon, the text was reversed
The solution was to edit the MCUFRIEND_kbv.cpp file and in the case 0xE300 line remove the INVERT_SS option

prenticedavid, 2021-01-27:
Please email your Supplier.   Ask for the Manufacturer and Part number of the 0xE300 controller.    Or preferably a datasheet.

Incidentally,  I added support for the 0xE300 which you should find in the current Beta (i.e. the master Branch)

I don't have this controller.   I rely on feedback from users.

David.

prenticedavid, 2021-01-29:
Oops.  My E300 edit had killed ST7781 and probably many other controllers.
I have just corrected it.   Please let me know if the current "master" Branch runs ok with your E300 shield.

David.

closed 2021-08-24 by prenticedavid