prenticedavid/MCUFRIEND_kbv #162: mcufriend_shield.h > Atmega328PB > ties up complete port C

ZL1CVD, 2021-01-04:
I am trying to use PC5 (A5, pin 28) on Atmega328PB as input. But I think mcufriend_shield.h is conflicting. Here is the section from my   mcufriend_shield.h 
`
#elif defined(__AVR_ATmega328P__) || defined(__AVR_ATmega328PB__)      //regular UNO shield on UNO
#define RD_PORT PORTC
#define RD_PIN  0
#define WR_PORT PORTC
#define WR_PIN  1
#define CD_PORT PORTC
#define CD_PIN  2
#define CS_PORT PORTC
#define CS_PIN  3
#define RESET_PORT PORTC
//#define RESET_PIN  4    //ZL1CVD - Atmega328P only 
#define RESET_PIN  6      //ZL1CVD - Atmega328PB only 

#define BMASK         0x03              //more intuitive style for mixed Ports
#define DMASK         0xFC              //does exactly the same as previous
#define write_8(x)    { PORTB = (PORTB & ~BMASK) | ((x) & BMASK); PORTD = (PORTD & ~DMASK) | ((x) & DMASK); }
#define read_8()      ( (PINB & BMASK) | (PIND & DMASK) )
#define setWriteDir() { DDRB |=  BMASK; DDRD |=  DMASK; }
#define setReadDir()  { DDRB &= ~BMASK; DDRD &= ~DMASK; }
#define write8(x)     { write_8(x); WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }
#define PIN_LOW(p, b)        (p) &= ~(1<<(b))
#define PIN_HIGH(p, b)       (p) |= (1<<(b))
#define PIN_OUTPUT(p, b)     *(&p-1) |= (1<<(b))

`
Any ideas?

prenticedavid, 2021-01-04:
PC5 is always unused on a Mcufriend Uno Shield.   You can use it for whatever you want. 

Your macros have selected a non-existent PC6

It looks as if you can use PE2 as GPIO as well as ADC6 on a mega328PB
On the mega328P ADC6 is only available as an Analog pin ADC6 on the TQFP-32 package.

Obviously you have to hand-wire the LCD_RST pin otherwise the Shield connects to PC4 (SDA)

This will release PC4, PC5 for I2C bus. 

David.

ZL1CVD, 2021-01-04:
OK, so I'm using a custom PCB with only a 328PB TQFP soldered directly to it.  Circuit snippet is here:
[image]

Lots of these are already in the field. I am only now looking to activate another software feature by sensing PC5 state, in the same way I'm already sensing PE0 & PE1, but have run into issues and thought on quick glance that mcufriend_shield.h was the issue :(

prenticedavid, 2021-01-05:
Ah-ha.  You have defined PC6 for TFT_RST.    And connected TFT_RST to the hardware Reset pin.
Since PC6 is the AVR RESET pin nothing will happen when MCUFRIEND_kbv wiggles TFT_RST in software.
AVR_RESET has a weak pullup.  You normally have a push button to GND.   In which case both AVR and TFT controller will receive the same reset pulse.   And both receive the same Power-On-Reset
If you change fuses i.e. RSTDISBL PC6 can be used as GPIO but you kill any re-programming or debugging.

Personally I would **never** change RSTDISBL.   I would define TFT_RST as a non-existent port pin e.g. PC7.   That would ensure no confusion.

PC4, PC5 would behave as regular GPIO pins unless you use Arduino Analog or Wire functions.

David.

ZL1CVD, 2021-01-05:
The 328PB RESET can also be defined as I/O. I had issues with the standard 0.1uF / 10K pullup not actually resetting some displays 100% of the time. It didn't matter how often I sent reset delay reset, etc in SW, those with a dodgy reset line would just stay white.  All sorted with this solution. Have also sorted this issue. It was elsewhere in the code + the 0.1uF filter cap cap was holding the EEPROM data for 5 seconds. I expected the change to be instant, wasn't. 
Thanks so much for a fantastic library BTW.

closed 2021-01-05 by ZL1CVD