prenticedavid/MCUFRIEND_kbv #165: HX8347 lcd display not working

elmahdi-dev, 2021-02-18:
Hello, I'm actually trying to use a display I bought years ago, I don't have any reliable link to pin point the exact display I have but I'll try to give you all the informations I can find. 

**LCD Display**
[image]
[image]

When I use an arduino and upload the graphictest_kbv script the lcd is blank, but when I use a mega this is what I see:
[image]

**What I have tried**
I enabled
> #define SUPPORT_8347D             //HX8347-D, HX8347-G, HX8347-I, HX8367-A +520 bytes, 0.27s
#define SUPPORT_8347A  

I tried many libraries without getting the screen to even change one pixel's color...

**Output of readReg**

> Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)
reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 3F 3F 3F 3F 3F	Status Register
reg(0x000A) 00 00	Get Power Mode
reg(0x000C) 01 01	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 80 80	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 47 47	RDID Himax HX8347-A
reg(0x0070) 06 06	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2021-02-18:
Your PCB is clearly marked.   HX8347
Please don't dismantle shields.   The double-sided sticky tape can be difficult to unstick.

Thanks for the LCD_ID_readreg.ino report.   It looks like a HX8347-A.
#define SUPPORT_8347A 
  
readID() should return 0x8347

diagnose_TFT_controller.ino should report 0x8347 to Serial and display something.

I have never seen an HX8347-A.    Nor has anyone ever reported having one.

So I would be very interested in getting feedback from you.   e.g. Serial output from diagnose_TFT_controller.ino

David.

elmahdi-lab, 2021-02-18:
@prenticedavid Thanks for your quick answer.

As for the diagnose output here it is:
> Diagnose whether this controller is supported
There are FAQs in extras/mcufriend_how_to.txt
tft.readID() finds: ID = 0x8347
MCUFRIEND_kbv version: 2.9.9
PORTRAIT is 240 x 320
Run the examples/graphictest_kbv sketch
All colours, text, directions, rotations, scrolls
should work.  If there is a problem,  make notes on paper
Post accurate description of problem to Forum
Or post a link to a video (or photos)
I rely on good information from remote users

As far as the display, I finally see something with the line you gave me, but it seems the image is inverted as well as the color.

elmahdi-lab, 2021-02-18:
**Here is the actual image output :**
[image]

prenticedavid, 2021-02-18:
Look to the end of the case 0x8347 block.

```
        // Atmel ASF code uses VCOM2-3: 0x38, 0x12. 50ms delays and no TEST mode changes.
        init_table(HX8347A_NHD_regValues, sizeof(HX8347A_NHD_regValues));
        //        init_table(HX8347A_CMO32_regValues, sizeof(HX8347A_CMO32_regValues));
        //        init_table(HX8347A_CMO24_regValues, sizeof(HX8347A_CMO24_regValues));
        //        init_table(HX8347A_ITDB02_regValues, sizeof(HX8347A_ITDB02_regValues));
        //        init_table(HX8347G_2_regValues, sizeof(HX8347G_2_regValues));
        break;
```

Try each line in turn.   e.g. comment the NHD line.   uncomment the CMO24 line.

Don't worry about vertical-mirror or horizontal-mirror.    Just compare the colours,  clear picture, ...

It is always best to run the graphictest_kbv example.
You can identify wrong colour, inverted colours, ..., scroll directions,  text directions, ... from the Penguin screens.

Take notes on paper.   Report back.
Or just take photos of Penguin screens.
Or take a short video.   Post link to video.

David.

elmahdi-lab, 2021-02-18:
I've tried one line after the other but they all gave the same result, everything seems  light green and blue only with inverted image.

The last line gave this error:
> C:\Users\Mehdi\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp: In member function 'void MCUFRIEND_kbv::begin(uint16_t)':
C:\Users\Mehdi\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:2064:28: error: 'HX8347G_2_regValues' was not declared in this scope
                 init_table(HX8347G_2_regValues, sizeof(HX8347G_2_regValues));
                            ^~~~~~~~~~~~~~~~~~~
C:\Users\Mehdi\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:2064:28: note: suggested alternative: 'HX8347A_NHD_regValues'
                 init_table(HX8347G_2_regValues, sizeof(HX8347G_2_regValues));
                            ^~~~~~~~~~~~~~~~~~~
                            HX8347A_NHD_regValues
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

prenticedavid, 2021-02-18:
Go back to the NHD table.   Remove the comment on the (0x70) line e.g.

```
        static const uint8_t HX8347A_NHD_regValues[] PROGMEM = {
            //Gamma Setting NHD
            (0x46), 12, 0x94, 0x41, 0x00, 0x33, 0x23, 0x45, 0x44, 0x77, 0x12, 0xCC, 0x46, 0x82,
            (0x01), 1, 0x06,    //Display Mode [06]
            (0x16), 1, 0xC8,    //MADCTL [00] MY=1, MX=1, BGR=1
            (0x70), 1, 0x05,    //Panel [06] 16-bit REMOVE COMMENT
```
That should give you 16-bit colours.    Please report back with the Penguin errors.

David.

elmahdi-lab, 2021-02-19:
@prenticedavid I did the modification, same result, the screen seems fast and fluid, the only issue is the color and image inversion. What are the penguin errors?

prenticedavid, 2021-02-19:
First off.    Are you comfortable with Forking GitHub projects and changing Branches etc ?

I can give you step by step instructions for regular IDE install.   Then minor edits to MCUFRIEND_kbv.cpp file

But I need accurate reports back from you.

graphictest_kbv.ino is the standard example sketch that comes with the library.

it displays several Adafruit graphics and reports the times

displays a named coloured band, some text and a Penguin.
These "penguin" screens show obvious problems e.g. The Band says BLUE but shows red or yellow or cyan
e.g. the text is vertical-mirror or horizontal-mirror
e.g. the Scroll says UP but shows down

I would expect the` is8347 = 1` edit and the `0x70` edit to give you proper colours (but the colours might be wrong)
If not,  please SAY SO.

David.

elmahdi-lab, 2021-02-19:
I am confortable enough to use git, so if you have any specific instructions I will be able to follow them. I will try my best to give you accurate feedback with pictures.

I will fork the project while I am waiting for your instructions.

Thank you very much.

prenticedavid, 2021-02-19:
You can just stick with the Release i.e. with these edits.

```
#define SUPPORT_8347A             //HX8347-A +500 bytes, 0.27s
...
        is8347 = 1;  //omitted this statement.  thanks elmahdi-lab
...
            (0x70), 1, 0x05,    //Panel [06] 16-bit REMOVE COMMENT

```

and run graphictest_kbv.ino from the examples.

Let me know if your colours are still bluish and washed out.
Report back with problems in graphictest_kbv.ino

If you want to Fork you really need to delete the existing Library Release.
Then change to the "test_8347A" Branch that I have just pushed on GitHub.

It might be safer to stick with the Release.

David.

prenticedavid, 2021-02-21:
Bump.

Let me know if your colours are still bluish and washed out.

David.

prenticedavid, 2021-02-25:
Bump again.

David.

elmahdi-lab, 2021-03-09:
Hi,

I am sorry for the delay. I've deleted and reinstalled the library with the suggested changes:

#define SUPPORT_8347A             //HX8347-A +500 bytes, 0.27s
...
        is8347 = 1;  //omitted this statement.  thanks elmahdi-lab
...
            (0x70), 1, 0x05,    //Panel [06] 16-bit REMOVE COMMENT

The result is a very responsible, all blue and green and text is inversed.

elmahdi-lab, 2021-03-09:
Hi,

I loaded graphictest_kbv.ino, I didn't see any penguin but here are multiple pictures of what was displayed:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

prenticedavid, 2021-03-09:
It looks as if you have still got 18-bit pixels.    reg(0x70) should have selected 16-bit pixels.
I think it is wise to start again i.e. with code where I have done ALL the edits.

Please can you follow these steps:

1.  go to https://github.com/prenticedavid/MCUFRIEND_kbv
2.  change Branch from "master" to "test_8347A"   (on left hand side of page opposite the Green [Code] button)
3.  select Green [Code] button.
4.  Download .ZIP
5.  save .ZIP on your PC

6.  leave the IDE.   
7.  Delete the MCUFRIEND_kbv directory.
8.  start the IDE
9.  Sketch->Include Library->Add .ZIP library (that you saved in (5)
10. File->Examples->MCUFRIEND_kbv:  select graphictest_kbv 

Please ask if you do not understand.    Photo #6 is a "Penguin" screen.    
One or Two photos are sufficient.    It took a very LONG time to receive your enormous files.

David.

elmahdi-lab, 2021-03-09:
Hi,
I've used your test branch, so the good news is that the text orientation is correct, but the colors seems be the same:

[image]

The purple dot in the center is a reflection of my phone camera ir sensor, so just ignore it.

prenticedavid, 2021-03-09:
I have altered the MCUFRIEND_kbv.cpp file on GitHub.

You can repeat the Download .ZIP procedure.
Or just apply my edits manually.   (no need to leave IDE etc )

I am sure that this is something very simple.    Other libraries support HX8347-A

David.

elmahdi-lab, 2021-03-09:
Hi, I see you added a condition with the variable is_hx8347, the problem is that it's not defined so the program does not compile. As there is a variable that is similar in the name 'is8347', i've used  edited the variable name just to test, and it compiled but the result in the display is the same, correct text orientation but colors are the same as my last image.

prenticedavid, 2021-03-09:
Yes,  that was my typo.   And you made the correction.

I am 100% convinced that your HX8347A is expecting 18-bit colour.
MCUFRIEND_kbv and almost all libraries work in 16-bit colour.

It is normally a User operation to set 16-bit or 18-bit colour mode.
And it is register 0x70 from the HX8347A data sheet that controls the CSEL bits,

HX8347-D, G, I use register 0x17 which is not present on a HX8347-A

Your display uses the 8-bit 8080-8 parallel interface which is set by hardware pins BS2-BS0 = 0b011 default 18-bit colour
Itead screens use the 16-bit 8080-16 parallel interface which is set by hardware pins BS2-BS0 = 0b000 default 16-bit colour

This accounts for why they do not specifically set 16-bit colour mode in software.

Please can you try writing a different value to reg(0x70) e.g. 0x25 instead of 0x05
`
            (0x70), 1, 0x25,    //see if 0x70 is a Protected register.
`
This should make horizontal mirror i.e. the text will be backwards.

Either reg(0x70) is locked or the datasheet has the wrong register.   (this is not unknown !! )

David.

elmahdi-lab, 2021-03-09:
I tried: 0x20, 0x21, 0x22, 0x23, 0x24, 0x25, 0x26, 0x27, 0x2A. Text is backward, color are still blue. I went to the datasheet, they say in order to have 16Bit colors BS0, BS1, BS2 should be all set to 0, so just to understand, the register 0x70 is the one responsible for these  values?

While editing the register, I noticed another line a little  up:
(0x70), 1, 0x05,    //Panel [06] 16-bit REMOVE COMMENT

prenticedavid, 2021-03-09:
That was the original statement.
I added the second statement to see if it needed to be inside the reg(0x57) pair

I have removed the Protected and is8347 edits.
And inserted a RGB_EN statement.   This _might_ change the colour mode.   But I expect that it will destroy any picture completely.

If it does destroy the picture,  uncomment the next RGB_EN line.

I am clutching at straws !

I don't know whether you are a native English speaker.   I should not use expressions like "clutching at straws"

The datasheet says about reg(0x70):
```
CSEL[2:0]:
This command is used to define the format of RGB picture data, which is to be
transferred via the RGB Interface. The formats are shown in the table
```

David.

elmahdi-lab, 2021-03-09:
I'm not a native English speaker, but I understand you very well ;-) ! I really appreciate your help! Thank you very much.

I tried your latest commit, unfortunately the LCD is all white, there's nothing happening like before.

prenticedavid, 2021-03-09:
Yes,  I expected a White screen.
Did you try uncommenting the other RGB_EN line ?
I would expect it to show the previous Blue-Green pictures.

It is my bedtime.   I will have a think in the morning.
I am beginning to think that this controller will only use 18-bit pixel colour with 8080-8 interface.

Yes,   I can display 18-bit colour in software but it slows things down.    I would prefer to configure 16-bit colour with the HX8347A registers. 
I only have HX8347-D controller which is VERY different.

David.

elmahdi-lab, 2021-03-09:
You are right, uncommenting RGB_EN returned the blue colors. Have a good night and thank you for your help!

prenticedavid, 2021-03-10:
One last chance before I give up.
Perhaps reg(0x72) defaults to 0x01.   Perhaps this will select 16-bit colour pixels.
I have put the change on GitHub.

To resolve the unknown power up default values of the HX8347A controller:
Please edit setup() in LCD_ID_readreg.ino from the examples.
Change this line 
from:

`
//    for (uint16_t i = 0; i < 256; i++) readReg(i, 7, "f.k"); 
`
to:

` 
   for (uint16_t i = 0; i < 256; i++) readReg(i, 2, "f.k"); 
`

Run the sketch.   Copy-paste the Serial Output to your message.   (or to a TXT file)

Perhaps I should have asked you to do this at the start !!

David.

elmahdi-lab, 2021-03-10:
Hi, this is the output of the sketch:

> Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)
reg(0x0000) 00 00	f.k
reg(0x0001) 02 02	f.k
reg(0x0002) 00 00	f.k
reg(0x0003) 00 00	f.k
reg(0x0004) 00 00	f.k
reg(0x0005) EF EF	f.k
reg(0x0006) 00 00	f.k
reg(0x0007) 00 00	f.k
reg(0x0008) 01 01	f.k
reg(0x0009) 3F 3F	f.k
reg(0x000A) 00 00	f.k
reg(0x000B) 00 00	f.k
reg(0x000C) 01 01	f.k
reg(0x000D) 3F 3F	f.k
reg(0x000E) 00 00	f.k
reg(0x000F) 00 00	f.k
reg(0x0010) 01 01	f.k
reg(0x0011) 40 40	f.k
reg(0x0012) 00 00	f.k
reg(0x0013) 00 00	f.k
reg(0x0014) 00 00	f.k
reg(0x0015) 00 00	f.k
reg(0x0016) 00 00	f.k
reg(0x0017) 00 00	f.k
reg(0x0018) 00 00	f.k
reg(0x0019) 48 48	f.k
reg(0x001A) 00 00	f.k
reg(0x001B) 09 09	f.k
reg(0x001C) 00 00	f.k
reg(0x001D) 05 05	f.k
reg(0x001E) 00 00	f.k
reg(0x001F) 04 04	f.k
reg(0x0020) 40 40	f.k
reg(0x0021) 10 10	f.k
reg(0x0022) FC 00	f.k
reg(0x0023) DC DC	f.k
reg(0x0024) DC DC	f.k
reg(0x0025) DC DC	f.k
reg(0x0026) 20 20	f.k
reg(0x0027) 02 02	f.k
reg(0x0028) 02 02	f.k
reg(0x0029) 02 02	f.k
reg(0x002A) 02 02	f.k
reg(0x002B) 00 00	f.k
reg(0x002C) 02 02	f.k
reg(0x002D) 02 02	f.k
reg(0x002E) 00 00	f.k
reg(0x002F) 06 06	f.k
reg(0x0030) 00 00	f.k
reg(0x0031) 00 00	f.k
reg(0x0032) 00 00	f.k
reg(0x0033) 00 00	f.k
reg(0x0034) 00 00	f.k
reg(0x0035) 19 19	f.k
reg(0x0036) 19 19	f.k
reg(0x0037) 22 22	f.k
reg(0x0038) 00 00	f.k
reg(0x0039) 00 00	f.k
reg(0x003A) 01 01	f.k
reg(0x003B) 00 00	f.k
reg(0x003C) 00 00	f.k
reg(0x003D) 00 00	f.k
reg(0x003E) 19 19	f.k
[... 202 lines cut ...]
reg(0x0065) 80 80	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 47 47	RDID Himax HX8347-A
reg(0x0070) 06 06	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 00 00 00 00 00 00 00 00	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 00 00	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

prenticedavid, 2021-03-10:
Thankyou for the Readreg report.

It looks as if reg(0x72) defaults to 0x00.   So my "last chance" does not work.
You could try writing 0x01 to reg(0x72).    The datasheet does not make sense in its TRI description.

From inspection of your Readreg values undocumented reg(0x2F) contains 0x06
I will attempt to write 0x05 which _might_ alter the pixel colour mode.

David.

closed 2021-08-24 by prenticedavid