prenticedavid/MCUFRIEND_kbv #166: Recent purchases Bangood of "red" 2.4" LCD + touchscreen.

Ray-electrotechie, 2021-02-28:
I have been investigating why the Bangood "red" LCD does not work out of the box with MCUFriend. After a baffling couple of hours I took to forcing various IDs into "begin" I found the one that worked and passed all the screen examples in your package was  ILI9341
The reason I had to do that was that MCUFriend failed to identify it as ILI9341 and returned code D3D3 and then set it to a chip which didn't work (even wrong resolution). I then checked all the readable registers of the ILI9341  using the "a-Si TFT LCD Single Chip Driver 240RGBx320 Resolution and 262K color Specification Version: V1.11" PDF file. None of the registers which identify the chip responded. The main one should of course have been D3. The chip seems to return the register address as its return parameters if the register is behaving as NOP i.e. asking for 4 bytes delivers D3 D3 D3 D3.

I then discovered the following statement in the spec "Note 1: Undefined commands are treated as NOP (00h) command.
**Note 2: B0 to D9 and DE to FF are for factory use of display supplier**. USER can decide if these commands are
available or they are treated as NOP (00h) commands before shipping to USER. Default value is NOP"

So, it would seem that whichever company produced the modules, they failed to enable any of the registers either those that identify the chip, or those which are writable by the module producer so nominally there are NO registers to identify the chip.

My apologies if you already know all this but I thought that it might save you some time. The only suggestion I can make is if D3 returns D3 D3 then would it be worth going through all the readable registers of the ILI9341 and check to ensure that all of them do NOT return their own address. If that passes, then perhaps there is  a chance that it really is an ILI9341 but of course, no guarantees. For example the ILI9342 looks very similar but it does have F7h which the 9341 doesn't so perhaps check that first.

Additionally it could be possible to attempt some innocent writes into status (say brightness) and make sure that they read back correctly.

All that being said I am both truly grateful for all the contributors hard work with MCUFriend and really impressed by what you have achieved. Please accept my apology and close this if you are executing another strategy to help people who have bought this LCD recently.
It has been an interesting few days working with this screen! My work is very amateurish but I have butchered your LCD_ID_readreg to only read registers which are readable on the ILI9341. It isn't hard to do but if you want the code I will post it.
Best wishes Ray

prenticedavid, 2021-02-28:
Please post the output from LCD_ID_readreg.ino

It is almost certainly **not** Ilitek ILI9341

I return 0xD3D3 for write-only displays.   Which means that you have to specify the controller ID manually.

LCD_ID_readnew.ino can write values to unlock the Manufacturer registers.    But the typical ID registers are normally readable.   You only need to "unlock" when accessing the Power registers.

David.

Ray-electrotechie, 2021-02-28:
Certainly. Here it is.
[readreg.txt](https://github.com/prenticedavid/MCUFRIEND_kbv/files/6057537/readreg.txt)

Ray-electrotechie, 2021-02-28:
And for completeness, the results of readnew.
[readnew.txt](https://github.com/prenticedavid/MCUFRIEND_kbv/files/6057578/readnew.txt)
Just for your info, this is on an Arduino Mega. I have two of these displays, both of which behave identically, the other currently plugged into an Arduino Uno.

prenticedavid, 2021-02-28:
You have the same Red Shield as this current message : https://forum.arduino.cc/index.php?topic=729776.msg4910611#msg4910611

There is no point in running readnew without saying which "unlock" sequence you have tried.
There is no point in posting anything unless you have revealed different registers.

First step is to uncomment this line in setup()
`
    //    for (uint16_t i = 0x00; i <= 0xFE; i++) readReg(i, 10, "f.k");
`
An un-implemented or locked register will return its address e.g. reg(0xD3) returns D3 D3 D3 ...
Since the Touch panel is connected to LCD_D6, LCD_D7 you will possibly find that bit6, bit7 read a fixed value on an "un-implemented" register.  e.g. reg(0xB4) returns F4 F4 F4 ...  

If you get something interesting from reading "all" registers,   SAY SO.
Otherwise,  repeat with un-commenting one unlock statement at a time e.g.
`
    //    unlock = unlock_1520;
`
If you reveal something SAY SO and WHICH unlock statement.

You appear to be a GitHub member living in the UK.
So we can probably run through "the usual steps" within a few minutes.

Note that User registers like 0x2E are going to be readable.   It is Manufacturer registers above 0x80 that I would like to see.

David.

Ray-electrotechie, 2021-02-28:
>  But the typical ID registers are normally readable. You only need to "unlock" when accessing the Power registers.

Yes, in my case 0x4 is readable and returns 0,0,0,0. On the ILI9341 (and I presume other chips) ID1, ID2 and ID3 can be written (or not as the case may be) by the module producer using command "NV Memory Write (D0h)" (I am just quoting the spec here, I have not tried to do it).

> There is no point in running readnew without saying which "unlock" sequence you have tried.
> There is no point in posting anything unless you have revealed different registers.

OK, I admit, I did not read anything about readnew, I just ran it. Here is the output from lcd_ID_readreg with the comment removed from the for loop. Finding "something interesting" is a bit tricky, I am new to this game. I will uncomment the unlocks as you requested, but unfortunately tomorrow. From my perspective, the only thing that I wasn't expecting from my original hypothesis of ili9341  or ILI9342 is that 68h responds thus it isn't either of those as you wrote, since neither define 68h as a valid read command. Otherwise all other read commands correspond as expected with the spec - though the lack of response to F7h rules out ILI9342 in any case.

I will try forcing the screen as ILI9338 (as in the URL you gave) and run all the examples tomorrow. I have downloaded the spec of ILI9338B: that also does not have a command 68h.
You are right, I am English and in the UK at the moment, trapped by COVID restrictions. I used to spend half my time in France.

Regards, Ray 

Meanwhile, below is the result of running readreg with the for loop:
Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) C0 C0 C0 C0 C0 C0 C0	f.k
reg(0x0001) C1 C1 C1 C1 C1 C1 C1	f.k
reg(0x0002) C2 C2 C2 C2 C2 C2 C2	f.k
reg(0x0003) C3 C3 C3 C3 C3 C3 C3	f.k
reg(0x0004) 00 00 00 00 00 00 00	f.k
reg(0x0005) C5 C5 C5 C5 C5 C5 C5	f.k
reg(0x0006) C6 C6 C6 C6 C6 C6 C6	f.k
reg(0x0007) C7 C7 C7 C7 C7 C7 C7	f.k
reg(0x0008) C8 C8 C8 C8 C8 C8 C8	f.k
reg(0x0009) 00 00 61 00 00 00 00	f.k
reg(0x000A) 08 08 08 08 08 08 08	f.k
reg(0x000B) 00 00 00 00 00 00 00	f.k
reg(0x000C) 06 06 06 06 06 06 06	f.k
reg(0x000D) 00 00 00 00 00 00 00	f.k
reg(0x000E) 00 00 00 00 00 00 00	f.k
reg(0x000F) 00 00 00 00 00 00 00	f.k
reg(0x0010) D0 D0 D0 D0 D0 D0 D0	f.k
reg(0x0011) D1 D1 D1 D1 D1 D1 D1	f.k
reg(0x0012) D2 D2 D2 D2 D2 D2 D2	f.k
reg(0x0013) D3 D3 D3 D3 D3 D3 D3	f.k
reg(0x0014) D4 D4 D4 D4 D4 D4 D4	f.k
reg(0x0015) D5 D5 D5 D5 D5 D5 D5	f.k
reg(0x0016) D7 D7 D7 D7 D7 D7 D7	f.k
reg(0x0017) D7 D7 D7 D7 D7 D7 D7	f.k
reg(0x0018) D8 D8 D8 D8 D8 D8 D8	f.k
reg(0x0019) D9 D9 D9 D9 D9 D9 D9	f.k
reg(0x001A) DA DA DA DA DA DA DA	f.k
reg(0x001B) DB DB DB DB DB DB DB	f.k
reg(0x001C) DC DC DC DC DC DC DC	f.k
reg(0x001D) DD DD DD DD DD DD DD	f.k
reg(0x001E) DE DE DE DE DE DE DE	f.k
reg(0x001F) DF DF DF DF DF DF DF	f.k
reg(0x0020) E0 E0 E0 E0 E0 E0 E0	f.k
reg(0x0021) E1 E1 E1 E1 E1 E1 E1	f.k
reg(0x0022) E2 E2 E2 E2 E2 E2 E2	f.k
reg(0x0023) E3 E3 E3 E3 E3 E3 E3	f.k
reg(0x0024) E4 E4 E4 E4 E4 E4 E4	f.k
reg(0x0025) E5 E5 E5 E5 E5 E5 E5	f.k
reg(0x0026) E6 E6 E6 E6 E6 E6 E6	f.k
reg(0x0027) E7 E7 E7 E7 E7 E7 E7	f.k
reg(0x0028) E8 E8 E8 E8 E8 E8 E8	f.k
reg(0x0029) E9 E9 E9 E9 E9 E9 E9	f.k
reg(0x002A) EA EA EA EA EA EA EA	f.k
reg(0x002B) EB EB EB EB EB EB EB	f.k
reg(0x002C) EC EC EC EC EC EC EC	f.k
reg(0x002D) ED ED ED ED ED ED ED	f.k
reg(0x002E) 00 FC FC FC FC FC FC	f.k
reg(0x002F) EF EF EF EF EF EF EF	f.k
reg(0x0030) F0 F0 F0 F0 F0 F0 F0	f.k
[... 217 lines cut ...]
reg(0x0065) E5 E5	RDID2 HX8347-A
reg(0x0066) E6 E6	RDID3 HX8347-A
reg(0x0067) E7 E7	RDID Himax HX8347-A
reg(0x0070) F0 F0	Panel Himax HX8347-A
reg(0x00A1) E1 E1 E1 E1 E1	RD_DDB SSD1963
reg(0x00B0) F0 F0	RGB Interface Signal Control
reg(0x00B4) F4 F4	Inversion Control
reg(0x00B6) F6 F6 F6 F6 F6	Display Control
reg(0x00B7) F7 F7	Entry Mode Set
reg(0x00BF) FF FF FF FF FF FF	ILI9481, HX8357-B
reg(0x00C0) C0 C0 C0 C0 C0 C0 C0 C0 C0	Panel Control
reg(0x00C8) C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8 C8	GAMMA
reg(0x00CC) CC CC	Panel Control
reg(0x00D0) D0 D0 D0	Power Control
reg(0x00D2) D2 D2 D2 D2 D2	NVM Read
reg(0x00D3) D3 D3 D3 D3	ILI9341, ILI9488
reg(0x00D4) D4 D4 D4 D4	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0 E0	GAMMA-P
reg(0x00E1) E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1 E1	GAMMA-N
reg(0x00EF) EF EF EF EF EF EF	ILI9327
reg(0x00F2) F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2 F2	Adjust Control 2
reg(0x00F6) F6 F6 F6 F6	Interface Control

prenticedavid, 2021-02-28:
Well,  that was a bit pointless.   As expected User registers 04, 09, 0A, 0B, 0C, 0D, 0E, 0F, 2E, 52, 56, 5F, DA, DB, DC are "active".

All the others read as garbage  (as I explained)

Repeat with different unlock statements.    If you discover anything different,  say so.
If you find that unlock_1520 and unlock_1526 make no difference,   please say so.

I doubt that the other unlock statements are worth trying.

David.

Ray-electrotechie, 2021-03-01:
Hello. I removed the comment from each of the unlock statements LC D_ID_readnew, one at a time. I kept the results of each in a text file and ran Ultracompare. All are identical. To be clear, I definitely included the 1520 and 1526 - they make no difference.
Regards, Ray

prenticedavid, 2021-03-01:
In which case I am stuck.

Most controllers will show an ID.   Even if you need a special unlock sequence to access the Manufacturer registers.
Not all MIPI-compliant controllers support 0x68 (Read Adaptive Brightness Control Content)
There is nothing unique about reg(0x04) matching DA, DB, DC content (or not).

So I can't find anything to distinguish your device.
However I can stop it returning 0xD3D3

I suggest that you request the controller make and model number from BangGood.
Quote your order number, date etc.

I could send you a sketch that writes to an ILI9341 style controller e.g. to invert panel, change directions, ...
If your controller uses the same registers it may or may not respond e.g. if locked

David.

Ray-electrotechie, 2021-03-01:
David, thank you for your efforts. I spent some time looking at the code and understand where the D3D3 comes from.
It is unfortunate that there is nothing that can distinguish this from another controller. I don't really expect anything at all from Bangood since they always say "or equivalent" anyway and they sell pretty well anything so the chance of that level of support I believe is low - but so was the price! This is quite an old design for a phone with icons across the bottom which are not illuminated by the backlight - just in the range of the touch screen. After a few more experiments I think I might just give up and buy a better screen with a known controller on it (hopefully).
My application is pretty simple, I am just showing a few numbers and a time/temperature graph, so the fact that the examples worked pretty well when I told the system it was and ILI9341 gives me confidence that my simple app may just work, even with the incorrect driver
Keep up the good work, there must be thousands of people beavering away with arduino and various screens - it is good to know that there are experts like you backing them up. Please close the issue now, I shall not bother you further.

prenticedavid, 2021-03-01:
Please email BangGood.

I share your scepticism but Chinese companies will never correct these things if no one tells them !!

Thanks for the feedback anyway.   Since I can tell that this is a read-write controller I need to distinguish from the write-only controller boards.    (Mega2560 shields tend to be write-only)

David.

Ray-electrotechie, 2021-03-02:
David, well, I have been round and round the Bangood website. It does say "contact us" and it does have "email" but before you can choose email, you have to select a question e.g. "need product guide". Then you have to choose what you expect to happen between the following: returning the product for a refund, return for repair or return for replacement. Note that "get a product guide" isn't listed despite that being the original choice. Since I do not wish to return anything, I am disinclined to tick any, but of course they are mandatory. This sort of crap irritates the hell out of me. https://uk.banggood.com/tickets-create.html?label_id=78  I deeply suspect that it means they have no intention of giving out any technical details or, more likely, they have nobody who knowns much. They are like the "China stores" that you see all around Spain. Those sell everything from fashion, to furniture, electronics, hardware,toys, porcelain - all in one giant store. The only thing missing is food.
Bangood's facebook page is full of coffee machines, shoes, bicycles etc. etc. I could post there, but I think unlikely to be seen by anyone who has the slightest idea about what I would write. There is a "cservice@bangood.com" noted, but whether anyone is listening....

prenticedavid, 2021-03-02:
First off.  You have to open an account.   Which you might have already done with the original purchase.

Open the item page.
Select CustomerQandAs
Select See All Questions and Answers
Select Ask a New Question

They probably acknowledge the question.   Unlikely that you get a reply.
I don't like to be rude to people,   But if you get no response you can give a 1-Star Review that says the Red Shield does not return a correct ID.

Anyway,   please edit MCUFRIEND_kbv.cpp readID() function in the v2.9.9-Release:
```
    ret32 = readReg32(0xD3);      //[xx 91 63 00]
    ret2 = ret;   // >>>> remember from reg(04) <<<<
    ret = ret32 >> 8;
    if (ret == 0x9163) return ret;
    ret = readReg32(0xD3);      //for ILI9488, 9486, 9340, 9341
    if (ret == 0x3229) return ret;
    msb = ret >> 8;
    if (msb == 0x93 || msb == 0x94 || msb == 0x98 || msb == 0x77 || msb == 0x16)
        return ret;             //0x9488, 9486, 9340, 9341, 7796
//    if (ret == 0x00D3 || ret == 0xD3D3)
//        return ret;             //16-bit write-only bus
      if (ret == 0xD3D3 && ret2 == 0x0404) return ret; // >>> new test <<<
```

This should make readID() return reg(0x00) which in your case is probably 0x0000
You will still need to force the ID to 0x9338 or whatever. 

David.

Ray-electrotechie, 2021-03-02:
Hi David, Sorry, but I sort-of assumed that the Q and A were about customer to customer communication rather than being read by Bangood. I have bought lots of electronic bits from them over the last couple of years so I have an account. I will definitely post there but will run my suggested post past you first.

I hope that I don't upset you too much by bringing up the ILI9341/2 again, but... I have looked extensively at what read(0) returns and also studied the 9338 and ILI9341/2 specs to try to understand why. 95% of the time it returns C0C0 and now and again 8080. The spec for the 9338 and ILI9341/2 is very clear - command 0 is NOP and returns no arguments, ditto the soft reset 01h.
```
Command Function D/CX RDX WRX D17-8 D7 D6 D5 D4 D3 D2 D1 D0 Hex
No Operation          0        1     ↑       XX        0    0    0    0   0   0   0   0    00h
Software Reset        0        1     ↑       XX        0    0    0    0   0   0   0   1    01h
```

So, if by happenchance it turned out to be one of the family of ILI controllers like 9338/9341/2 I am not sure what would happen trying to do a read cycle when the operation has zero return arguments - maybe just noise on the inputs? (but C0C0 is fairly persistent). Could that be related to the connections with the resistive touch screen?

I am a bit embarassed because I have just looked carefully at the second card I bought. It looks identical and behaves identically, but this one has some sort of ID written in white on the red card (very tiny). It reads 487089AP3-200829

I haven't tried the examples with 0x9338 yet, but will do so later today.

prenticedavid, 2021-03-02:
As expected User registers 04, 09, 0A, 0B, 0C, 0D, 0E, 0F, 2E, 52, 56, 5F, 68, DA, DB, DC are "active".
The controller chip output driver is providing an output value(s).

A non-implemented register does not turn on any output driver.   So you "read" whatever the AVR had last put on the port.    And since the Touch Panel is on bit6, bit7 reading reg(0x00) will get `0x00 | 0b10000000 | 0b01000000`

Most MIPI style controllers can read an implemented register.
Since your controller does not seem to read any regular User "write" registers I would not expect it to read reg(0x00).

A non-MIPI controller like SSD1289 returns the ID for every register (except GRAM register)
ILI9320 reads several write registers and reads non-implemented as 0x0000

Controllers vary.   Such is life.

David.

Ray-electrotechie, 2021-03-02:
I feel particularly thick, but I think you are getting through to me. So, on this controller, whatever it is, all unimplemented commands,  AND commands that return no values as parameters, return their own command code OR with 0xC in the upper 4 bit nibble. Thus command zero (NOP) reads itself but (in both cases of the pcbs I have) with bits 6 and 7 set to 1 by the touch screen ORd into the command code giving C0. Reading 2 bytes gives C0C0. Soft reset gives C1C1 etc. as above. See, I eventually got it!  I will try to construct a message for the Q/A on bangood which asks for the chip number but tries to explain why without getting too technical.

closed 2022-02-12 by prenticedavid