prenticedavid/MCUFRIEND_kbv #167: BlackPill STM32F411CE  stm32duino core ver 1.9  unable to compile

JackDanl, 2021-03-14:
Maybe I am just being an idiot due to lack of sleep or something else. But I am not able to get most of the programs to compile, The compiler is giving several errors

C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp: In function 'void write24(uint16_t)':
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:84:5: error: 'write8' was not declared in this scope; did you mean 'write24'?
   84 |     write8(r);
      |     ^~~~~~
      |     write24
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp: In member function 'void MCUFRIEND_kbv::reset()':
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:92:5: error: 'setWriteDir' was not declared in this scope; did you mean 'setWriteError'?
   92 |     setWriteDir();
      |     ^~~~~~~~~~~
      |     setWriteError
In file included from C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:35:
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1090:31: error: 'RD_PORT' was not declared in this scope
 1090 | #define RD_OUTPUT  PIN_OUTPUT(RD_PORT, RD_PIN)
      |                               ^~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1111:37: note: in expansion of macro 'RD_OUTPUT'
 1111 | #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
      |                                     ^~~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:93:5: note: in expansion of macro 'CTL_INIT'
   93 |     CTL_INIT();
      |     ^~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1090:40: error: 'RD_PIN' was not declared in this scope
 1090 | #define RD_OUTPUT  PIN_OUTPUT(RD_PORT, RD_PIN)
      |                                        ^~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1111:37: note: in expansion of macro 'RD_OUTPUT'
 1111 | #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
      |                                     ^~~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:93:5: note: in expansion of macro 'CTL_INIT'
   93 |     CTL_INIT();
      |     ^~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1090:20: error: 'PIN_OUTPUT' was not declared in this scope; did you mean 'WR_OUTPUT'?
 1090 | #define RD_OUTPUT  PIN_OUTPUT(RD_PORT, RD_PIN)
      |                    ^~~~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1111:37: note: in expansion of macro 'RD_OUTPUT'
 1111 | #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
      |                                     ^~~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:93:5: note: in expansion of macro 'CTL_INIT'
   93 |     CTL_INIT();
      |     ^~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1093:31: error: 'WR_PORT' was not declared in this scope
 1093 | #define WR_OUTPUT  PIN_OUTPUT(WR_PORT, WR_PIN)
      |                               ^~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1111:48: note: in expansion of macro 'WR_OUTPUT'
 1111 | #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
      |                                                ^~~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:93:5: note: in expansion of macro 'CTL_INIT'
   93 |     CTL_INIT();
      |     ^~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1093:40: error: 'WR_PIN' was not declared in this scope
 1093 | #define WR_OUTPUT  PIN_OUTPUT(WR_PORT, WR_PIN)
      |                                        ^~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1111:48: note: in expansion of macro 'WR_OUTPUT'
 1111 | #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
      |                                                ^~~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:93:5: note: in expansion of macro 'CTL_INIT'
   93 |     CTL_INIT();
      |     ^~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1096:31: error: 'CD_PORT' was not declared in this scope
 1096 | #define CD_OUTPUT  PIN_OUTPUT(CD_PORT, CD_PIN)
      |                               ^~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1111:59: note: in expansion of macro 'CD_OUTPUT'
 1111 | #define CTL_INIT()   { GPIO_INIT(); RD_OUTPUT; WR_OUTPUT; CD_OUTPUT; CS_OUTPUT; RESET_OUTPUT; }
      |                                                           ^~~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:93:5: note: in expansion of macro 'CTL_INIT'
   93 |     CTL_INIT();
      |     ^~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1096:40: error: 'CD_PIN' was not declared in this scope
 1096 | #define CD_OUTPUT  PIN_OUTPUT(CD_PORT, CD_PIN)
      |                                        ^~~~~~
[... 653 lines cut ...]
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1094:37: error: 'CD_PIN' was not declared in this scope
 1094 | #define CD_COMMAND PIN_LOW(CD_PORT, CD_PIN)
      |                                     ^~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:860:13: note: in expansion of macro 'CD_COMMAND'
  860 |             CD_COMMAND;
      |             ^~~~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:861:13: error: 'write8' was not declared in this scope; did you mean 'write24'?
  861 |             write8(cmd);
      |             ^~~~~~
      |             write24
In file included from C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:35:
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1095:20: error: 'PIN_HIGH' was not declared in this scope
 1095 | #define CD_DATA    PIN_HIGH(CD_PORT, CD_PIN)
      |                    ^~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:864:17: note: in expansion of macro 'CD_DATA'
  864 |                 CD_DATA;
      |                 ^~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\utility/mcufriend_shield.h:1098:20: error: 'PIN_HIGH' was not declared in this scope
 1098 | #define CS_IDLE    PIN_HIGH(CS_PORT, CS_PIN)
      |                    ^~~~~~~~
C:\Users\JackDanl\OneDrive\Documents\Arduino\libraries\MCUFRIEND_kbv\MCUFRIEND_kbv.cpp:872:13: note: in expansion of macro 'CS_IDLE'
  872 |             CS_IDLE;
      |             ^~~~~~~
exit status 1
Error compiling for board Generic STM32F4 series.

JackDanl, 2021-03-14:
I was able to get LCD_ID_ReadReg working 
using 
#define LCD_RST PB9
#define LCD_CS PB8
#define LCD_RS PB7
#define LCD_WR PB6
#define LCD_RD PB0

#define LCD_D0 PA0
#define LCD_D1 PA1
#define LCD_D2 PA2
#define LCD_D3 PA3
#define LCD_D4 PA4
#define LCD_D5 PA5
#define LCD_D6 PA6
#define LCD_D7 PA7

Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 54 80 66	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 00 08	Get Power Mode
reg(0x000C) 00 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 93 30 93 30	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 02 02 3B 3B	Display Control
reg(0x00B7) 00 06	Entry Mode Set
reg(0x00BF) 00 00 00 00 00 00	ILI9481, HX8357-B
reg(0x00C0) 00 0E 0E 0E 0E 0E 0E 0E 0E	Panel Control
reg(0x00C8) 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA
reg(0x00CC) 00 04	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 00 00 00	NVM Read
reg(0x00D3) 00 00 94 86	ILI9341, ILI9488
reg(0x00D4) 00 00 00 00	Novatek ID
reg(0x00DA) 00 54	RDID1
reg(0x00DB) 00 80	RDID2
reg(0x00DC) 00 66	RDID3
reg(0x00E0) 00 09 1C 3D 02 15 09 74 BD 0D 0F 1A 06 03 23 00	GAMMA-P
reg(0x00E1) 00 05 15 3B 0F 11 0C 05 68 52 07 09 05 05 2C 02	GAMMA-N
reg(0x00EF) 00 80 00 10 60 40	ILI9327
reg(0x00F2) 00 18 A3 12 02 B2 12 FF 10 00 00 00	Adjust Control 2
reg(0x00F6) 00 54 80 66	Interface Control

prenticedavid, 2021-03-14:
Ok,  so you have used the wiring scheme intended for a regular ARDUINO_BLUEPILL_F103CB but on a non-standard board.
```
//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1**
```

That means that you need a SPECIAL.

The library is intended for Arduino Shields that plug into Arduino header sockets.    If you want me to write a SPECIAL to match your wiring,  please say so.

David.

JackDanl, 2021-03-14:
Thank you for your quick response
I was trying to figure it out myself, but if you wouldn't mind writing it, I can see exactly where I am going wrong with mine.

prenticedavid, 2021-03-14:
Insert this block in mcufriend_special.h
Enable the SPECIAL in the usual way i.e. USE_SPECIAL and USE_BLACKPILL_JACKDANL

Obviously untested.   Please let me know how you get on.
It should work with board set to "Generic_F411CE" or preferably "BlackPill_F411CE"

```
//############################ BLACKPILL_JACKDANL F411CE    ################################
#elif defined(USE_BLACKPILL_JACKDANL) && (defined(ARDUINO_GENERIC_F411CE) || defined(ARDUINO_BLACKPILL_F411CE))
#warning Using special for BLACKPILL_JACKDANL

//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1**

#if defined(__STM32F1__)   //weird Maple Core
#define REGS(x) regs->x
#else                      //regular ST Core
#define REGS(x) x
#endif
#define PIN_HIGH(port, pin)   (port)-> REGS(BSRR) = (1<<(pin))
#define PIN_LOW(port, pin)    (port)-> REGS(BSRR) = (1<<((pin)+16))
#define PIN_MODE2(reg, pin, mode) reg=(reg&~(0x3<<((pin)<<1)))|(mode<<((pin)<<1))
#define GROUP_MODE(port, reg, mask, val)  {port->REGS(reg) = (port->REGS(reg) & ~(mask)) | ((mask)&(val)); }

#define WRITE_DELAY { WR_ACTIVE2; WR_ACTIVE; } //100MHz
#define READ_DELAY  { RD_ACTIVE4; RD_ACTIVE2; }
#define GPIO_INIT()   { RCC->AHB1ENR |= RCC_AHB1ENR_GPIOAEN | RCC_AHB1ENR_GPIOBEN | RCC_AHB1ENR_GPIOCEN; }
#define PIN_OUTPUT(port, pin) PIN_MODE2((port)->MODER, pin, 0x1)

#define RD_PORT GPIOB
//#define RD_PIN  5
#define RD_PIN  0  //hardware mod to Adapter.  Allows use of PB5 for SD Card
#define WR_PORT GPIOB
#define WR_PIN  6
#define CD_PORT GPIOB
#define CD_PIN  7
#define CS_PORT GPIOB
#define CS_PIN  8
#define RESET_PORT GPIOB
#define RESET_PIN  9

// configure macros for the data pins
#define write_8(d)    { GPIOA->REGS(BSRR) = 0x00FF << 16; GPIOA->REGS(BSRR) = (d) & 0xFF; }
#define read_8()      (GPIOA->REGS(IDR) & 0xFF)
//                                         PA7 ..PA0
#define setWriteDir() { setReadDir(); \
                        GPIOA->MODER |=  0x5555; }
#define setReadDir()  { GPIOA->MODER &= ~0x5555; }

#define write8(x)     { write_8(x); WRITE_DELAY; WR_STROBE; }
#define write16(x)    { uint8_t h = (x)>>8, l = x; write8(h); write8(l); }
#define READ_8(dst)   { RD_STROBE; READ_DELAY; dst = read_8(); RD_IDLE; }
#define READ_16(dst)  { uint8_t hi; READ_8(hi); READ_8(dst); dst |= (hi << 8); }

```

prenticedavid, 2021-03-16:
Bump.

Have you tried it ?
Or not had the opportunity yet ?

David.

JackDanl, 2021-03-16:
I have only have the chance to run a few tests on it, and then had to break it down till later this week. But when I did have it setup, the graphics test at least worked perfectly. 

Thank you again for your help.

prenticedavid, 2021-03-16:
Jolly good.

JackDanl, 2021-03-21:
[image]

Here is the timings screenshot for the board/LCD

Thank you again for your help!

prenticedavid, 2021-03-21:
That looks about right.    That would be 0.78 sec on a 240x320 screen.   Which is about as fast as most controllers will go.
Are you sure that there are no glitches anywhere ?
Is the read ok ?   (Software Scroll)

David.

closed 2021-08-24 by prenticedavid