prenticedavid/MCUFRIEND_kbv #169: Support for ILI9488/HX8357C

fajarnurprasetyo, 2021-04-05:
[image]

Arduino Uno shield

prenticedavid, 2021-04-05:
Please post a link to the actual display that you have bought.  e.g. Ebay sale page.

It looks like an 18x2 header.   So it is probably a Shield designed for a Mega2560.
From extras/mcufriend_how_to.txt:

```
17. If you do not have a standard Uno Shield, you can add a SPECIAL to the mcufriend_special.h
    Edit mcufriend_shield.h:  #define USE_SPECIAL
    Edit mcufriend_special.h: e.g. #define USE_MEGA_16BIT_SHIELD
    If your "special" is write-only,  the library can not read the ID.  It always returns 0xD3D3

```

Mega2560 Shields are almost always read-only.   So you need to force the correct ID e.g. `tft.begin(0x9488);`

David.

fajarnurprasetyo, 2021-04-05:
@prenticedavid
I was try regular example sketch Glue_Demo_480x360
I've added `#define USE_SPECIAL` to mcufriend_shield.h
and edit UTFTGLUR.h like this
```
    void InitLCD(byte orientation=LANDSCAPE) {
        MCUFRIEND_kbv::reset();
//      uint16_t ID = MCUFRIEND_kbv::readID();
//       if (ID == 0) ID = 0x9341;        //DealExtreme with EXTC=0
//       if (ID == 0x8989) ID = 0x1289;
//       if (ID == 0xD3D3) ID = 0x9481;   //write-only controller
//       if (ID == 0xD3D3) ID = 0x9486;   //write-only controller
//       if (ID == 0xD3D3) ID = 0x9090;   //write-only controller HX8357-D
//       if (ID == 0x9327 && orientation == LANDSCAPE) orientation = 3;
        MCUFRIEND_kbv::begin(0x9488);
        MCUFRIEND_kbv::setRotation(_orient = orientation);
        _radius = 4;
    }
```
Same result but the white area now changes to black,
**This is a description of the product i bought**
Features 
Support Arduino Mega2560 Directly inserted 
OnBorad level conversion chip for 5V/3.3V MCU 
Compatible with 3.3/5V operation voltage level 
Compatible with Arduino-Series development Board 
Compatible with UTFT / UTFT_Buttons /Utouch Library for arduino 
Provided 12-examples with Arduino ,3-examples with STM32? 
With SD Card Socket 
With SPI FLASH circuit 

Specifications: 
Input Voltage: DC 3.3-5.5V 
Screen Size: 3.5" 
Screen Material: TFT 
Resolution: 320X480 
Drive IC: ILI9488 / HX8357C
Interface mode: parallel 8 bit 
Module Pin: 28Pin (8+8+6+6)

prenticedavid, 2021-04-05:
Please provide a link.   You might not find the actual shop that you bought from.   But you can probably find the identical item on Ebay, AliExpress, ... and post a link to **that** item.

Your text description does not make sense.    It is much better if I can see a photo of the pcb from the Ebay link.

David.

fajarnurprasetyo, 2021-04-05:
https://www.ebay.com/itm/174508022307
https://a.aliexpress.com/_mtpoEqB
similar but not the same
sorry, can't find on ebay or aliexpress

Here the picture of the pcb lcd i've bought,,
[image]

prenticedavid, 2021-04-05:
My apologies.   Your photo in #0 was difficult to see.   I thought that I saw 18x2 header pins on the display shield.   On reflection it was probably the 18x2 header socket on the Mega2560.

Anyway,  your new photos show a regular Blue 3.5 inch Uno Shield.
Please do not dismantle Shields.   I only needed to see the top photo.

You don't need any "USE_SPECIAL".
Just install MCUFRIEND_kbv **via the IDE Library Manager.**
Plug in the Shield.
Run **all** of the examples.    Every example should run 100% out of the box.

If you have a problem with any of the examples,  quote example name and copy-paste the output from the Serial Terminal.

If you have installed via ZIP or altered any library files I strongly advise you to delete the folder.   Then install a fresh copy via the IDE Library Manager.

If you are experienced with Git we can do things differently.   Otherwise it is best to use Arduino IDE tools.

David.

fajarnurprasetyo, 2021-04-05:
i was delete all library then clone library directly from github but the result is same, i was try graphictest_kbv.ino
[image]

prenticedavid, 2021-04-05:
If you follow my advice we can get you going within minutes.

fajarnurprasetyo, 2021-04-06:
@prenticedavid ok, how..

prenticedavid, 2021-04-06:
Just install MCUFRIEND_kbv **via the IDE Library Manager.**
Plug in the Shield.
Run all of the examples. Every example should run 100% out of the box.

If you have a problem with any of the examples, **quote** example name and **copy-paste** the output from the Serial Terminal.

It does not matter whether the Arduino IDE is running on Linux, Mac or Windows.   There is always a Serial Terminal.   And you can always copy-paste.

David.

prenticedavid, 2021-04-07:
How is the copy-paste doing ?

You can always type the text manually.   But copy-paste is MUCH safer.

David.

fajarnurprasetyo, 2021-04-07:
I haven't tried it yet.
I will try tomorrow, still busy with my work.

closed 2021-08-24 by prenticedavid