prenticedavid/MCUFRIEND_kbv #170: tft.readID() finds: ID = 0xC5C

prochor666, 2021-04-07:
Full serial output;

tft.readID() finds: ID = 0xC5C

MCUFRIEND_kbv version: 2.9.9

This ID is not supported
look up ID in extras/mcufriend_how_to.txt
you may need to edit MCUFRIEND_kbv.cpp
to enable support for this ID
e.g. #define SUPPORT_8347D

New controllers appear on Ebay often
If your ID is not supported
run LCD_ID_readreg.ino from examples/
Copy-Paste the output from the Serial Terminal
to a message in Displays topic on Arduino Forum
or to Issues on GitHub

Note that OPEN-SMART boards have diff pinout
Edit the pin defines in LCD_ID_readreg to match
Edit mcufiend_shield.h for USE_SPECIAL
Edit mcufiend_special.h for USE_OPENSMART_SHIELD_PINOUT

==================================================================

Eshop suggest this:

uncomment "#define USE_SPECIAL" in utility/mcufriend_shield.h
uncomment "#define USE_OPENSMART_SHIELD_PINOUT" in utility/mcufriend_special.h
uncomment "#define SUPPORT_8352A" (or SUPPORT_8352B, in new cersions) in /mcufriend_kbv.cpp

https://www.laskarduino.cz/open-smart-240x400-barevny-lcd-tft-displej-3-2-shield-arduino/

But; 

There is no USE_OPENSMART_SHIELD_PINOUT in mcufiend_special.h

prenticedavid, 2021-04-07:
1. Install MCUFRIEND_kbv via IDE Library Manager.
2. read C:\Users\...\Documents\Arduino\libraries\Mcufriend_kbv\extras\mcufriend_how_to.txt
```

19. OPEN-SMART Shields have different wiring to regular Uno Shields:
    Edit utility/mcufriend_shield.h:  #define USE_SPECIAL
    Edit utility/mcufriend_special.h: #define USE_OPENSMART_SHIELD_PINOUT
    Edit MCUFRIEND_kbv.cpp:           #define SUPPORT_8352B

20. OPEN-SMART Shields can not read LM75A on a Mega because there are no SDA/SCL pins next on AREF header.
    Uno:  LM75A on pcb works.  Difficult to add external I2C devices when Shield is plugged in.
	Uno:  All use of SPI bus should use SPI.beginTransaction(), SPI.endTransaction()
	Leo:  I do not support USE_OPENSMART_SHIELD_PINOUT_LEO
	Mega: Old boards do not have SDA/SCL on AREF header.   Only external I2C devices on COMMS header
	Due:  MAX809 Reset chip interferes with 3.3V RST on SPI header.  Manual reset for Upload / Run.

```	

In C:\Users\...\Documents\Arduino\libraries\Mcufriend_kbv\utility\mcufriend_special.h
```
//#define USE_OPENSMART_SHIELD_PINOUT_UNO
//#define USE_OPENSMART_SHIELD_PINOUT_MEGA
//#define USE_OPENSMART_SHIELD_PINOUT_DUE //thanks Michel53
```

Un-comment the appropriate macro.   e.g.  USE_OPENSMART_SHIELD_PINOUT_MEGA

Thanks.   I will update the how_to.txt file to say appropriate: USE_OPENSMART_SHIELD_PINOUT_xxxx

Do you have HX8352A or HX8352B ?

If you have HX8352A I would appreciate your assistance.    You need to use the Beta on GitHub instead of the Release-v2.9.9

David.

prochor666, 2021-04-09:
1 . Sorry, I missed you do not support Leonardo
2. With UNO display worked well, but died after few tests and I need to buy another one.

prenticedavid, 2021-04-09:
Do you have HX8352A or HX8352B ?

It should work with Uno.   It could work with Leonardo (if you really want).

Personally,   I think that the Open-Smart Shield is a bit pointless.

David.

prochor666, 2021-04-09:
It's HX8352A. Can you suggest better shield? Leonardo is not required, I jsut tried what I have.

prenticedavid, 2021-04-09:
I don't have a HX8352A.   But am fairly confident that the HX8352A (ID=0x5252) code should work ok with the Uno.   Enable `#define SUPPORT_8352A`

If you can describe any problem,  I can probably sort it out.

I do have an Open-Smart HX8352B to compare with.    However Himax controllers vary considerably with the suffix-letter.

If Leonardo is important,  I can add it.   But Leo USB bootloader and CDC code uses up a lot of Flash.

David.

closed 2021-08-24 by prenticedavid