prenticedavid/MCUFRIEND_kbv #172: Mini maple compilation errors

zfranco55, 2021-04-10:
I'm trying the example: graphic test.ino on the PlatformIO platform.
with Blue Pill card stm32f103c8t6 64k totto ok.
with Maple 128K mini card i get the following errors:

[image]
[image]

Excuse me, if I send the error images, but I don't know how to save the errors to the file in PlatformIO

prenticedavid, 2021-04-10:
I happen to have a MAPLE_REV3 board which has Arduino headers.

That is the only "Maple_xxxx" board that is supported in mcufriend_shield.h

If you have a MapleMini board you need a SPECIAL.    It does not have any Arduino header sockets.

God gave you the Arduino IDE.   I suggest that you install the official STM32 Core (from ST)
If you choose a wiring scheme,   I will write a SPECIAL for you.   (but only after you have verified with LCD_ID_readreg.ino)

When it is running successfully in the IDE,   we can think of building in PlatformIO.

David.

zfranco55, 2021-04-11:
are in difficulty for the definition of the LD_RST pin which is connected to the Reset pin of the board.
I tried to rename it as "Reset" and as #define LCD_RST NRST, but it always gives me error:
'NRST' was not declared in this scope
could you help me ?
thank you
my board is:
[https://stm32-base.org/boards/STM32F103C8T6-Maple-Mini-Clone.html
](url)
[image]
Also not being an expert, I can't understand what you mean official STM32 core (from ST)

zfranco55, 2021-04-11:
installed on an arduino uno board, the program gives me the following values:

> 
Read Registers on MCUFRIEND UNO shield

controllers either read ⸮Read Registers on MCUFRIEND UNO shield
controllers either read as single 16-bit
e.g. the ID is at readReg(0)
or as a sequence of 8-bit values
in special locations (first is dummy)

reg(0x0000) 00 00	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 00 00 00 00	Manufacturer ID
reg(0x0009) 00 00 61 00 00	Status Register
reg(0x000A) 08 08	Get Power Mode
reg(0x000C) 66 66	Get Pixel Format
reg(0x0061) 00 00	RDID1 HX8347-G
reg(0x0062) 00 00	RDID2 HX8347-G
reg(0x0063) 00 00	RDID3 HX8347-G
reg(0x0064) 00 00	RDID1 HX8347-A
reg(0x0065) 00 00	RDID2 HX8347-A
reg(0x0066) 00 00	RDID3 HX8347-A
reg(0x0067) 00 00	RDID Himax HX8347-A
reg(0x0070) 00 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 00 00 00	RD_DDB SSD1963
reg(0x00B0) 00 00	RGB Interface Signal Control
reg(0x00B4) 00 00	Inversion Control
reg(0x00B6) 00 00 00 00 00	Display Control
reg(0x00B7) 00 00	Entry Mode Set
reg(0x00BF) 00 02 04 94 81 FF	ILI9481, HX8357-B
reg(0x00C0) 00 10 3B 00 02 11 00 00 00	Panel Control
reg(0x00C8) 00 00 44 06 11 44 55 0A 08 17 33 77 11	GAMMA
reg(0x00CC) 00 00	Panel Control
reg(0x00D0) 00 00 43	Power Control
reg(0x00D2) 00 01 22 00 00	NVM Read
reg(0x00D3) 00 01 22 00	ILI9341, ILI9488
reg(0x00D4) 00 01 22 00	Novatek ID
reg(0x00DA) 00 00	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) 00 00	RDID3
reg(0x00E0) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-P
reg(0x00E1) 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00	GAMMA-N
reg(0x00EF) 00 00 00 00 00 00	ILI9327
reg(0x00F2) 00 00 00 00 00 00 00 00 00 00 00 00	Adjust Control 2
reg(0x00F6) 00 00 00 00	Interface Control

>

prenticedavid, 2021-04-11:
You have an ILI9481 controller
`
reg(0x00BF) 00 02 04 94 81 FF ILI9481, HX8357-B
`

If you have a successful wiring scheme on your Maple Mini LCD_ID_readreg.ino will report the same values.

Google "Arduino IDE STM32 Core".   I got https://github.com/stm32duino/Arduino_Core_STM32

It explains how to add the .JSON file and how to install with the IDE Boards Manager.

David.

zfranco55, 2021-04-11:
with the mini maple card and the ili9481 display I got:
on serial monitor:

> 
Read Registers on MCUFRIEND UNO shield
controllers either read 
reg(0x0000) 50 38	ID: ILI9320, ILI9325, ILI9335, ...
reg(0x0004) 04 04 3C 3C	Manufacturer ID
reg(0x0009) 59 39 39 39 00	Status Register
reg(0x000A) 5A 00	Get Power Mode
reg(0x000C) 5C 04	Get Pixel Format
reg(0x0061) 71 00	RDID1 HX8347-G
reg(0x0062) 72 00	RDID2 HX8347-G
reg(0x0063) 73 00	RDID3 HX8347-G
reg(0x0064) 74 04	RDID1 HX8347-A
reg(0x0065) 75 3D	RDID2 HX8347-A
reg(0x0066) 76 3E	RDID3 HX8347-A
reg(0x0067) 77 3F	RDID Himax HX8347-A
reg(0x0070) 70 00	Panel Himax HX8347-A
reg(0x00A1) 00 00 B9 B9 B9	RD_DDB SSD1963
reg(0x00B0) F0 B8	RGB Interface Signal Control
reg(0x00B4) 04 BC	Inversion Control
reg(0x00B6) F6 BE BE BE 04	Display Control
reg(0x00B7) 04 04	Entry Mode Set
reg(0x00BF) FF 04 04 BF BF BF	ILI9481, HX8357-B
reg(0x00C0) D0 B8 00 00 B8 B8 B8 B8 00	Panel Control
reg(0x00C8) D8 00 00 00 B8 B8 B8 00 00 00 B8 B8 B8	GAMMA
reg(0x00CC) 04 04	Panel Control
reg(0x00D0) 00 00 00	Power Control
reg(0x00D2) 00 00 BA BA BA	NVM Read
reg(0x00D3) 00 00 BB BB	ILI9341, ILI9488
reg(0x00D4) D4 BC 04 04	Novatek ID
reg(0x00DA) DA BA	RDID1
reg(0x00DB) 00 00	RDID2
reg(0x00DC) DC BC	RDID3
reg(0x00E0) 00 00 00 B8 B8 B8 00 00 00 B8 B8 B8 00 00 00 B8	GAMMA-P
reg(0x00E1) F1 00 00 B9 B9 B9 B9 00 00 B9 B9 B9 00 00 00 B9	GAMMA-N
reg(0x00EF) FF 04 04 04 BF BF	ILI9327
reg(0x00F2) 00 00 BA BA BA BA 00 00 BA BA BA 00	Adjust Control 2
reg(0x00F6) F6 04 04 04	Interface Control

> 
on terminal IDE:

> 
Sketch uses 13816 bytes (10%) of program storage space. Maximum is 131072 bytes.
Global variables use 844 bytes (4%) of dynamic memory, leaving 19636 bytes for local variables. Maximum is 20480 bytes.
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.6.0                  
      -------------------------------------------------------------------

ST-LINK SN  : 1C2A05012612344D314B4E00
ST-LINK FW  : V2J31S7
Board       : --
Voltage     : 3,23V
SWD freq    : 4000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x410
Revision ID : Rev X
Device name : STM32F101/F102/F103 Medium-density
Flash size  : 128 KBytes
Device type : MCU
Device CPU  : Cortex-M3

Memory Programming ...
Opening and parsing file: LCD_ID_readreg.ino.bin
  File          : LCD_ID_readreg.ino.bin
  Size          : 14108 Bytes
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 13]
Download in Progress:

File download complete
Time elapsed during download operation: 00:00:00.915

RUNNING Program ... 
  Address:      : 0x8000000
Application is running
Start operation achieved successfully

> 
and now what should I do?

prenticedavid, 2021-04-11:
Go on.   Your LCD_ID_readreg report is clearly garbage !!

I suggest that you identify which Core you are using.
e.g. with your working BluePill.
e.g. run LCD_ID_readreg on BluePill

Then wire up your Maple Mini with the same scheme.
e.g. run LCD_ID_readreg on Maple Mini.

Adapting a BluePill SPECIAL to Maple Mini is trivial.

David.

prenticedavid, 2021-04-11:
Did you look at https://github.com/prenticedavid/MCUFRIEND_kbv/blob/Release_v2.9.9/extras/unused/mcufriend_special_2.h

There is a SPECIAL block : USE_VADIMEL_MAPLEMINI

I have not tested it.   It is pretty old.   But presumably Vadimel had designed a sensible wiring scheme.
I suggest that you use different conditional for the REGS() macro e.g. look at the current mcufriend_shield_h

David.

zfranco55, 2021-04-12:
I did as you told me and with the Arduino IDE I found that with the same connections (between blupill and miniMaple) and the display works, if set as a BluePill F103CB board (or C8 with 128k) it seems to work now I will try to verify.
Unfortunately on the PlatformIO platform I have not found the same setting for the card. I have tried some f103CB 128k, but they don't work at compile time and give me errors. If you know which card to set, can you tell me?
Otherwise I will pass the project under the arduino IDE.
However, thanks for the support and patience you had to follow me.

prenticedavid, 2021-04-12:
What wiring did you use ?

What settings in the Arduino IDE ?
there are multiple options.

I don't use PlatformIO.   But I might have a look at it _if I get appropriate feedback from you_.
e.g. answering those questions _accurately_.

David.

zfranco55, 2021-04-12:
I hope the information is sufficient.
I am attaching the IDE screen, the wired table and the image I referred to
for the wires / names

prenticedavid, 2021-04-12:
Please read any message after you have posted it.

I do not see any attachments.

zfranco55, 2021-04-12:
sorry but I see them, now I try to put them on drive as attachments
 collegamenti.odt
<https://drive.google.com/file/d/1hQKe2E3HtfXDgQw4qsRjBwXZD6GuicGQ/view?usp=drive_web>

 921px-Maple_Mini_Pinout.svg
<https://drive.google.com/file/d/14XmE6a01dxzSDUUDJJi8ateQwe-Jhc-O/view?usp=drive_web>

 IDE_Board_Setting.jpg
<https://drive.google.com/file/d/1JZdMRmX_4OSwbZwLnBzwuaNniiJg0-g4/view?usp=drive_web>

zfranco55, 2021-04-12:
the cause, perhaps it is due to my reply made directly from my e-mail.
Now I am attaching the files.
I'm sorry.
[collegamenti.pdf](https://github.com/prenticedavid/MCUFRIEND_kbv/files/6298944/collegamenti.pdf)
[image]
[image]

prenticedavid, 2021-04-12:
It looks as if you have used the regular mcufriend_shield.h wiring for BLUEPILL but moved the LCD_RST pin.
```

#warning Uno Shield on BLUEPILL

//LCD pins  |D7 |D6 |D5 |D4 |D3 |D2 |D1 |D0 | |RD |WR |RS |CS |RST| |SD_SS|SD_DI|SD_DO|SD_SCK|
//STM32 pin |PA7|PA6|PA5|PA4|PA3|PA2|PA1|PA0| |PB0|PB6|PB7|PB8|PB9| |PA15 |PB5  |PB4  |PB3   | **ALT-SPI1**

```

Yes,  you are using the official STM32 Core.   
You could enable the USB Support: "CDC generic Serial" or use a USB-UART Adapter.
You seem to upload via a STLink-2 
You could use a bootloader.

I hope that everything is working now.
Please note that the links in your previous message are all "Top Secret".    No one is allowed to read them.

David.

closed 2021-08-24 by prenticedavid