prenticedavid/MCUFRIEND_kbv #175: Slow operation Using UNO Module(ILI9486 with touch) with arduino Mega!

tusharvb19, 2021-04-29:
I tried to use the UNO module on arduino Mega and found that it is very slow!
With UNO it tookes 14 seconds while with Mega it is taking 37 seconds to load the graphictest_kbv!
Need help to make it fast.
I don't know why the pins used are same for UNO and Mega!
TIA.

prenticedavid, 2021-04-29:
You chose to use a Uno shield in a Mega2560.
If you want the best performance on a Mega2560,  you should buy a Mega2560 Shield.   (and use the appropriate SPECIAL e.g. USE_MEGA_16BIT_SHIELD )

I am guessing that you have a 320x480 display.   The Adafruit Tests will take about 4.5 seconds with USE_MEGA_16BIT_SHIELD.

David.

tusharvb19, 2021-04-29:
I don't have the Mega shield here..
Need to use UNO only. I understood it is 8 bit. But why it runs fast on UNO with same arduino pins?
Can you help to make it fast on Mega?

prenticedavid, 2021-04-29:
God invented Shields.   You plug them into the Arduino header sockets.   Everything works.

If you want it to run faster on the Mega2560 you need to make your own Shield.   Or buy a ready-made Mega2560 Display Shield.

Of course you could just read the mcufriend_special.h file.   There is a clue.    No one has ever mentioned it.    It is not as impressive as buying the correct `USE_MEGA_16BIT_SHIELD` display in the first place.

David.

closed 2021-08-24 by prenticedavid